z88dk/z88dk #1714: base_graphics undefined for zx81 compile

emuslor, 2021-03-14:
Appears to be a result of this changeset, which moves the PUBLIC declaration of the base_graphics symbol:
https://github.com/z88dk/z88dk/commit/29e55c27450bef6ac3bfa09c5921385464dce434#diff-6d27a27e64075cf175256c387b1191b96ba68719a8372e911dd1cca4e7b5166c
Since the symbol is not publicized in the other location where DEFINED_basegraphics may be defined, zcc can no longer build existing hrg programs.  Example command line: zcc +zx81 -subtype=wrx -clib=wrx -s -o demo.bin demo.c
Error at file 'target/zx81/graphics_hr/_clg_hr.asm' line 31: symbol 'base_graphics' not defined
Error at file 'target/zx81/graphics_hr/copytxt.asm' line 31: symbol 'base_graphics' not defined
Error at file 'target/zx81/graphics_hr/copytxt.asm' line 119: symbol 'base_graphics' not defined
Error at file 'target/zx81/graphics_hr/pixladdr.asm' line 46: symbol 'base_graphics' not defined
Errors in source file D:/devel/cross/z88dk/lib/config\..\..\\lib\target\zx81\classic\zx81_crt0.asm:
Error at file 'target/zx81/graphics_hr/_clg_hr.asm' line 31: symbol 'base_graphics' not defined
Error at file 'target/zx81/graphics_hr/copytxt.asm' line 31: symbol 'base_graphics' not defined
Error at file 'target/zx81/graphics_hr/copytxt.asm' line 119: symbol 'base_graphics' not defined
Error at file 'target/zx81/graphics_hr/pixladdr.asm' line 46: symbol 'base_graphics' not defined
Moving the single-line change back in crt_section.asm corrects the build issue.

zx70, 2021-03-14:
thank you for the heads up, I can't see a direct connection to the krokha target though.

emuslor, 2021-03-14:
Yeah, I don't know anything about that target, except that the new code for it includes the existing common file that was changed.

suborb, 2021-03-14:
It looks like I was trying to trim down memory usage - the Kroha only has 512 bytes of RAM.

I think that variable just needs to be pulled into the library and linked on demand tbh.

suborb, 2021-03-14:
Sticking it in the library was a bit tricky since some targets need an extra -l to gain graphics support.

So, since it was just +zx81 that was affected, I've exported the definition when this condition is hit, hopefully that will sort it out.

zx70, 2021-03-14:
it will surely work,  thank you

emuslor, 2021-03-15:
Yeah, that works fine (at least for zx81 targets).

closed 2021-03-15 by suborb