z88dk/z88dk #1715: (classic) CP/M target: adding primitives for GSX

zx70, 2021-03-14:
- [ ] Struct definitions for parameter passing
- [ ] GIOS function code/name declarations
- [ ] Support API library

zx70, 2021-03-15:
[image]

zx70, 2021-03-15:
[image]

zx70, 2021-03-16:
Testing the TS803 is tricky, the B drive is mounted R/O, so the file must be copied to A: before issuing GENGRAF.
The B drive will give an error but the file will be still accessible.
The keyboard needs repeated keypresses because of a faulty emulation.

By the way here's a screenshot explaining how to enable the GSX graphics before running a program.

[image]

zx70, 2021-03-18:
[image]

+ turtle graphics  ;)    This one is reasonably fast and should support the graphics on  CLISP  as well.

zx70, 2021-03-18:
[image]

This one is stretched, cropped and so slow !!

zx70, 2021-03-24:
I managed to compute the exact position for the physical pixel out from the virtual addresses, no crop/stretch anymore!
Moreover we now have the real values for getmaxx() and getmaxy()

zx70, 2021-03-25:
-pragma-redirect=fputc_cons=_putc_gsx

[image]

zx70, 2021-03-25:
[image]

zx70, 2021-03-25:
I've been putting some time on the Acron CP/M disk formats.
It is quite a pain, for some reason the only disk images I was able to boot are in "dsd" format, which I suspect having 2 separate disks/sides in a single file ???

zx70, 2021-03-30:
Too bad, I got this faster stencil implementation by using a trick (using the 'BAR' objects and XORing the bar borders with the contained patterns.
Sadly this does not work on the TS-803, and probably on other targets as well.

zx70, 2021-03-30:
[image]
This is crap, but works everywhere.   Uses the dotted line features.

zx70, 2021-03-30:
crossing the line drawing directions
[image]

zx70, 2021-03-30:
It is way too crap.. I'm reverting to the first version.
The problem is that it will work correctly only on those GSX drivers which:
  - support fill patterns
  - apply the XOR drawing styles to area borders

Apparently there's no way to exclude the borders, I tried by changing its colour, setting the line width to zero, etc..

zx70, 2021-04-01:
I gave up with the Acorn format, I'm still hoping to find a way to transfer the com files to a cpm disk image,  though

zx70, 2021-04-02:
Systems which declared the GSX compatibility

Televideo TS803, TPC-1, TS800 (640x240)
Ontel Amigo (640x300)
Apple II (CP/M card)
Amstrad CPC
Amstrad PCW
BBC Micro (Z80 card)

zx70, 2021-04-03:
[image]

JohnCElliott, 2025-07-19:
This is years late but you're unfortunately correct that GSX can't turn off polygon borders. A function was added to GEM to support this (number 104, vsf_perimeter) and it's also present in the GSX drivers supplied with the 'Activity' GUI on the Apricot PC. Bit late to start adding it to all the other GSX drivers that are out there, though.

JohnCElliott, 2025-07-23:
I haven't had the BBC drivers running, but looking at the disassembly they can't do filled polygons and don't even fall back to drawing the outline as DRI recommended, so I can well believe they weren't adequate for DR Graph.