z88dk/z88dk #1718: newlib - GSX-80 & VDI Specification

feilipu, 2021-03-18:
Noting that #1715 is being worked on for `classic`, perhaps it is worth discussing a thought process I'm working on now towards getting a "standardised" graphics library available for the RC2014, for which graphics device drivers can be written.

And, from reading, I've developed some (potentially unpopular) opinions that I'd like to test.
Any thoughts / comments / welcome, please. :fire: 

The [GSX-80 Graphics System eXtension](https://www.seasip.info/Cpm/gsx.html) for 8080 (Z80), is the only thing available that has some history. And its method of using a G/BIOS call to trap graphics commands is still appropriate for what we need.

1. In the context of z88dk `newlib` the `GENGRAPH.ASM` tool to "graphicise" an application and `GSX.SYS` library to support the GIOS / BIOS reroute are obsolete and there's no point to build to them.

2. The `GSX.SYS` code does nothing more than trap a the G/BIOS call, and assign a VDI driver. These functions are much more elegantly supported as an integral z88dk library, during application build time. The ability to swap VDI drivers on the fly is not a high priority today, as video hardware options are limited and tied to specific machines anyway.

3. For z88dk `newlib`, it makes more sense add an integral GSX library (that can support all the calling options we need), that can go in / near the CP/M `newlib` library and be automatically linked to any relevant application at build time.

4. The VDI Specification is defined in [Appendix B of the GSX-80 Graphics Extension Programmers Guide](https://github.com/feilipu/ux_module/files/6154091/GSX-80_Graphics_Extension_Programmers_Guide_Nov82.pdf). The VDI seems to be the best (most complete, and most appropriate) specification for a z88dk GSX library to work towards.

_Background_
I'm working on a RC2014 UX Module, and the z80 interface to the [VJET library](https://github.com/feilipu/ux_module/issues/1) (by @Wuerfel21) running on a Parallax Propeller. And rather than re-invent all the wheels, I'd rather use the best of what's already around.<br>
So I'm fishing for some direction. :fishing_pole_and_fish: <br>
TIA.

zx70, 2021-03-18:
Hi @feilipu !
First of all let me explain the reasons behind my current effort regarding the GSX support:
 - To extend the z88dk kit to those few targets still lacking an emulation and documentation.   E.g. a "Kyber Minus", with a 512x512 resolution and its own historical DD driver could now run some of the z88dk graphics bits.
 - To get more interest on z88dk from the retrocomputing community (the D.R. GSX is an interesting piece of the IT history), and possibly compensate a bit our weakness regarding the Amstrad stuff  :D
 - To have some fun with the new z88dk features:  the sections helped a lot in creating the proper structures, the CP/M disk creation stuff on appmake is a beauty,  and Dominic did a wonderful work in sorting out the graphics libraries, so I'm trying to mess them up again  🤣  

The GFX APIs.   The way GSX was structured is not my ideal one.. the 32768x32768 px windows was probably chosen to be compatible with the 16bit int parameters from most of the programming language.   It is too small for very high resolutions (plotters, etc) and too big for a video display.   I could not find a way to gather the real rez of the device when in graphics mode (have I missed it?).
In my opinion we have already an interesting base to work on (turtle graphics, raster output, vectors, areas, polygons..), portable versions of many of those functions and a bit of demos to test a new port.

The VDI.    This one is an interesting topic, when I began I did not know there was a text interface too !   I'm not sure though, should we add yet another way to deal with the console ?

feilipu, 2021-03-19:
> To get more interest on z88dk from the retrocomputing community (the D.R. GSX is an interesting piece of the IT history), and possibly compensate a bit our weakness regarding the Amstrad stuff.

From what I read it is really the [Amstrad driver support](https://www.seasip.info/Cpm/gsxint.html) that makes GSX relevant today (apart from historical reasons). There are only a handful of D.R. applications (DR Draw, DR Graph) and not much more. So it would be great to try to expand this catalogue. And get some new applications running.

> I could not find a way to gather the real rez of the device when in graphics mode (have I missed it?).

I think it could be [Function 27](https://www.seasip.info/Cpm/gsx.html#27) - Inquire cell array?
It talks mainly about colour arrays, but it provides an error return if the requested pixel doesn't exist.
So I imagine it could be used to confirm your suspicions.

> In my opinion we have already an interesting base to work on (turtle graphics, raster output, vectors, areas, polygons..), portable versions of many of those functions and a bit of demos to test a new port.

Yes, I agree.

I see the GSX function definitions in [`cpm.h`](https://github.com/z88dk/z88dk/blob/master/include/cpm.h#L200), but that is making me uncertain as to where the API is being drawn. I thought the GSX interface was via the GDOS function `$73`?

The function calls all refer to `gios_xxx`, but the documented GSX application interface definition is via the GDOS implemented in `GSX.SYS`. So this sounds like the function calls should really be named __`gdos_xxx`__. Or, have I missed something?

Anyway, I guess my view is that.

- a new z88dk graphic interface should implement the GSX VDI, but there's not much value in that in and of itself. So, I'll work on getting that working.
- 
 - and, that the GDOS does so little (that it seems forgotten), and that writing directly to the VDI API directly seems like the most useful outcome. I won't try to put the driver into a `PRL` file and then need `GSX.SYS` and `ASSIGN.SYS` to link, but rather the VDI driver can be linked into the application at build time.

zx70, 2021-03-19:
Let me rephrase, english is not my first language and I need to be sure I'm understanding correctly:   are you proposing to make the application able to simply pick the external PRL driver, allocate it alone at the memory bottom together with the appropriate graphics functions, and have a wrapper jumping directly to it rather than via GDOS  ?

If so, I like it.. it is stimulating  :)

I'll experiment a bit more on inquiring the returned functions to pick the exact resolution, that could force me in reworking most of the stuff I've published already ..but it is part of the fun, right ?
If so, I will try to rename the gios_* stuff to whichever name we decide to be appropriate,

So far I'm already happy I was able to get some kind of output, even stuffing the results in a valid disk images and turning around the emulators to boot and mount such images was tricky !
Again.. I really couldn't get up to here without the massive cleanup Dom recently did.

zx70, 2021-03-26:
I'm keeping the gios_* naming convention for the calls.
For what I could understand reading the docs, the structure to be forwarded to the drivers is still the same (well the documentation is not the best ever, so I may be wrong)..  but I see little benefit in renaming the functions in 'BDOS' and I'm avoiding to abuse with the GSX* definitions to avoid confusion.

feilipu, 2021-03-26:
> For what I could understand reading the docs, the structure to be forwarded to the drivers is still the same 

Yes. I believe you're right. The GDOS layer does nothing except read the `ASSIGN.SYS` file to work out to which driver the calls should be mapped. It doesn't do any translation as far as I can tell.

Still ruminating on this issue, so no real code progress. I've a simple use case, but trying to generalise the solution makes the problem bigger but makes the solution perhaps more useful.

I've also the issue #1568, that the RC2014 has multiple different serial interfaces with each subtype, so ideally there would be a way to generalise the way the terminal interface is applied. The GSX solution also supports pure serial devices too (but not in graphics mode obv.) so it could be useful as a guide.

I get the feeling that a generalised newlib terminal instantiation which included graphics orientated terminals would be very useful. Perhaps the driver API could look like the GSX structures and calling mechanisms? Then one driver could be used across multiple subtypes and even targets (potentially), and would be useful also to GSX too?

But, if I scope the solution so broadly then I don't have the ability to write an elegant solution. And, that's why there's no progress.

zx70, 2021-03-26:
To be honest, I do not like the GSX calling mechanisms, even skipping the BDOS trap (not to be intended only in the technical way !).

Trying to evaluate the pros and cons, I see very little opportunities on focusing on the GSX, it was clearly intended to provide vector graphics support, to deal with serial terminals, plotters, printers supporting some sort of GL in a period where companies were investing in graphics add-ons.
In the past I was tempted to import the existing RS232 drivers built for the ancient terminal programs.  It is quite simple, they have a fixed size and position but I couldn't test them, and very few people is interested in that (thus, I had nobody testing it).
GSX is slightly different because of the recurring enthusiasm around it (a lot of people dreaming to work it out and very few succeeding  :D)  and due to the existence of a number of targets/devices we're not supporting already.
I recently discovered that even those targets capable of GSX graphics had non-GSX based programs to work around its limits (memory requirements, weakness in raster graphics, limited coverage of the GSX specs of some drivers, complex and cumbersome structures to deal with..).
This was the case of GraphPlan for the MicroBee or the font editors on the PCW.

This is why I'm keeping the GSX as a nice option rather than a base for developing a standard.
I'd rather try to get closer to existing APIs, like the one existing for CC65, something possibly available for the Arduino or those which were provided with some other old dev environment, like the Borland ones.

Back to the intriguing things we can do, there's also the possibility of using z88dk to build valid drivers for the GSX and provide support for those targets which didn't have it.

zx70, 2021-04-05:
I did a bit more of research, your comment regarding Amstrad is quite confirmed  :)
The rough timeline I was able to recreate:
1981
  - The Ontel company, growing up dramatically, gets hold of a CP/M based computer originally designed in Asia, sold in the US as tge Ontel Amigo.  It's capable of 640x300 pixel graphics.
1982
- Ontel is sold to Visual Technology
- Northstar launches its own graphics capable computer,  the "Advantage".   It is one of those system bridging from CP/M 80 to MSDOS  and CP/M 86 (with an optional add-on).  The 640x240 graphics was enabled by a clever trick, a GMGRADD command appending a driver loader to the customer's executable thus limiting the waste of memory to those only applications needing graphics.
1983
- Digital Research is continuously working to make the passage between 8 to 16 bit systems totally transparent to the users and preserve its market position regardjng the 'Business/Finance' customers.  It launches GSX aiming to provide portable graphics, but the customers are generally sick of the excessive hype (the same reason is bringing to the videogames market crash in the US).  In example, many companies immediately announce GSX support in the 'near future'.  DR uses the GENGRAF trick, similar to GMGRADD but hooked to the BDOS entry for further compatibility with any CP/M system and overlaid driver support (workstation ID, etc..).
- Cortex, a British company starts distributing a computer named C/WP (computer for Word Processing), based on the Ontel Amigo design.   GSX support is provided.   Another company in Brazil sells an Ontel Amigo clone called LatinData Professional.
- Televideo launches the TS803, derived from the TS802 (1982) but featuring 640x240 pixel graphics and GSX supports.
1984
- the Amstrad journey begins.   I read on the web that Cortex and Amigo systems are basically Amstrad CPC, I still haven't cheched it.

feilipu, 2021-07-02:
Just too add to / close off on the the discussion. I'm now intending to write to the [ReGIS](https://en.wikipedia.org/wiki/ReGIS) interface, rather than to GSX. Why is that?

I think that ReGIS is more generalised for my application using a terminal graphical interface, than relying on specific hardware or device drivers. In this way the application can be supported on a variety of platforms connected to a variety of VT125, VT240/VT241, and VT330/VT340 terminals and terminal emulators.

What could be nice though would be to implement a GSX -> ReGIS library in z88dk.
Still thinking about how this could work.

closed 2021-07-16 by feilipu