z88dk/z88dk #1723: (classic) improve Bondwell 2

zx70, 2021-03-20:
While improving stencil_render() in the bw2 I  noticed it's not in working order anymore.
@suborb , could it have to do with the startup enhancements (font, etc?)
I noticed few odd characters being displayed at the bottom of the display when the program is initializing the graphics page,  e.g. something stays visible while running coswave.c
stencil.c in turn now goes crazy (it was very slow but worked well)

suborb, 2021-03-20:
Hmmm, the only things I changed were:

- Changing the cls routine
- Changing the swapgfx code so it was no longer smc

I guess register usage could have changed?

zx70, 2021-03-20:
perfect,  I'll focus on that

zx70, 2021-03-21:
I'm running out of guesses,  I think I have only one left:  could it be that we have something in the init section which depends on a to-be-relocated part in himem?
The odd behaviour resembles some sort of overflow, but moving sp did not help.

suborb, 2021-03-21:
So, it looks like regular plotting works ok.

There is a problem where some code has ended up in the wrong section (in generic_console_cls), but correcting that to just print the character doesn't resolve the problem.

suborb, 2021-03-21:
Ah, right, solved, or at least problem found.

The stencil[256*4] inside examples/graphics/stencil.c is accessed from HIMEM whilst the display is switched in.

So, I think the swapgfxbks need a bit of juggling around.

[image]

zx70, 2021-03-21:
ouch, I was convinced it was something else because it worked !
I'm halfway in getting a byte oriented version of stencil render, but yes it's not good to have that kind of traps.
Thank you for spotting it

zx70, 2021-03-22:
So, there was still something unclear to me.  The PCW port is working fine, but probably that's only a matter of temporary luck.
I think that the problem is elsewhere, the functions setting up the 'stencil' element are swapping the graphics in and out, sharing  "__graphics_end" on exit.
This can occasionally make them 'plot' on the wrong page, I was probably keeping the stencil data in the high part of the memory to avoid this kind of problems, but there could be even three options, including the insane idea of getting its space somewhere in the paged video memory.
Moving it all on himem is simple and tempting but it is a lot of memory (about 800 bytes, but it depends on the screen vertical rez).

suborb, 2021-03-22:
I do still think that the `swapgfxbk` is too broad. The way I've done graphics for the past few years has been to noop that call and do the bankswitching in plot/res/xor/point target specific implementation.

When there's several screen modes to support and colour, this is usually the easiest way to do it: the old z88 graphics library was probably a bad model to follow since you can end up with interrupts disabled for a long time when drawing - hindsight is wonderful though!

zx70, 2021-03-22:
I agree.   I think that the swapgfxbk calls should be excluded at least from the inner stencil subs (I know, it was me adding it!).
I only hope it won't break things somewhere else (e.g. I remember the Spectrum target re-setting IY).

closed 2021-03-22 by zx70

zx70, 2021-03-22:
the much you boosted the pc88 ! 
impressive really,  and colours are a wonderful extra