z88dk/z88dk #1726: (classic) (sam) Errors and issues compiling example programs

dandoore, 2021-03-22:
I went through and compiled all the example code, here are the results for those ones that didn't quite work as they should:

**/examples/**

_othello.c_

Command line: zcc +sam othello.c -lm -v -lndos
or
Command line: zcc +sam othello.c -lm -v -lndos -DGRAPHICS

Keyboard input very, very bouncy
Scroll artifacts on Move: box, also loses play area as it scrolls (might be normal)

_wall.c_

Command line: zcc +sam wall.c -lm -v -lndos -DJOYSTICK -DBANNERS -Dspritesize=8 -DSOUND -DCLOCK -DBORDERS

```
z88dk-z80asm -mz80 -b -d  -o"a.bin" -L.   -I"C:\Users\xbox\AppData\Local\Temp\zcc00002F7CF7183" -D__SCCZ80 -L"d:\z88dk\lib\config\..\..\/lib/clibs" -I"d:\z88dk\lib\config\..\..\/lib" -l"genmath" -lndos -lsam_clib -lndos -l"z80_crt0"     "C:\Users\xbox\AppData\Local\Temp\zcc00002F7CF7184.o" "C:\Users\xbox\AppData\Local\Temp\zcc00002F7CF7182.o"
Error at file 'target/shared/clock.asm' line 13: symbol '_FRAMES' not defined
Error at file 'target/shared/clock.asm' line 14: symbol '_FRAMES' not defined
Errors in source file d:\z88dk\lib\config\..\..\\lib\target\sam\classic\sam_crt0.asm:
Error at file 'target/shared/clock.asm' line 13: symbol '_FRAMES' not defined
Error at file 'target/shared/clock.asm' line 14: symbol '_FRAMES' not defined
 ^ wall failed, press a key.
```

Command line: zcc +sam wall.c -lm -v -lndos -DJOYSTICK -DBANNERS -Dspritesize=8 -DSOUND -DBORDERS

Works fine

**/examples/sam**

_ansitext.c_

Command line: zcc +sam -lm ansitext.c -clib=ansi -lndos 

Has odd coloured bars on the Back Text Colour lines
The 'x' is now inverted ink/paper
Bottom lines of screen seems replicated

Also fails to compile under basic mode:

Command line: zcc +sam -lm -subtype=basic ansitest.c -clib=ansi -lndos

```
Compiling ansitest.c...
Error at file 'd:/z88dk/lib/../lib/target/sam/classic/basic.asm' line 22: symbol 'CRT_KEY_DEL' already defined
Errors in source file d:\z88dk\lib\config\..\..\\lib\target\sam\classic\sam_crt0.asm:
Error at file 'd:/z88dk/lib/../lib/target/sam/classic/basic.asm' line 22: symbol 'CRT_KEY_DEL' already defined
                   ^ ----     defc    CRT_KEY_DEL = 12
```
				   
_dstar.c_

Command line: zcc +sam -lm dstar.c 

Locks up on startup - although the one in **/examples/** compiles and runs fine.

_enigma.c_

Command line: zcc +sam -lm enigma.c 

Long text line wraps around screen incorrectly overprinting previous line but a pixel down, but only on every other line (keep typing to see this)

Command line: zcc +sam -lm enigma.c  -clib=ansi 

Output is fine.

_graygfx.c_

Command line: zcc +sam -lm graygfx.c 

Does not compile, missing library?

_keyget.c_

Command line: zcc +sam -lm keyget.c 

Scrolling cuts off characters tops, bottom lines of screen seems replicated

suborb, 2021-03-22:
Thanks for taking the time to do that - I thought there might be a few bugs handing around!

Thankfully, they're not too major.

The main issues were:

- The screen was set to be 64 characters wide rather than 32 - hence the offset by 1 pixel
- The scroll routine wasn't clearing the bottom row

othello.c: I'm not sure how this is meant to behave @zx70 ?

wall.c: Looks like I need to install a tick ISR by default!

ansitest.c: I think what you're seeing outside of basic mode is expected: just before the cross is printed the program issues a "reset attributes" command. Unfortunately the way it's been bridged to the console driver it's not possible to reset the colours.

I'm not sure what you mean "odd coloured bars"

Duplicated last line is the scrolling issue.

I've done a bit of tweaking and have made -clib=ansi work with the basic subtype again - just a few toggles to switch to get it to behave.

dstar.c: I've deleted this now since there's at least two other versions of dstar that work. I think the issue here was that it was trying to print in mode 1 at address 16384.

graygfx.c: I'm not sure what this TI example was doing there, so that's gone as well now.

I think that's all of those issues covered, and I found a couple of others whilst testing which I've fixed as well.

zx70, 2021-03-23:
othello.c  tries to keep the spirit of the original, teletype oriented, game.
Thus, in text and UDG modes it simply works by dumping text and counting on the target system for a scroll feature.

In graphics, it clears the screen in both text and graphics mode to home the cursor.   The picture is drawn "conveniently" shifted on the right to leave the target system space enough for text, hoping that text and graphics mixing works on the target system.
Sometimes switching to the "ansi" or the simplified 4x6 text option should work.

The keyboard is a simple getchar().   You can ignore the references to getk(), it is used only to escape from the "0 players" option.

Need me to investigate?

dandoore, 2021-03-23:
_Othello_
Scrolling in Othello is fixed but the keyboard input is so bouncy it fails 50% of the time as it seems to be picking up phantom keypresses - when starting you now don't get the choice of who goes first as it just flips past with the computer going first as if 'n' was selected.

_ANSItest_
subtype=basic generates an 'Off Screen' error 
[image]

In default memory model you now get full character coloured bars on the BackTextColour lines
[image]
 (before it was a couple of pixels, possibly due to the scrolling issue that has been fixed)

Also I just spotted that using 'bordercolour' compiles but fails as it's trying to call a missing ZX library?
```
z88dk-z80asm -mz80 -b -d  -o"a.bin" -L.   -I"C:\Users\xbox\AppData\Local\Temp\zcc00003EBCB9E53" -D__SCCZ80 -L"d:\z88dk\lib\config\..\..\/lib/clibs" -I"d:\z88dk\lib\config\..\..\/lib" -l"genmath" -lsam_clib -lndos -l"z80_crt0"     "C:\Users\xbox\AppData\Local\Temp\zcc00003EBCB9E54.o" "C:\Users\xbox\AppData\Local\Temp\zcc00003EBCB9E52.o"
Error at file 'bordercolour.asm' line 17: symbol 'zx_border' not defined
Errors in source file d:\z88dk\lib\config\..\..\\lib\target\sam\classic\sam_crt0.asm:
Error at file 'bordercolour.asm' line 17: symbol 'zx_border' not defined
```
I'm compiling a table of the native 'Sam C' functions against those in Z88DK with a view to working out what maps to what in Z88DK libraries and anything handy that is missing (BLOCKS, CSIZE, FAXPIX, ROLL and SCROLL come to mind) and will try and build something to exercise these.

suborb, 2021-03-23:
Ah right, `sam_get_screenmode()` is bypassing some of the settings for the console driver so things are end up in the wrong place. I tried to do something different with the mode branching, but it's obviously not working in the case. I'll go back to conventional way of doing it that should bypass it.

I think that scrolling bar is expected if the paper attribute is set when the screen is scrolled (though I do know it's broken in mode 1).

I've seen the basic driver one, I think it's trying to print to line 23 which is out of the range of the upper screen, so I need to find the ROM call that can sort that out.

dandoore, 2021-03-23:
> I think that scrolling bar is expected if the paper attribute is set when the screen is scrolled (though I do know it's broken in mode 1).

Not like that with the offset lines - here's a quick demo for how the scrolling should be in each mode:

https://youtu.be/vup5fnZVptw

> I've seen the basic driver one, I think it's trying to print to line 23 which is out of the range of the upper screen, so I need to find the ROM call that can sort that out.

Cool.

suborb, 2021-03-23:
> > I think that scrolling bar is expected if the paper attribute is set when the screen is scrolled (though I do know it's broken in mode 1).
> 
> Not like that with the offset lines - here's a quick demo for how the scrolling should be in each mode:

It depends on *when* the scroll occurs. In this case the scroll is happening when the \n is printed.

In that program we've got:

```
        printf("%c[%umBack text color %u.\n", 27, x, x);
```

So the scroll happens whilst we're on the colour that's defined in that line. So the blank line is filled in with those attributes.

If we simulate what happens in basic:

```
        printf("%c[%um\nBack text color %u.", 27, x, x);
```

The the attributes are changed, screen scrolled and we start printing in that blank line at the bottom.

suborb, 2021-03-23:
I've tracked down the odd keyboard behaviour to reading the 9th row of the keyboard - sometimes it thinks the left cursor is pressed.

suborb, 2021-03-23:
It turns out I wasn't reading the last row of the keyboard, but some other random port!

Hopefully that's all of the issues you've found now sorted.

dandoore, 2021-03-24:
> It depends on _when_ the scroll occurs. In this case the scroll is happening when the \n is printed.

Ah, gotcha - that makes sense, moving the \n gives the result I was expecting

dandoore, 2021-03-24:
> Hopefully that's all of the issues you've found now sorted - apart from the tick ISR.

Awesome, all gnarly issues fixed but some oddities in _ansitest.c_ 

Command line: zcc +sam  ansitest.c -lm -v -clib=ansi -lndos -create-app 
Added: 
```
	int mode=<as per mode);
	console_ioctl(IOCTL_GENCON_SET_MODE,&mode);
```
Mode 1: 
[image]
(@suborb you mentoned this was broken so this might be known)
Background is white rather than black (i.e. unchanged) so scroll paper colour is mismatched
Should show the text back colour lines?
Should it default to pseudo 64 col font? 

Mode 2:
[image]
Background is white rather than black (i.e. unchanged)
Mode 3:
[image]
Odd step at very right hand side of the text back colour lines
Mode 4:
[image]
Odd step at very right hand side of the text back colour lines

suborb, 2021-03-24:
Mode 3/4 scrolling: Looks like that's an off by one - it took me long enough to spot it in your screenshots so I'm not surprised I missed it! :)

Mode 3 palette: I'm deliberately not changing the palette and not touching the CLUT bits in HMPR. Accessing them should be pretty easy for user code though using SFRs: `IO_HMPR = IO_HMPR | 0x20` for example.

Mode 2/1: Looks like I'm not changing the default colours from the +zx black/white. I do want to rewrite the mode1 scrolling since it also affects +ts2068 and ROM based +zx.

dandoore, 2021-03-25:
Cool, ansitest is working:)

-subtype=basic is failing though:

Command line: zcc +sam -subtype=basic ansitest.c -lm -v -clib=ansi -lndos-create-app 

```
PROCESSING CRT
z88dk-z80asm -I"C:\Users\xbox\AppData\Local\Temp\zcc00001A485CC23" -D__SCCZ80  -s  -mz80 -I"d:\z88dk\lib\config\..\..\/lib" -I"d:\z88dk\lib\config\..\..\\lib\target\sam\classic"  "C:\Users\xbox\AppData\Local\Temp\zcc00001A485CC24.asm"

z88dk-z80asm -mz80 -b -d  -o"ansitest" -m -s -L.   -I"C:\Users\xbox\AppData\Local\Temp\zcc00001A485CC23" -D__SCCZ80 -L"d:\z88dk\lib\config\..\..\/lib/clibs" -I"d:\z88dk\lib\config\..\..\/lib" -l"genmath" -lndos -lsam_clib -lndos -l"z80_crt0"     "C:\Users\xbox\AppData\Local\Temp\zcc00001A485CC24.o" "C:\Users\xbox\AppData\Local\Temp\zcc00001A485CC22.o"
z88dk-appmake +mgt --exec --default-dosfile=d:\z88dk\lib\config\..\..\\lib\target\sam\classic\samdos2.bin  -b "ansitest" -c "C:\Users\xbox\AppData\Local\Temp\zcc00001A485CC24"
Building application code failed
```

closed 2021-03-25 by suborb

dandoore, 2021-03-26:
-subtype=basic working.

Command line: zcc +sam -subtype=basic ansitest.c -lm -v -clib=ansi -lndos-create-app

[image]