z88dk/z88dk #1727: (classic) (sam) -create-app ignoring manually specified dos-file

dandoore, 2021-03-23:
Command line: zcc +sam miniman.c -lm -v -lndos -o miniman -create-app -Cz--dosfile=B-DOS16

Adds --dosfile to default (samdos2.bin) instead of B-DOS16 in this case, seems to specify it twice

```
PROCESSING CRT
z88dk-z80asm -I"C:\Users\xbox\AppData\Local\Temp\zcc00004430D05C3" -D__SCCZ80  -s  -mz80 -I"d:\z88dk\lib\config\..\..\/lib" -I"d:\z88dk\lib\config\..\..\\lib\target\sam\classic"  "C:\Users\xbox\AppData\Local\Temp\zcc00004430D05C4.asm"

z88dk-z80asm -mz80 -b -d  -o"microman" -m -s -L.   -I"C:\Users\xbox\AppData\Local\Temp\zcc00004430D05C3" -D__SCCZ80 -L"d:\z88dk\lib\config\..\..\/lib/clibs" -I"d:\z88dk\lib\config\..\..\/lib" -l"genmath" -lndos -lsam_clib -lndos -l"z80_crt0"     "C:\Users\xbox\AppData\Local\Temp\zcc00004430D05C4.o" "C:\Users\xbox\AppData\Local\Temp\zcc00004430D05C2.o"
z88dk-appmake --dosfile=B-DOS16 +mgt --exec --dosfile=d:\z88dk\lib\config\..\..\\lib\target\sam\classic\samdos2.bin  -b "microman" -c "C:\Users\xbox\AppData\Local\Temp\zcc00004430D05C4"
```

suborb, 2021-03-23:
Specifying it twice would be okay if it were in the right order (so that the newer overwrites the older).

Sadly it's not so I'll have to make the default a new option that user doesn't use.

closed 2021-03-23 by suborb

dandoore, 2021-03-24:
Fixed, thank you!

commit 431908f9 (2021-03-23): Fix #1727: Rename options to allow --dosfile override by the user