z88dk/z88dk #1733: (sam) (classic) Errors and issues in compiling other examples

dandoore, 2021-03-25:
*/examples/sound/etracker/*

_main.c_

Command line: zcc +sam main.c axel.asm -lm -v -lndos -create-app 

Works but slow (running at about 1/3 speed?)

*/examples/clisp/*

_clisp.c_

Command line: zcc +sam clisp.c -lm -v -lndos -create-app 
```
z88dk-z80asm -mz80 -b -d  -o"clisp" -m -s -L.   -I"C:\Users\xbox\AppData\Local\Temp\zcc00001D245C7E3" -D__SCCZ80 -L"d:\z88dk\lib\config\..\..\/lib/clibs" -I"d:\z88dk\lib\config\..\..\/lib" -l"genmath" -lndos -lsam_clib -lndos -l"z80_crt0"     "C:\Users\xbox\AppData\Local\Temp\zcc00001D245C7E4.o" "C:\Users\xbox\AppData\Local\Temp\zcc00001D245C7E2.o"
Error at file 'clisp.c::init::0::8' line 608: symbol 'time' not defined
Errors in source file d:\z88dk\lib\config\..\..\\lib\target\sam\classic\sam_crt0.asm:
Error at file 'clisp.c::init::0::8' line 608: symbol 'time' not defined
```

*/examples/console/*

_adv_a.c_

Command line: zcc +sam adv_a.c -lm -v -lndos -create-app -DPICS -DCOMPUTE_SCORE

No Graphics (black on black?) since 24th nightly build, previously graphics not displaying correctly compared to ZX Version same issue as */examples/graphics/marilyn.c* ?

*/examples/console/*

_vstone.c_

Command line: zcc +sam vstone.c -lm -v -lndos -create-app -clib=ansi -pragma-define:ansicolumns=80

With or without
```
 	int mode=3;
	console_ioctl(IOCTL_GENCON_SET_MODE,&mode);
```

Needs to display 80 columns (CSIZE 8x6) in MODE 3 to work but not supported?

_world.c_

Command line: zcc +sam world.c -lm -v -lndos -create-app
Command line: zcc +sam world.c -lm -v -clib=ansi -lndos -create-app

No output.

suborb, 2021-03-25:
Yeah, I totally mucked up the graphics last night - my recompilation didn't pick up the optimisation I made so I assumed it work. Missing "and a" from the core graphics routines so it thinks every pixel is out of bounds :(

I thought etracker was slow, but didn't have a reference - can you let me know a good delay time - I'm guessing something like 20millis should be correct to match the interrupt. Unfortunately for me it sounds awful at 20millis, but I think that's because my laptop is practically a retro machine in itself and only runs emulators at a variable speed between 50-75% speed most of the time.

Regarding the graphics in Marilyn and presumably adv_a, the draw_profile does an odd compile time scaling based on the horizontal resolution of the screen. Since we configure the max dimension to match that of mode 3 the scaling takes things off screen in a weird way.  I think this also affects other targets that have variable width graphic modes where one of the modes is >= 320 pixels.

I suspect that when you switch to mode 3 everything appears correctly. @zx70 is it possible to call getmaxx() to do the scaling at runtime  inside draw_profile?

World.c is a great one (I'm laughing to be honest), it's got this line:

        printf("Hello world! %c\n",12);

The %c was originally a %d to do a little bit of maths for 8080 machines and (oddly) trim down the binary size. Unfortunately printing character 12 directly clears the screen! So of course nothing can be seen!

A few of the others are because there's no system clock for the SAM moment, though since I think there's an RTC module which I suspect would be better to hook up to?

80 column mode would be covered by #1707 - it probably makes sense to do it for the SAM only in mode 3.

zx70, 2021-03-25:
Oops
. I wanted the screen to be cleared,  but the intention was to have it before printing

dandoore, 2021-03-25:
> I thought etracker was slow, but didn't have a reference - can you let me know a good delay time - I'm guessing something like 20millis should be correct to match the interrupt. Unfortunately for me [...]

Here is the song running: https://www.youtube.com/watch?v=1e05RqzJQPY&t=90s
 
> The %c was originally a %d to do a little bit of maths for 8080 machines and (oddly) trim down the binary size. Unfortunately printing character 12 directly clears the screen! So of course nothing can be seen!

:)
 
> A few of the others are because there's no system clock for the SAM moment, though since I think there's an RTC module which I suspect would be better to hook up to?

Yes, there is a standard RTC's that is used - reffered to as the SamBus/DALLAS clocks - implementation details in clock.cpp https://github.com/simonowen/simcoupe/tree/master/Base if that helps.
 
> 80 column mode would be covered by #1707 - it probably makes sense to do it for the SAM only in mode 3.

Cool.

zx70, 2021-03-25:
draw_profile() should be better now

dandoore, 2021-03-25:
I've got a fix for etracker, the calling routine needs to be called once a frame so extarcting the `pause` function from the SamC libraries works a charm like this:
```
void pause(int c)
{
	#asm
	ei
	ld a,h
	or l
	ret z
	halt
	dec hl
	ld a,0x7f
	in a,(254)
	rrca
	ret nc
	jr $-11
	#endasm
}

<SNIP>

   // Just loop
   while  ( 1 ) {
#ifdef NO_INTERRUPT
       saa_etracker_play();
       pause(1);
#endif
```
pause is a handy fuction in other stuff too - can it be added in somewhere?

suborb, 2021-03-25:
The other option is to get it to run on the interrupt, this seems to run okay with Mame but is deathly silent with Simcoupe so something is missing there. I'm a little baffled:

[image]

[image]

Why is it reading 0xff if the debugger thinks there's a frame interrupt waiting?

I've just tried it out your version and it still sounds terrible on my machine with the % indicator bouncing around all over the place. You can do the same thing with:

```
#include <intrinsic.h>

intrinsic_halt()
```

dandoore, 2021-03-25:
> 
> 
> The other option is to get it to run on the interrupt, this seems to run okay with Mame but is deathly silent with Simcoupe so something is missing there. I'm a little baffled:
> 
> [image]
> 
> [image]
> 
> Why is it reading 0xff if the debugger thinks there's a frame interrupt waiting?

This is @simonowen, @stefandrissen and @frodete 's area - I know literally nothing about assembler :(

> I've just tried it out your version and it still sounds terrible on my machine with the % indicator bouncing around all over the place. You can do the same thing with:
> 
> ```
> #include <intrinsic.h>
> 
> intrinsic_halt()
> ```

Ah, that's neater - Under SimCoupé it's spot on in terms of timing.

suborb, 2021-03-25:
Oh wow...I've not tracked down the right bit of the emulator source, but if I read the status port before I fiddle with HMPR it works ok.

And it sounds better if I hide the emulator window :) I'll update the example so it's interrupt driven.

simonowen, 2021-03-25:
> [image]

In the image above the frame interrupt (F) is indeed active in the Status register, but the Events list in the bottom right also shows the end-of-interrupt even is due in 2 CPU cycles. It'll become inactive even before the next opcode byte is read, so the port read will show no interrupts are active.

> [image]

In the image above you can see the pink dash where the F has disappeared, so the interrupt is no longer active.

I'm not sure I understand about hiding the window improving the sound. Is that in SimCoupe? Does it report that it's running at 100% speed? If you've built from source the Debug version might run a chunk slower, but I'd be surprised if it can't run at full speed on any system from the last 10 years  :)

suborb, 2021-03-25:
Thanks @simonowen, totally didn't spot nor understand the events section, but yes, reading the register a little bit earlier fixed it. Obviously there's an omission in the Mame which doesn't emulate that behaviour.

I'm using a  1.1 binary that I downloaded a few years ago, the % result drops down to 70 at times which is when the sound becomes out of sync. 

To be honest, although it's an old machine (2012) it probably just needs a reinstall and cleanup since I have performance issues with most emulators.

dandoore, 2021-03-26:
**/examples/console/**

_adv_a.c_

Fixed \o/

**/examples/sound/etracker/**

_main.c_

Command line: zcc +sam main.c axel.asm -lm -v -lndos -create-app

Working under interrupts but putting a basic printf("*") in the main loop results in corruption of the characters being printed:

[image]

I added the interrupt-driven routing to one of my other test program that is printing lines on the screen with draw() and that is unaffected though.

suborb, 2021-03-26:
Looks like etracker uses af' as well so that will need to be saved on the interrupt.

zx70, 2021-03-26:
I got back to draw_profile() to make its memory allocation dynamic using getmaxx() and getmaxy().  This will save lots stack space on the GSX target but it works also dynamically in those targets supporting video mode changes at runtime.

suborb, 2021-03-26:
@zx70 that change doesn't seem to work for adv_a.c for the SAM, I'm guessing the stack is being adjusted incorrectly and we end up in space.

zx70, 2021-03-26:
Probably I need to extend the test to more targets, thank you for alerting.

zx70, 2021-03-26:
odd, the resolution is rather static on the SAM, it must be a stupid bug

zx70, 2021-03-26:
Weekend time, if it's annoying to have it hanging around for a couple of days, feel free to revert.

dandoore, 2021-03-27:
Etracker test is spot on:
[image]

I presume the issue with _clisp.c_ is just doen to support for RTC/time or is there more to it?

zx70, 2021-03-27:
You can try building it by setting the -DNOTIMER flag.
There's almost no loss in excluding it.
You'd be rather interested in adding -DGRAPHICS.

zx70, 2021-03-27:
@suborb , I think I understood what went wrong with draw_profile.
Stencil_render works on a single vector having the subsequent left and right sides of the picture, of 1 or 2 bytes depending on the horizontal rez.   Setting dynamically a different segment length (and reduce the stack space) in stencil_render  without instructing draw_profile accordingly will obviously mess up the pictures and probably crash.

It is then going to be an extra option in gfx/common, which I'll surely use for GSX ..ATM it is allocating 768x4=3Kbytes on stack, when (the bottom of) the memory is already bearing the GSX layer itself.

I'm sorry I didn't test it right,  I think I was trapped by an unexpected behavior of the ts2068, staying in standard zx mode.

dandoore, 2021-03-27:
> You can try building it by setting the -DNOTIMER flag.
> There's almost no loss in excluding it.
Cool that works:
[image]

> You'd be rather interested in adding -DGRAPHICS.

I shall give the turtle a run out :)

zx70, 2021-03-27:
nope, draw_profile will stay like this,  it'll be eventually quoted in the GSX thread

closed 2021-03-27 by zx70

zx70, 2021-03-27:
I see you chose to build CLISP in SpecLisp mode:  be aware that it is not the way you should program in LISP nowadays,  it was an obscure, now totally forgotten dialect  :)

dandoore, 2021-03-31:
Frames issue added to enhancement request  #1741