z88dk/z88dk #1737: z80asm: ORG directive is not honoured by ASMPC

apoloval, 2021-03-30:
Given the following code:

```asm
        ORG     $1000

hello:
        DEFM    "Hello World!", 0

        DEFS    $1100-ASMPC, 0
```

One would expect to start assembly in `$1000`, have the `Hello World!\0` string defined there, and then padding with zeroes until `$1100`. 

However, the `ASMPC` is not taking into account the ORG directive. The assembler is assuming everything starts at 0. So instead of padding `$100 - len(hello)` bytes, it is placing `$1100 - len(hello)`. 

In terms of listing, this is what is expected:

```
1     0000                      ORG     $1000
2     0000
3     0000  48 65 6C 6C 6F 20 57 6F 72 6C 64 21 00
                                DEFM    "Hello World!", 0
4     000D
5     000D  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
      002D  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
      004D  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
      006D  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
      008D  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
      00AD  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
      00CD  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
      00ED  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
                                DEFS    $1100-ASMPC, 0
6     00EE
7     00EE
```

But this is what we obtain:
```
1     0000                      ORG     $1000
2     0000
3     0000  48 65 6C 6C 6F 20 57 6F 72 6C 64 21 00
                                DEFM    "Hello World!", 0
4     000D
5     000D  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
      002D  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
      004D  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
      006D  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
      008D  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
      00AD  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
      00CD  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
      00ED  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
      010D  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
      012D  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
      014D  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
      016D  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
      018D  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
      01AD  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
      01CD  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
      01ED  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
      020D  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
      022D  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
      024D  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
      026D  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
      028D  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
      02AD  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
      02CD  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
      02ED  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
      030D  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
      032D  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
      034D  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[... 88 lines cut ...]
      0E6D  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
      0E8D  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
      0EAD  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
      0ECD  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
      0EED  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
      0F0D  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
      0F2D  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
      0F4D  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
      0F6D  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
      0F8D  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
      0FAD  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
      0FCD  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
      0FED  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
      100D  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
      102D  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
      104D  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
      106D  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
      108D  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
      10AD  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
      10CD  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
      10ED  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
                                DEFS    $1100-ASMPC, 0
6     1100
7     1100
```

suborb, 2021-03-31:
That does seem a bit unexpected. I'm not sure of the implications, but this works as you'd expect:

```
        org     $1000
        phase   $1000
        defm    "Hello"

        defs    $1100-ASMPC
        defm    "World"
loop:
        jp      loop
```

Hopefully @pauloscustodio can explain...

apoloval, 2021-03-31:
That's the worst part: the implications. 

As far as I see, the ORG and the ASMPC are totally separated for a section. This reveals another odd thing: the listing file shows the ASMPC starting at 0. The numbers shown there are not relative to ORG. What makes the listing file very difficult to read, as it does not reflect the actual addresses where code is expected to be assembled when an explicit origin is indicated. 

I know there is a reason for this: relocation. Assembled code during pass 1 contains addresses relative to ASMPC. They will be relocated during pass 2 with actual values that take into account the origin. But `DEFS` will determine its size during pass 1. That means the code above is broken. 

IMO, it could be worthy trying to force the `ORG` directive to call `set_PC()` with that origin value. This would make the assembler to set the current position to the origin address without affecting the generated output. As it would do with the `PHASE` directive shown above. This will solve both problems. `DEFS` will work as expected, and listing file will show the addresses starting at the indicated origin. 

I just wonder if the relocation process could be affected by this. Perhaps the unresolved expressions in the object file depend on ASMPC to have an offset relative to 0. But I don't think so from what I see [in the change that added support to PHASE/DEPHASE](https://github.com/z88dk/z88dk/commit/d75a9107e7628d08ccb8ce2cc0a128730dd132ce).

apoloval, 2021-03-31:
I just tried to apply the following change to `set_origin_directive()`:

```c
/* define a new origin, called by the ORG directive
*  if origin is -1, the section is split creating a new binary file */
void set_origin_directive(int origin)
{
	if (CURRENTSECTION->origin_found)
		error_org_redefined();
	else
	{
		CURRENTSECTION->origin_found = true;
		if (origin == -1) /* signal split section binary file */
			CURRENTSECTION->section_split = true;
		else if (origin >= 0)
		{
			if (CURRENTSECTION->origin_opts && CURRENTSECTION->origin >= 0)
				; /* ignore ORG, as -r from command line overrides */
			else
			{
				CURRENTSECTION->origin = origin;
				CURRENTSECTION->asmpc = origin;  // This line was added
			}
		}
		else
			error_int_range(origin);
	}
}
```

Now the following ASM code:
```asm
        ORG	$0100

        DEFM    "Hello ", 0

        DEFS    $0120-$, 0
```

Generates this listing:

```
1     0000                      ORG	$0100
2     0100              
3     0100  48 65 6C 6C 6F 20 00 
                                DEFM    "Hello ", 0
4     0107              
5     0107  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
                                DEFS    $0120-$, 0
6     0120              
```

Which is exactly what one would expect. However, I still have to test how relocation would perform.

apoloval, 2021-03-31:
Buuuut, it was not enough. If we have the following sources:

```asm
        PUBLIC  hello
        PUBLIC  world

        ORG	$0100

hello:  DEFM    "Hello ", 0

        DEFS    $0120-$, 0

world:  DEFM    "World!", 0
```
...and...
```asm
        EXTERN  hello
        EXTERN  world

        ld      hl, hello
        ld      de, world
```
.. when they are linked, the following addresses are assumed in the map file and the binary output:

```
hello                           = $0200 ; addr, public, , test, , test.asm:6
world                           = $0220 ; addr, public, , test, , test.asm:10
__head                          = $0100 ; const, public, def, , ,
__tail                          = $012D ; const, public, def, , ,
__size                          = $002D ; const, public, def, , ,
```

It looks like the origin is added twice.

apoloval, 2021-03-31:
I think it comes from `sections_alloc_addr()`:

```
void sections_alloc_addr(void)
{
	Section *section, *next_section;
	SectionHashElem *iter;
	int addr;

	init_module();

	/* allocate addr in sequence */
	addr = 0;
	for (section = get_first_section(&iter); section != NULL; section = next_section)
	{
		if (section->origin >= 0) /* break in address space */
			addr = section->origin;

		section->addr = addr;
```

This is invoked in the link process. And the address is reseted to the origin for each section. Somewhere else the relocatable values that already have the origin in their offset are added to that `section->addr`. And that's why it is applied twice. 

It looks more difficult than I thought.

apoloval, 2021-03-31:
Confirmed: if I remove the following block from `sections_alloc_addr()`:

```c
if (section->origin >= 0) /* break in address space */
			addr = section->origin;
```

The addresses are generated as expected. 
```
hello                           = $0100 ; addr, public, , test, , test.asm:6
world                           = $0120 ; addr, public, , test, , test.asm:10
__head                          = $0000 ; const, public, def, , ,
__tail                          = $002D ; const, public, def, , ,
__size                          = $002D ; const, public, def, , ,

```

Considering I am not familiarized with this codebase, I'd say I am like a monkey with a typewriter at this point. But perhaps this investigation is useful to figure out if we can update `ASMPC` when origin is set.

feilipu, 2021-04-01:
For consideration, you could try with the `ALIGN` directive, rather than calling on the `ASMPC` at a low level.

`ALIGN` can work either as an alignment to an absolute address, or align to a relative address. I'm [making use of](https://github.com/RC2014Z80/RC2014/blob/master/ROMs/CPM-IDE/acia/cpm22bios.asm) `PHASE`, `DEPHASE`, and `ALIGN` to specify a number of relocatable code blocks for CP/M and to specify address and page aligned BSS storage.

pauloscustodio, 2021-04-12:
ASMPC is a relocatable assembler; this means there are the normal 2 phases in the assembler, but these generate a relocatable object file that needs to be linked to get the actual addresses. ORG is a linker command, telling where the section should start in real memory. All object-modules contain code assembled with an origin of zero.

So, during assembly, ASMPC always starts at zero for each section in each module; during linking, the base address of the section (the ORG value) plus the size of any other modules linked in the same section before the current one are added to the module addresses. You need to take this into account when using DEFS based on ASMPC or ALIGN.

The list file is an output of the assembler, therefore has the zero-based addresses.

The map file is an output of the linker, therefore has the real addresses.

The PHASE directive is intended for code that will run at a fixed address, but is assembled at a different one; the code within a PHASE block is not relocatable.

Hope this clarifies the ORG issue.

closed 2021-04-12 by pauloscustodio

apoloval, 2021-04-13:
Many thanks for the explanation! I think the key point is understanding that `ORG` is a linker directive. And `ASMPC` will never be relative to it. 

I've found this problem while assembling the source code of the MSX System (BIOS+Basic). Which uses many `DEFS` directives to fill the remaining space until a given absolute address. That code expects a non-relocatable assembler indeed. If we want to port it to z80asm, the `DEFS` should be replaced by `ALIGN` directives. 

Again, many thanks for your time and your maintenance to this fabulous z88dk!