z88dk/z88dk #1741: (sam) (classic) Enhancement - Support for SAM system clock

dandoore, 2021-03-31:
Support for SamBus/DALLAS Real Time Clocks, both exist in real hardware and as options in SimCoupé to support the FRAMES commands etc.

Implementation details in clock.cpp https://github.com/simonowen/simcoupe/tree/master/Base if that helps.

Also this from Simon Cooke's CleanDOS routine
[CleanDOS.txt](https://github.com/z88dk/z88dk/files/6204936/CleanDOS.txt)  for detecting:

;Check for presence of realtime clock.

check.clok:
               LD   BC,&B0EF       ;BC = Year MSB of realtime clock
               LD   DE,&0F00

               IN   A,(C)          ;read in the Year MSB -- we can pretty much
               PUSH AF             ;guarantee that this won't change while
                                   ;testing for the clock's presence.
clk.llp:
               OUT  (C),D          ;output 15 different values to the clock
               IN   A,(C)          ;and ensure that they can be read back
               AND  15             ;strip unused bits
               CP   D
               JR   NZ,no.clk

               DEC  D
               JR   NZ,clk.llp

               LD   E,1            ;set E to 1 if clock is present
no.clk:
               POP  AF
               OUT  (C),A          ;restore YEAR data.

               LD   A,E
               OR   A                ;finished init messages
               JR   Z,noclock

               LD   HL,found       ;put the CLOCK FOUND message into the text
               LD   DE,nsambus
               LD   BC,5
               LDIR
noclock:

suborb, 2021-04-25:
Support for the SAMbus clock has now been added. So time(), mktime(), ctime() should now be available and working:

[image]

dandoore, 2021-04-26:
Tested this with the example that didn't work last time under #1733 

/examples/clisp/

clisp.c

Command line: zcc +sam clisp.c -lm -v -lndos -create-app

`z88dk-z80asm -mz80 -b -d  -o"clisp" -m -s -L.   -I"C:\Users\xbox\AppData\Local\Temp\zcc0000196C93443" -D__SCCZ80 -L"d:\z88dk\lib\config\..\..\/lib/clibs" -I"d:\z88dk\lib\config\..\..\/lib" -l"genmath" -lndos -lsam_clib -lndos -l"z80_crt0"     "C:\Users\xbox\AppData\Local\Temp\zcc0000196C93444.o" "C:\Users\xbox\AppData\Local\Temp\zcc0000196C93442.o"
Error at file 'target/sam/time/target_read_structtm.asm' line 9: symbol '_time_from_structtm' not defined
Errors in source file d:\z88dk\lib\config\..\..\\lib\target\sam\classic\sam_crt0.asm:
Error at file 'target/sam/time/target_read_structtm.asm' line 9: symbol '_time_from_structtm' not defined
 ^ clisp failed, press a key.`

Also from #1732

/examples/graphics/

clock.c

Command line: zcc +sam clock.c -lm -v -lndos -create-app -llib3d -DDETAILED

`PROCESSING CRT
z88dk-z80asm -I"C:\Users\xbox\AppData\Local\Temp\zcc00001D90987F3" -D__SCCZ80  -s  -mz80 -I"d:\z88dk\lib\config\..\..\/lib" -I"d:\z88dk\lib\config\..\..\\lib\target\sam\classic"  "C:\Users\xbox\AppData\Local\Temp\zcc00001D90987F4.asm"

z88dk-z80asm -mz80 -b -d  -o"clock" -m -s -L.   -I"C:\Users\xbox\AppData\Local\Temp\zcc00001D90987F3" -D__SCCZ80 -L"d:\z88dk\lib\config\..\..\/lib/clibs" -I"d:\z88dk\lib\config\..\..\/lib" -l"genmath" -lndos -llib3d -lsam_clib -lndos -l"z80_crt0"     "C:\Users\xbox\AppData\Local\Temp\zcc00001D90987F4.o" "C:\Users\xbox\AppData\Local\Temp\zcc00001D90987F2.o"
Error at file 'target/shared/clock.asm' line 13: symbol '_FRAMES' not defined
Error at file 'target/shared/clock.asm' line 14: symbol '_FRAMES' not defined
Errors in source file d:\z88dk\lib\config\..\..\\lib\target\sam\classic\sam_crt0.asm:
Error at file 'target/shared/clock.asm' line 13: symbol '_FRAMES' not defined
Error at file 'target/shared/clock.asm' line 14: symbol '_FRAMES' not defined`

Although I suspect FRAMES is the subsecond tick?

suborb, 2021-04-26:
That was dumb of me, I had it working and then made a small typo whilst getting something else working.

Anyway, I've now blown a few bytes and added a sub second tick so _FRAMEs is available.

As an added bonus if you compile clock.c with -DHAVE_TIME it now seeds the hours/minutes from system clock.

dandoore, 2021-04-27:
Awesome stuff!

clisp now compiles fine and clock looks great:

[image]

Switching to the DALLAS clock type in SimCoupé doesn't return a value (as the clock type is slightly different) - is this easy to add in as most of the new clock devices being made are DALLAS?

suborb, 2021-04-30:
Hopefully I've got it right....

dandoore, 2021-04-30:
Cool, I'll check it out on the nightly build

dandoore, 2021-05-01:
Working but oddly when you compile and launch SimCoupé with clock.c executable immediatly it doesn't read the clock but resetting the emulation with F12 and then rebooting the disk image with F9 or starting the emulation first and then booting the disk reads fine for both SamBus and DALLAS.

Not sure if this is a timing thing with the detection but it didn't happen on just the SamBus implementation, or it's a bug in SimCoupe

suborb, 2021-05-02:
The detection is pretty much the same as the detection in B-DOS 15a so there's nothing unusual with it.

However, given that the feature actually works, I'm going to close this issue.

closed 2021-05-02 by suborb

dandoore, 2021-05-02:
Fair enough :)