z88dk/z88dk #1743: Using AMALLOC in multi file build.

jayacotton, 2021-04-02:
I am attempting to use calloc (thus AMALLOC) in a multi file build.
I have included an 'inc' file on every compile and link line in my Makefile.
The inc file has 
#pragma output REGISTER_SP = 0xc000
#pragma output USING_amalloc

In the target program main, I calloc and print the pointer.   In a downstream function I 
calloc and print the pointer.  Its a zero. 

The full repo is at https://github.com/jayacotton/inettools-z80

suborb, 2021-04-02:
I've just tried this out on a clean project and it works, so I'll need to dig further into your project - which sub-build is it? The obvious question is "is there enough heap available?"

Makefile
```
test.bin: main.o func.o
        zcc +test -o $@ $^ -pragma-include:zpragma.inc

%.o: %.c
        zcc +test -o $@ -c $^

clean:
        $(RM) test.* *.o
```

main.c:
```
#include <stdio.h>
#include <stdlib.h>

extern void func2();

int main() {
        void *ptr;

        ptr = calloc(1,100);
        printf("Main %p\n",ptr);

        func2();
}
```

func2.c:
```
#include <stdio.h>
#include <stdlib.h>

void func2() {
        void *ptr;

        ptr = calloc(1,100);
        printf("Func2 %p\n",ptr);
}
```

```
#pragma define USING_amalloc
```

```
% make
zcc +test -o main.o -c main.c
zcc +test -o func.o -c func.c
zcc +test -o test.bin main.o func.o -pragma-include:zpragma.inc
% z88dk-ticks test.bin
Main c1be
Func2 c158

Ticks: 87538
```

jayacotton, 2021-04-02:
git clone

make ping

ping server

I may need to hack it a bit, since it checks for h/w but, lets try it as is.

suborb, 2021-04-02:
If you use this code:

```
  {
        int max;
        int total;

        mallinfo(&total,&max);

        printf("total %d max %d\n",total,max);
  }
```

Then we can get diagnostics on the heap. I've just commented out the Ethernet initialisation code in main.c and get this output:

```
zxcc  PING.COM
b727
total 6569 max 6569
heap 9d86
HostRoot b727, b1 b6d5, b2 b683
```

Which I think are the log lines you added when you were having trouble - the allocation in main.c and the allocation in libhost.c? The total/max is my logging from mallinfo.

jayacotton, 2021-04-02:
Yes that is the correct assumption.  I will try the exact same thing on my machine, and see what I get.

jayacotton, 2021-04-02:
Well....  
C>k:ping server
total 7820 max 7820
heap 9702
HostRoot 0, b1 0, b2 0
libhost.c 50  

I will try this without cp/net running.

suborb, 2021-04-02:
Just to check, this is the diff of what I'm testing:

```
-- a/ping.c
+++ b/ping.c
@@ -112,6 +112,16 @@ free(t);
          skip_dns = 1;
        }
     }
+  {
+       int max;
+       int total;
+
+       mallinfo(&total,&max);
+
+       printf("total %d max %d\n",total,max);
+
+  }
+#if 0
   if (Ethernet_begin (mac) == 0)
     {
       if (Ethernet_hardwareStatus () == EthernetNoHardware)
@@ -121,6 +131,7 @@ free(t);
     }
   Ethernet_localIP (gWIZNETINFO.ip);
   Ethernet_localDNS (gWIZNETINFO.dns);
+#endif
   run_user_applications = 0;
   if (skip_dns == 0)
```

I don't suppose anything in the commented out bit could end up overwriting the area between the program and stack?

jayacotton, 2021-04-02:
I am on the same track.  Looking for possible heap overwrite.

jayacotton, 2021-04-02:
Does the heap code  honor the stack setting ?  See zpragma.inc.
I ask because it seems the changing the stack value has not affect on the
heap size as seen by mallinfo.

Going to try to set up the heap to use 1/2 of the available space.

suborb, 2021-04-02:
It makes me think that mallinfo might be a bit buggy. The amalloc code definitely uses the stack to calculate how much heap to give though.

jayacotton, 2021-04-02:
I moved the stack down, and the free space got smaller, so that is as expected.

C>ping nasa
total 5060 max 5060
total 4884 max 4884
heap 9562
HostRoot a91e, b1 a8cc, b2 a87a
libhost.c 58  
libhost.c 61  
libhost.c 66  
libhost.c 73  
libhost.c 77  
libhost.c 81  
total 4864 max 4864
heap 9562
HostRoot a91e, b1 a8cc, b2 a87a
192.168.0.120   SERVER  LINUX-HOST
libhost.c 73  
libhost.c 77  
libhost.c 81  
total 4820 max 4820
heap 9562
HostRoot a91e, b1 a8cc, b2 a87a
209.237.226.90  WWW.OPENSOURCE.ORG      OPENSOURCE
libhost.c 73  
libhost.c 77  
libhost.c 81  
total 4788 max 4788
heap 9562
HostRoot a91e, b1 a8cc, b2 a87a
13.227.75.41    WWW.NASA.GOV    NASA
PING NASA (13.227.75.41) 

I think this one is solved.  Just need to figure out why I am stomping on ccp or ndos.

suborb, 2021-04-02:
I've not analysed too much, but, is DHCP_BUFFER correctly sized? It looks like if your server gives you more than 20 or so bytes of options you'll overflow it?

jayacotton, 2021-04-03:
I did cut it down during the 'great trimming exercise' last year.  Have not visited that code since.  (until now).
I find that if I don't call dhcp_run, then the code seem better behaved, so I think you may have hit on something
there.   Was in fact just about to start running that down.  That buffer needs to 548 bytes, not 256.

suborb, 2021-04-05:
@jayacotton I'm going to close this issue now. Feel free to open another issue if you need.

closed 2021-04-05 by suborb