z88dk/z88dk #1745: GSX autoloader

zx70, 2021-04-02:
This is the totally untested code I'm working on, its possible position could be after the full initialization and the redirection files being assigned already, e.g. after  INCLUDE "crt/classic/crt_command_line.asm".

I'm open to comments, DDSCREEN.PRL is a single driver which, in my intention, couldn't be overlaid at runtime as it happens with the standard GENGRAF tool.   No Workstation ID assignment should then be necessary and the allocated space after SP will be the exact one required by the only chosen driver.
To me, even keeping a static name like DDSCREEN.PRL could be a valid option.   The file could be placed in the same drive or forced to be on drive A.

```

EXTERN	open
EXTERN	readbyte
EXTERN	close
EXTERN	l_neg

PUBLIC _gsxfname

	jr	__skipfname

_gsxfname:
	defm "A:DDSCREEN.PRL"
	defb 0

_gsx_entry:
	defw 0
	
__skipfname:
	ld		hl,_gsxfname
	push	hl	; filename
	ld	de,0	; O_RDONLY
	push	de
	ld	hl,3 	; _IOUSE | _IOREAD
	push	hl	; flags
	call	open
	pop	bc
	pop	bc
	pop	bc
	pop	de
	
	ld (fileid_smc+1),hl
	push hl	; file id

	call	readbyte	; skip first byte
	
	pop hl
	push hl
	call	readbyte	; LSB
	ld	a,l
	pop hl
	push hl
	push af
	call	readbyte	; MSB
	pop af
	ld	b,l	; file size
	ld	c,a

	pop hl
	push bc
	
	ld	b,253	; 256-3  (should it be 128-3 ?)
skip_header:
	push hl	; file id
	push bc
	call	readbyte	; skip first byte
	pop hl
	pop bc
	djnz skip_header

	ex de,hl	; de=file ptr
	pop bc		; bc=file size

	ld h,b
	ld l,c
	call l_neg	
	add hl,sp
	
	ld	a,l
	and a
	jr z,page_boundary
	dec h
	ld	l,0
page_boundary:

	ld	(_gsx_entry),hl
	ld	sp,hl	; lower SP
	
	push de		; file ptr
	push bc		; driver size
	push hl		; gsx_entry

ld_loop:
	push    bc
	push    hl
	ld      h,d
	ld      l,e
	call    readbyte
	pop     hl
	pop     bc
	
	ld 		(hl),a
	inc		hl
	dec		bc
	ld 		a,c
	or		b
	jr		nz,ld_loop
	
	pop		hl		; gsx_entry
	ld 		b,h
	dec		b		; adjust page (MSB)
	pop		de		;
	
	pop     hl		; gsx_entry
	pop     de      ; driver size

; bit mapped reloc section
reloc:
	ld 		c,8 			; group of 8 bytes to be relocated 

	push    de
	push    bc
	push    hl
fileid_smc:
	ld      hl,0			; file id
	call    readbyte
	pop     hl
	pop     bc
	pop     de
reloc8:
	rlca					; test whether the current byte needs to be relocated
	push	af
	jr		nc,no_reloc
	ld 		a,b
	add		a,(hl)
	ld 		(hl),a
no_reloc:
	inc		hl
	dec		de
	ld 		a,d
	or		e
	jr		z,close_file
	pop		af
	dec		c
	jr		nz,reloc8
	jr		reloc

close_file:
	ld      hl,(fileid_smc+1)
	call    close

```

zx70, 2021-04-07:
It took me forever to debug the above routine, It was oddly shifting the allocated data depending on the padding bytes I was trying to remove from the first 256 bytes block.
It could be a bug in the CP/M read() function happening when the data to be loaded terminates at the exact sector boundary, I'm investigating.

zx70, 2021-04-08:
I think I met this problem already:  https://github.com/z88dk/z88dk-ext/commit/c27da99675c51e93b2f06adb9aa26580d7f5658a
It could be consequence of the recent improvements to boost the sequential read() performances.

zx70, 2021-04-08:
The root cause of #1644 could have been read() for CP/M, being indirectly invoked by fread()

fread() -> fgetc() -> readbyte() -> read([...],1)

zx70, 2021-04-08:
Ok, I'm still running in circles, the test program I've built to show the bug works perfectly  :(

suborb, 2021-04-08:
Are you opening the file as binary?

zx70, 2021-04-08:
Good point, I'll re-check everything tomorrow.

zx70, 2021-04-09:
I finally went for a workaround, I'm now closing the file after getting the block size from the first 3 bytes, opening it back and skipping the first 256 bytes at once rather than 253.
In this way it works fine   :)

suborb, 2021-04-25:
@zx70 - see this comment: although it's about write, I wonder if it's the same issue: https://github.com/z88dk/z88dk/issues/1458#issuecomment-826398159 - intermingling of sequential and random reads.

zx70, 2021-04-26:
It might be the case.   Not an easy bug to track down (they can be 2, read+write) but I was getting to very close conclusions while debugging.
It is worth to spend a bit more time on it, I'm sure there's a fix or a workaround if we're able to recreate the bug.

zx70, 2021-04-26:
Let me elaborate a little more on this:  in the above code, it seems that the pointers were working well in general but failing when a successive read command hits exactly the sector edge.
Oddly closing the file, opening it back and reading a whole 256 bytes chunk lines it up perfectly.

suborb, 2021-04-26:
>  failing when a successive read command hits exactly the sector edge.

So, with the code that was there, if you did:

1. read 1
2. read 1
3. Read 254

Then these got mapped into:

1. set pointer
2. Read rand 1
3. set pointer
4. Read rand 1 
5. set pointer
6. Read rand 126
7. Read sequential 128

The issue as alank figured out is this:

(from https://www.seasip.info/Cpm/bdos.html)
```
BDOS function 33 (F_READRAND) - Random access read record

Supported by: CP/M 2 and later.

Entered with C=21h, DE=FCB address. Returns error codes in BA and HL.

Read the record specified in the random record count area of the FCB, at the DMA address. The pointers in the FCB will be updated so that the next record to read using the sequential I/O calls will be the record just read. 
```

The random read sets the sequential filepointer to the start of record that's just being read. So the sequential read reads the original record in again.

The sequential optimisation just doesn't play nicely with `lseek()` either, so, sorry @feilipu I've backed it out - I guess it could be restored with an O_DIRECT flag and disable seeking on the fd.

feilipu, 2021-04-26:
I'll take working over not working any day.

Still with that front of mind, it may be possible to rework.

Also, lifting the userID calls helped the most anyway. So as long as that isn't broken too?

suborb, 2021-04-27:
Regarding the `setuid()` changes, I don't know why a call to `setuid()` is needed here at all. I would think it should only be needed at `open()` time and possibly `close()`. 

Maybe I'm missing an MP/M requirement though.

zx70, 2021-04-27:
Could it be useful when we are contemporairly dealing with many files on different userid areas?

suborb, 2021-04-27:
Oh, @zx70 I'm so sorry, I just looked at my bookcase and look what I found...

[image]

Any info you need or have you managed to reverse engineer all of it?

suborb, 2021-04-27:
> Could it be useful when we are contemporairly dealing with many files on different userid areas?

I had thought the user area was in the FCB but it's not. So I'm guessing it's needed to find the directory entry when we need to grab a new extent.

That makes sense then. With us saving and restoring it then we don't trample on any user code.

zx70, 2021-04-27:
:)   I think I have all I need, but it doesn't mean I'll surely succeed !

I think I finished decoding the existing disassembly of gsx.sys, if I correctly understood the way it works, it includes the expressions to scale the driver resolution and make it compatible with the VDI metrics (forth and back.. also for the results of the inquiry command).   It appears to copy the "xy" parameters locally while doing it, which might be a requirement to permit the video memory paging when needed.    I think it picks the declared driver resolution at startup to make it work...   it is the first clue I ever had of a separations between GIOS and GDOS, I tried to adopt that code to check the way that scaling trick worked but no success so far.

zx70, 2021-05-14:
I confirm the new fixed FCNTL version works, thankyou Dom!
I tested it on the GSX autoloader code and on the "lar.c" example.

Reverting the loader to its original shape saves few bytes and probably makes it slightly faster.

```
EXTERN	open
EXTERN	read
EXTERN	readbyte
EXTERN	close
EXTERN	l_neg

PUBLIC _gsxfname
PUBLIC _gsx_entry
PUBLIC _gsx_pos

	jr	__skipfname

_gsxfname:
	defm "A:DDSCREEN.PRL"
	defb 0

_gsx_entry:
;	jp	0
	defb 195
_gsx_pos:
	defw 0

__skipfname:
	ld		hl,_gsxfname
	push	hl	; filename
	ld	de,0	; O_RDONLY
	push	de
	ld	hl,3 	; _IOUSE | _IOREAD
	push	hl	; flags
	call	open
	pop	bc
	pop	bc
	pop	bc
	
;	ld (__fileid+1),hl
	ld	a,h
	and l
	cp 255	; -1 ?
	jp z, __notfound

	push hl	; file id
	call	readbyte	; skip first byte
	pop hl

	push hl
	call	readbyte	; LSB
	ld	a,l
	pop hl
	push hl
	push af
	call	readbyte	; MSB
	pop af
	ld	b,l	; file size
	ld	c,a
	pop hl

	push bc
	
	ld	b,253	; 256-3
skip_header:
	push bc	; file size
	push hl	; file id
	call	readbyte	; skip
	pop hl
	pop bc
	djnz skip_header

	ex de,hl	; de=file id
	pop bc		; bc=file size

	ld h,b
	ld l,c
	call l_neg	
	add hl,sp

;	ld	a,l
;	and a
;	jr z,page_boundary		; we're at a byte boundary already
	dec h
	ld	l,0
;page_boundary:

	ld	(_gsx_entry+1),hl
	ld	sp,hl	; lower SP

	
	push hl		; position
	push bc		; size

	; Load driver body
	
	ld		(__fileid+1),de

	push	de
	push	hl		; driver position (gsx_entry)
	push	bc		; driver size
	call	read
	pop		bc
	pop		hl
	pop		de

	pop	de	; size
	pop hl	; position

	ld 		b,h
	dec		b		; Adjust page offset (MSB), the PRL file default position is 100h

; bit mapped reloc section
reloc:
	ld 		c,8 			; group of 8 bytes to be relocated 

	push    de
	push    bc
	push    hl
__fileid:
	ld      hl,0	; file id
	call    readbyte
	ld		a,l
	pop     hl
	pop     bc
	pop     de

__reloc8:
	rlca			; check whether the current byte needs to be relocated
	push	af
	jr		nc,__no_reloc
	ld 		a,b
	add		a,(hl)
	ld 		(hl),a
__no_reloc:
	inc		hl
	dec		de
	ld 		a,d
	or		e
	jr		z,__close_file
	pop		af
	dec		c
	jr		nz,__reloc8

	jr		reloc

__close_file:
	pop		af				; balance stack
	ld      hl,(__fileid+1)	; file id
	push	hl
	call    close
	pop		hl

	ld		hl,(_gsx_entry+1)
	jr		__gsx_init_done

__notfound:
	ld      hl,(exitsp)

__gsx_init_done:
```

closed 2021-05-14 by zx70