z88dk/z88dk #1746: Much large code

varmfskii, 2021-04-02:
When producing a zxnext dot file from the code in [https://gitlab.com/varmfskii/showsimg](url) there is a 37% increase between the size of the code produced by the 2.1 release as opposed to the 2.0.

suborb, 2021-04-02:
So, varying the sdcc versions but keeping the lib the same and looking at the size of the executable:

Today (12069) -> 10454
2.1 (12036) -> 10454
2.0 (11556) -> 10399

I'm not seeing a 37% increase in the code size, you're going to give me a hint!

varmfskii, 2021-04-02:
When I build with 2.0 (including the libraries that shipped with 2.0):  7693 bytes.
with 2.1 and the current release: 10454 bytes
7693/10454 = 1.358 so 36%, not 37%

This e-mail message (including any attachments) is for the sole use of
the intended recipient(s) and may contain confidential and privileged
information. If the reader of this message is not the intended
recipient, you are hereby notified that any dissemination, distribution
or copying of this message (including any attachments) is strictly
prohibited.

If you have received this message in error, please contact
the sender by reply e-mail message and destroy all copies of the
original message (including attachments).

commit 92c606de (2021-04-02): Fix #1746: Temporary solution to this issue

The option -startupoffset was matching -startup which was evaluating
to 0, thus pulling in the wrong console drivers and bloating the
size.
closed 2021-04-02 by suborb