z88dk/z88dk #1748: (sam) (classic)

dandoore, 2021-04-05:
On the SAM Coupé the POINT(X,Y) function should return the pixels colour value (0 to 15) in MODE 3&4 but 0-1 (not paper) in MODE 1/2/

Documented here: https://github.com/z88dk/z88dk/wiki/SAM-Coup%C3%A9-Library-Reference

suborb, 2021-04-05:
Funnily enough I was looking at `point()` the other day.

I agree that there should be `point()` that returns the pixel colour - unfortunately the `point()` function by design returns 0/1 for notset/set.

Need to think about this one a bit more...

dandoore, 2021-07-07:
@suborb Any more thoughts on this one or is it a PITA?

suborb, 2021-07-13:
It's a fair bit of work to add it to all the appropriate targets and screen modes - it is a genuinely useful extension so I'm not going to close this issue, but at the moment I just don't have the energy/time/enthusiasm/etc to make a change like that.

For a SAM specific function (and to lay the ground work for the generic version), take a look at the files in here: https://github.com/z88dk/z88dk/tree/master/libsrc/target/sam/graphics - the pixel_MODE.inc files contain the implementation for the various graphic modes. The approach would be:

* Add a IF POINTC clause that returns the colour of the pixel
* Create a cpointc(x,y) function in there that queries the screen mode and then calls the appropriate routine: the pattern is in the appropriate plot/xor/unplot files

Mode 1 files are located as MODE0 here: https://github.com/z88dk/z88dk/tree/master/libsrc/target/zx-common/graphics