z88dk/z88dk #1750: (classic) Update sp1 to newlib version

suborb, 2021-04-06:
Following on from the section issue and the pending ASMDISP issue, I've noticed that sp1 in classic doesn't play nicely (or at all with sdcc).

So, lets switch the +zx sp1 over to newlib version and pull in the sdcc bindings. Looking at it, it might be easier to force sccz80 to use `__stdc` bindings rather than `__smallc`.

The +zx81 implementation should remain the same, I'll review the ts2068 implementation to see if we can share the C bindings.

feilipu, 2022-05-11:
Resolved by #2006 ?

jorgegv, 2022-07-22:
This was indeed what @suborb asked me to do, I hope I was up to the task!

But the `share_sp1` branch is still not merged to master, will it be? Or are there some other things that need to be done?

suborb, 2022-07-22:
It will be, but I did think there was something still to do? Maybe I'm mis-remembering though.

I'll setup a merge to go in after the snapshot.

jorgegv, 2022-07-22:
Sorry, there was still a mistery bug with one of the examples, but all the rest is done. For some reason I could not debug (and boy I tried!), `ex5e.c` did not work correctly. Still working on it, but I think the rest is ready for merging.

feilipu, 2022-10-02:
Is this closed by #2048 ?

jorgegv, 2022-10-02:
Apparently yes! I haven't been able to spend time debugging the `ex5e.c` problem, but I just updated my repo now and rebuilt everything, and now `ex5e.c` compiles _and runs fine_ in Classic with newlib SP1 sources, both with SCCZ80 and with SDCC. The error does not happen any more.

Since I was getting crazy with this and everything looked ok, it might be that indeed it was triggering some obscure compiler bug that got fixed between april and today...

I'll try to do a `git bisect` one of this days and I'll update.

closed 2022-10-04 by feilipu