z88dk/z88dk #1751: Console scrolling & textbackground color of new line.

everslick, 2021-04-07:
I'm developing a shell for 8-bit computers [github](https://github.com/everslick/push) and I'm also targeting z88dk supported platforms. My first target is ZX Spectrum. After some initial fighting with the toolchain and different libs, compilers and console drivers, I have the console part mostly working. One of the remaining issues is, that whenever the prompt reaches the last line and text is about to scroll up, the new scrolled in line will have the default BG color WHITE. IMHO it should have the textbackground() that was last used. Right?

everslick, 2021-04-07:
That fixed my issue! WOW

Thank you very much for this quick response!

The "initial fighting" I mentioned above was more me not reading/following the documentation. I guess there is not much you can do about THAT! ;-)

Should you ever do more work on the console driver, give [push](https://github.com/everslick/push) a shot. It is a nice testcase/benchmark, IMHO.

closed 2021-04-07 by everslick

suborb, 2021-04-07:
True, there's only so much that can done about that particular scenario!!

Yeah, it looks like a nice driver for conio and also for fileio when that bit is padded out. It's going on my test cases lists!

 I was just browsing through some of the code, going "this looks a familiar style", and then I got to lined.c and went "of course". I may have used linenoise quite a few times in several projects!

Was there anything majorly missing from conio? Other projects have successfully switched between cc65 and z88dk without issues so it would be good to know what could be missing.

everslick, 2021-04-07:
Yes FILEIO is next on my list - stay tuned! :)

The only .ifdefs for the ZX code are:

* missing revers() which I have implemented by setting textbackground() and textcolor() manually.
* I had to make `bgcolor()` an alias for `textbackground()` (what is not quite the same).
* `cursor()` is missing from conio.h, and so is a blinking harware curser, so I draw an inverted space on its place instead.
* linefeed is tricky: it is \r on ZXSpec but \r\n everywhere else, so I have to write each printf like `printf(".." LF);`

The cross platform magic for the console happens mostly in `term.c`.

I still have to figure out a workable key-mapping, but that is beyond the scope of z88dk. :)

suborb, 2021-04-07:
> missing revers() which I have implemented by setting textbackground() and textcolor() manually.

There's the inverse escape codes ESC p and ESC q  which I can easily wrap up to deal with that one.

> I had to make bgcolor() an alias for textbackground() (which is not quite the same).

That one is tricky - it'll work nicely on textmode (where a background colour is supported) or on attribute byte based systems, but chunky pixel targets would hate it! It's easy enough to write some custom code for +zx though since it's attribute based.

> cursor() is missing from conio.h, and so is a blinking harware curser, so I draw an inverted space on its place instead.

#1245 has been raised for a while. Any blinking would be machine dependent, eg the 6845 machines support it in hardware, but software cursors wouldn't blink since we don't by default hook the interrupt.

> linefeed is tricky: it is \r on ZXSpec but \r\n everywhere else, so I have to write each printf like printf(".." LF);

The battles that have been had over linefeeds.....

> I still have to figure out a workable key-mapping, but that is beyond the scope of z88dk. :)

You say that, but...compile with --hardware-keyboard and you'll get our control key mapping....

Xalior, 2021-04-07:
> cursor() is missing from conio.h, and so is a blinking harware curser, so
> I draw an inverted space on its place instead.
>
> #1245 <https://github.com/z88dk/z88dk/issues/1245> has been raised for a [...]
I generally “cheat” this on +ZX with the `FLASH` attribute bit.

-Dx

everslick, 2021-04-07:
> There's the inverse escape codes ESC p and ESC q which I can easily wrap up to deal with that one.

I tried, "\e[2m" and "\e[27m", that did not work.
 
> #1245 has been raised for a while. Any blinking would be machine dependent, eg the 6845 machines support it in hardware, but software cursors wouldn't blink since we don't by default hook the interrupt.

I'd consider this not a problem. If it is so important, we can always roll some target specific code.

> > linefeed is tricky: it is \r on ZXSpec but \r\n everywhere else, so I have to write each printf like printf(".." LF);
> 
> The battles that have been had over linefeeds.....

indeed.

maybe some pragma-define?
 
> You say that, but...compile with --hardware-keyboard and you'll get our control key mapping....

oh, nice, will check that out1 THX!!!

everslick, 2021-04-07:
> I generally “cheat” this on +ZX with the `FLASH` attribute bit.

I think I tried something like that, but I couldn't make it work. Maybe i was only stupid. dunno.

suborb, 2021-04-07:
> > There's the inverse escape codes ESC p and ESC q which I can easily wrap up to deal with that one.
> 
> I tried, "\e[2m" and "\e[27m", that did not work.

Ah sorry, I've now played around with the project and see you're using VT100 codes. \e[7m and \e[27m seem to be working for me.

> > The battles that have been had over linefeeds.....
> 
> indeed.
> 
> maybe some pragma-define?

Well....I've found the code responsible and I've undone that "temporary" patch which existing for about 20 years or so: https://github.com/z88dk/z88dk/commit/9c6673ec0b9e7ccab04afed3e3f1525c447d5d38

We've always used \n to move to the next line at column 0. The mapping of this changed a few years ago from 13d to 10d which leaves 13 free to fulfil its original purpose.

> > You say that, but...compile with --hardware-keyboard and you'll get our control key mapping....
> 
> oh, nice, will check that out1 THX!!!

That actually works really well - just press symbol + caps to emulate the ctrl key and it looks like everything "just works"

everslick, 2021-04-08:
> > I tried, "\e[2m" and "\e[27m", that did not work.
> 
> Ah sorry, I've now played around with the project and see you're using VT100 codes. \e[7m and \e[27m seem to be working for me.

will give it another shot, thx.
 
> Well....I've found the code responsible and I've undone that "temporary" patch which existing for about 20 years or so: [9c6673e](https://github.com/z88dk/z88dk/commit/9c6673ec0b9e7ccab04afed3e3f1525c447d5d38)

oh, wow, just wow. I didn't expect to get that patch reverted. But it is great, feels so much more natural. I have already removed my LF hack. so no backsies!!! ;-)