z88dk/z88dk #1753: Strange code generation bug for `uint8_t += uint8_t` (both on stack)

everslick, 2021-04-08:
How can code like:

```
// sz and len are uint8_t on the stack
printf("sz=%u, len=%u\n", sz, l);
sz += l;
printf("sz=%u, len=%u\n", sz, l);
```

ever produce output like:

[image]

The code that shows the bug is here:

https://github.com/everslick/push/blob/master/src/cli.c#L185-L229

the offending line is this:

https://github.com/everslick/push/blob/master/src/cli.c#L215

I already shuffled the variables on the stack and reduced the stack size, by making buf[] smaller, to rule out stack corruption, but the bug does not look like stack corruption anyway.

everslick, 2021-04-08:
Expected output:

[image]

suborb, 2021-04-08:
There's a mistake in z80rules.1

I've got other rules outstanding, but line 3830 currently has this:

        ld      hl,%1   ;const

It needs to be:

        ld      hl,%eval(%1 2-)   ;const

[Tabs are important] - I'll run through some more checks this evening to make sure it's actually correct.

everslick, 2021-04-08:
fantastic! thank you!

everslick, 2021-04-08:
I tried your fix, but it actually made things worse. now the `l` variable is 255 after each `+=`. if this is of any help to you.

Interestingly `sz` is now increased by double the value it should be e.g. if `l` is **3** `sz` increases by **6**.

everslick, 2021-04-08:
I tried your fix, and it works flawlessly. Ignore my comment from 10 minutes ego, which i have since deleted. (i testet MY test code, without realizing).

suborb, 2021-04-08:
That's great, the fix went in this commit: https://github.com/z88dk/z88dk/commit/030732620e7098322acc0a4b97c63440b38d3a94

closed 2021-04-08 by suborb