z88dk/z88dk #1754: (sam) (calssic) Large array causes crash

dandoore, 2021-04-08:
The following code causes a crash when defining an array:

```
#include <stdio.h>

int main(void) {
	printf("Before...\n");
   	unsigned char array[1500];
	printf("After...\n");
}
```

Same code compiled under ZX works fine under FUSE.

suborb, 2021-04-08:
It took me a while to find something that I knew I'd written: https://github.com/z88dk/z88dk/wiki/Platform--Sam-Coupe.md#the-stack

dandoore, 2021-04-08:
Ah, I see - should I be using malloc() or something similar here instead?

suborb, 2021-04-08:
Just use the pragma to increase the stack size if you're doing something quick and easy.

It looks like I've not set malloc up properly yet - thanks for the reminder.

closed 2021-04-08 by dandoore