z88dk/z88dk #1755: Unresolved symbols when linking with esxdos

everslick, 2021-04-10:
My apologies if this is more of an user question, then a bug or issue in z88dk. I'm about to implement directory browsing and navigation and while my code compiles, it fails to link missing some symbols like: rename, remove, _pwd, mkdir, opendir, closedir. I guess I'm missing something obvious here.

```
clemens@alien:~/Devel/8bit/push/src$ make zx
make SDK=z88dk MACHINE=zx COLS=32 z88dk
make[1]: Entering directory '/home/clemens/Devel/8bit/push/src'
zcc +zx -vn -startup=1 -create-app -I../z88dk/include  -pragma-define:CLIB_EXIT_STACK_SIZE=0 -pragma-define:CLIB_CONIO_NATIVE_COLOUR=1 -pragma-define:ansicolumns=32 -pragma-need=ansiterminal -pragma-export:__zx_console_attr=0x5C8d -pragma-export:__zx_mode0_console_w=0 -lesxdos -O3 -DVERSION=0.0.2 -DZ88DK -DAMALLOC -DHAVE_CONIO -DHAVE_OSD -DHAVE_SWCURSOR -DHAVE_HISTORY -DHAVE_HINTS -DHAVE_COMPLETION -DZX -DCOLUMNS=32 -DHAVE_FILEIO -DHAVE_DIVMMC -DTOOLCHAIN=z88dk -DMACHINE=zx -o push-zx main.c term.c lined.c cli.c parse.c zxspec.c divmmc.c
Error at file 'cli.c::cmd_mv::0::12' line 148: symbol 'rename' not defined
Error at file 'cli.c::cmd_rm::0::15' line 163: symbol 'remove' not defined
Error at file 'divmmc.c::fileio_getcwd::0::3' line 40: symbol '_pwd' not defined
Error at file 'divmmc.c::fileio_getcwd::0::3' line 40: symbol '_pwd' not defined
Error at file 'divmmc.c::fileio_mkdir::0::7' line 58: symbol 'mkdir' not defined
Error at file 'divmmc.c::fileio_ls::0::10' line 70: symbol 'opendir' not defined
Error at file 'divmmc.c::fileio_ls::0::10' line 70: symbol 'closedir' not defined
Error at file 'divmmc.c::fileio_ls::0::10' line 70: symbol 'closedir' not defined
Errors in source file ../../z88dk/lib/config//../..//lib/target/zx/classic/spec_crt0.asm:
Error at file 'cli.c::cmd_mv::0::12' line 148: symbol 'rename' not defined
                   ^ ---- static void cmd_mv(uint8_t argc, char **argv) {
Error at file 'cli.c::cmd_rm::0::15' line 163: symbol 'remove' not defined
                   ^ ---- static void cmd_rm(uint8_t argc, char **argv) {
Error at file 'divmmc.c::fileio_getcwd::0::3' line 40: symbol '_pwd' not defined
                   ^ ---- char *fileio_getcwd(char *buf, uint8_t size) {
Error at file 'divmmc.c::fileio_getcwd::0::3' line 40: symbol '_pwd' not defined
                   ^ ---- char *fileio_getcwd(char *buf, uint8_t size) {
Error at file 'divmmc.c::fileio_mkdir::0::7' line 58: symbol 'mkdir' not defined
                   ^ ---- int8_t fileio_mkdir(const char *dir) {
Error at file 'divmmc.c::fileio_ls::0::10' line 70: symbol 'opendir' not defined
                   ^ ---- int8_t fileio_ls(uint8_t flags, char *path) {
Error at file 'divmmc.c::fileio_ls::0::10' line 70: symbol 'closedir' not defined
                   ^ ---- int8_t fileio_ls(uint8_t flags, char *path) {
Error at file 'divmmc.c::fileio_ls::0::10' line 70: symbol 'closedir' not defined
                   ^ ---- int8_t fileio_ls(uint8_t flags, char *path) {
make[1]: *** [Makefile:97: z88dk] Error 1
```

suborb, 2021-04-10:
Nope, you're not doing anything silly, they've just not been bridged from the device API to a POSIXy API. 

From reading commits (and from a fuzzy memory), it looks like ESXDOS doesn't actually have a stable public API which complicates things "a little":

Running through the issues you've got:

* remove: I've just corrected this so it's being exported correctly
* pwd: There's `getwd()` and `getcwd()` available already

The following are only implemented/exposed in the ZX Next version of ESXDOS (but would still need bridging) - handful of lines of code:

* rename
* mkdir

opendir/readdir/closedir: 

Wonderfully, there's two versions of this api - one for ZX and one for ZX Next. Looking through the code, it looks like the header is mostly a placeholder that I setup so I could write the API for the +3. That does seem to have stalled back in 2017 though.

The "raw" ESXDOS apis are exposed in `<arch/zx/esxdos.h>` and `<arch/zxn/esxdos.h>` which for the ZX Next does have all the functionality you need. I realise it's a cheeky ask, but if you do bridge them, do you mind if I borrow them back and add them to the library?

everslick, 2021-04-10:
ahh, thx for the detailed answer. that makes a lot of sense now.

while trying to get a hang on all the inner workings of esxdos I pondered the idea to not directly interface with the esxdos API, nor the standard libc interface, but instead call the external .dot commands. especially on the next, we have fully working implementations for mv, ls, cd, cp, ... and friends. For that to work, I would need a way to start those commands and resume execution (or restart the shell) after the commands exit. I'm not sure how this could work though, if at all possible. any comments or ideas, are very much appreciated!

Having said that, I will need to read the current directory anyway, otherwise TAB-completion cannot work in the shell. so at least I need to implement that. I will try to bridge them, as you said, and whatever I come up with, you have my explicit permission to integrate that in z88dk with the license of your choice.

btw: missing `_pwd` was a typo in my code anyway.

suborb, 2021-04-10:
There's the nextos API doc here: https://github.com/z88dk/techdocs/blob/master/targets/zx-next/nextos/nextzxos_api.pdf

It's not been updated in a while, but there appears to be an esxdos execute call. I've no idea what it does regarding memory - I guess it depends on whether the dot executable confine themselves to the 8k RAM?

ped7g, 2021-04-11:
you can get latest from the Next repository: https://gitlab.com/thesmog358/tbblue/-/blob/master/docs/nextzxos/NextZXOS_and_esxDOS_APIs.pdf
There's some info how the memory paging is expected to be for certain API calls, but you need to scavenge that file quite completely, and I'm always afraid there's some further silent assumptions (also I don't remember all details from head), so yes, what you do to your memory layout and whether you are running as dot executable, or as tap/nex code, makes difference when calling the API.
The esxdos-like stubs are usually the least demanding ones, the Next API BASIC interpreter related ones the most demanding.

Ad:
> while trying to get a hang on all the inner workings of esxdos I pondered the idea to not directly interface with the esxdos API, nor the standard libc interface, but instead call the external .dot commands

It's impossible\* to execute dot command from dot command, and overall I'm failing to see the benefit of adding such complexity instead of calling the API directly? I get the point about esx API being not public yet, but the esxdos-like API documented by NextZXOS is already public, and to execute dot command you need some API either, so... If you are trying to avoid the non-public esxdos API, I guess using directly the +3 API is then better as that is public API and "native" for NextZXOS, so for Next-specific SW it offers most options.

\*) well, you can "finish" your dot command with follow-up code being already in regular memory and continue as non-dot code, then you can execute dot commands from within, but you can't easily return to OS except soft-reset - which by the way is quite legit and simple way how to exit for example a game or other type of SW which mostly takes over machine and restoring OS would be too tricky

aralbrec, 2021-05-07:
The esxdos api was given to me by Miguel and is documented in the implementation here:

https://github.com/z88dk/z88dk/tree/master/libsrc/_DEVELOPMENT/arch/zx/esxdos/z80

That is not going to change; there is too much software using it.  There may be extensions that come later however.

The api is formally documented by NextZXOS in the pdf already linked above.  NextZXOS extends the API so comparing the esxdos code in the repo link above with the pdf will provide more insights than the terse documentation from Miguel.

Although Migual is aware of NextZXOS and its extensions to the esxdos api, there is a chance he will extend esxdos in an incompatible direction.  It has been many years however without major updates to esxdos so there's no telling if that will happen.

In any case, z88dk is separating the esxdos and nextzxos apis (the above link and https://github.com/z88dk/z88dk/tree/master/libsrc/_DEVELOPMENT/arch/zxn/esxdos/z80 respectively) even though the esxdos subset is nearly identical just in case there is a divergence.  If that happens, it would be a simple matter of recompiling to make the same program work in both environments.

aralbrec, 2021-05-07:
> while trying to get a hang on all the inner workings of esxdos I pondered the idea to not directly interface with the esxdos API,
> nor the standard libc interface, but instead call the external .dot [...]

I will be putting file i/o redirection and piping into the z88dk for dot commands soon.  Then it will be possible to redirect output of commands like ls to a temporary file that you can read.  However it's not hard to read the directory contents directly using the esxdos api.  In fact the dot commands you listed (`ls`, `mv`, `cd`, etc) are written in c and do exactly that.  The source code is at https://gitlab.com/thesmog358/tbblue/-/tree/master/src/c/DotCommands.

The `run` dot command contains relatively simple code that walks a directory:
https://gitlab.com/thesmog358/tbblue/-/blob/master/src/c/DotCommands/run/run.c#L232

It opens the directory twice, once to read short filenames and the second to read long filenames.  Directory entries are in the same order so reading an entry from each corresponds to the same file.  Some filename matching occurs on the short filename and the long filename, the latter using the function `glob_fat()` which does wildcard matching with `*` and `?`.

closed 2022-05-11 by feilipu