z88dk/z88dk #1756: (sam) Keyboard input difference compared to ZX

dandoore, 2021-04-10:
I have been trying to use kbhit() followed by getch() to capture keystokes and SAM seems to be extremely sensative compared to the same code under ZX;

```
#include <stdio.h>
#include <conio.h>

void main()
{
	clrscr();
    char ch;
    while (1)
		{
		if ( kbhit() )
			{
			ch = getch();
			if (ch == 'x') break;
			printf("%c",ch);
			}
		}
	printf("\nEnd.\n");
}
```
Pressing keys alphabetically at normal speed gives this on ZX;

[image]

And this on SAM

[image]

ZX loses some keystrokes (g&h didn't register) but SAM goes like it has an auto-repeat of 0ms!

suborb, 2021-04-11:
So I suspect you'll see the same behaviour on the +zx if you compile with `--hardware-keyboard`.

`kbhit()` reads the instantaneous state of the keyboard and then stores any keypresses to be picked up with `getch()`. If there's no keypress stored, `getch()` calls `fgetc_cons()`

With +zx by default we read the state of the keyboard by looking at (23560) which is of course updated by the ROM, after debouncing etc

On the Sam and with --hardware-keyboard there's no debouncing with `kbhit()`, but there is with `fgetc_cons()` so that's why you're seeing the different result.

dandoore, 2021-04-11:
I've not specified  `--hardware-keyboard` so I presume this is default behaviour but for the ZX it's not - is there a way to get the debounced input on SAM for kbhit() so that I can act on the kbhit() (in this case, `break` out of a loop) and then getch() the character out of the buffer?

closed 2021-04-11 by suborb

dandoore, 2021-04-12:
Awsome, [that](https://github.com/z88dk/z88dk/commit/eade0617f577d05c7ceb8f019e7fee23b0bdb511) fixes the issue :)

commit eade0617 (2021-04-11): Fix #1756: Hardware keyboard kbhit doens't debounce

So don't store the key obtained from getk, it should mostly
work unless there's a long delay between kbhit and getch being
called
commit 65204587 (2021-05-15): Issue #1791: Fix #1756 a different way.

Rather than not storing the keypress, lets just delay a little -
this achieves the same aim and seems to make everything compatible
and matches the delay in fgetc_cons when we don't have a key
pressed.