z88dk/z88dk #1759: Potential invalid opcode in stdlib

jayacotton, 2021-04-13:
Running a large program (under development) I am hitting this situation
+++ INVALID Z180 OPCODE @8647H: DD 6A 4D 44 03 16 00 6A
The DD 6A does not decode to anything in z80 or z180 instruction sets.

This is in:
Section 'code_stdlib' size: 184 bytes ($85D8 to $868F)
Section 'code_string' size: 124 bytes ($8690 to $870B)

Here is a disassembled output from DDTZ

8637  7A        MOV  A,D
8638  CDF582    CALL 82F5
863B  57        MOV  D,A
863C  7E        MOV  A,M
863D  CDE382    CALL 82E3
8640  3848      JRC  868A
8642  BA        CMP  D
8643  3045      JRNC 868A
8645  5F        MOV  E,A
8646  7A        MOV  A,D
8647  DD6A      MOVX L,D

It seems that DDTZ is attempting to decode this anyway.

suborb, 2021-04-13:
I think somewhere along the line you're using `strtoul()`- but you may not be calling it directly.

Looking at the implementation I can see there's a code path to handle Z180 and CPUs that don't support splitting the index registers.

However, the only targets this code path are in (besides Rabbit code) will be +scz180, +yaz180, +z180, and I'm guessing you're using +cpm?

feilipu, 2021-04-13:
> the only targets this code path are in (besides Rabbit code) will be +scz180.

Yes, that is possibly the only reason to use the `+scz180 -subtype=cpm` subtype.

suborb, 2021-04-13:
A quick search shows the 8 bit index registers are only used in a couple of non-performance sensitive functions.  

As a result I don’t think I’ve got an issue switching classic to using the generic code to sidestep this issue.

feilipu, 2021-04-13:
> Obviously floating point is a different matter entirely.

I guess it depends. For example, math32 is z80 clean, but not 8080 clean. So it probably shouldn't be used for a conservative (8080) CP/M. But it's not a big deal to waste the z180 optimised mlt solution in this case, IMHO, given the code might also be run on a z80.

jayacotton, 2021-04-13:
Yes, I am using +cpm.  I was unaware of this index register issue.  Must
study up.

tnx
jc

jayacotton, 2021-04-13:
excerpts from the make file FYI
CFLAGS  = +cpm --list --c-code-in-asm  -Wall -pragma-include:zpragma.inc
LINKOP  = +cpm -Cl -v -create-app -m -pragma-include:zpragma.inc -DAMALLOC2
STORE = -DFRAM
htlist: htlist.o w5500.o dhcp.o spi.o socket.o ethernet.o dns.o wizchip_conf.o libinet.o libhost.o
        zcc $(LINKOP) -ohtlist htlist.o w5500.o dhcp.o spi.o socket.o ethernet.o dns.o wizchip_conf.o 
htlist.o: htlist.c
        zcc $(CFLAGS) $(STORE) -c htlist.c

jayacotton, 2021-04-13:
Is there a subtype=z180 ?

If so, would that code work with z80 ?

suborb, 2021-04-13:
No, there's not a subtype=z180, the intention is that the core classic z80 library (i.e. the one you are using) should work on the z180 as well so I've disabled usage of the undocumented opcodes.

feilipu, 2021-04-14:
@jayacotton, I've re-read [the thread on Mark's Shift Register SPI Module](https://groups.google.com/g/rc2014-z80/c/T7j-9CAsKAo/m/7UboeCMIAgAJ) for the RC2014, and I hope that I can offer some more general input / discussion back on that thread.

But here some specifics...

On the `CFLAGS` in your make file, the `--c-code-in-asm` option is not good to include for the classic `+cpm` target because it will break some peephole optimisations in sccz80. The c code appears as comments in the assembly, which will cause the peephole optimisation to miss opportunities. Best to only use it when you're debugging, otherwise don't use it. For sdcc I don't think it is such an issue because the compiler does more optimisation so peephole optimisation is not so critical.

> Is there a `subtype=z180` ?

No, there is a [newlib](https://github.com/z88dk/z88dk/wiki/Introduction) [target for z180](https://github.com/z88dk/z88dk/tree/master/libsrc/_DEVELOPMENT/target/z180). It is closely related to the z80 target and essentially an [embedded (raw metal) target](https://github.com/z88dk/z88dk/wiki/NewLib--Platform--Embedded) with few if any drivers. The assumption is that you'll be writing your own drivers. I wouldn't use this as a starting point.

If you're targeting general RC2014 usage, then you need to have z80 support, and shouldn't invoke z180 optimisations. So the fixes @suborb has already made to the classic `+cpm` target are pretty much all you need to worry about. You could also use the [`+rc2014` target](https://github.com/z88dk/z88dk/blob/master/lib/config/rc2014.cfg) and the `-subtype=cpm`, which would invoke sdcc and the newlib by default.

But, if you're keen on only using z180 then  the [`+scz180` target](https://github.com/z88dk/z88dk/blob/master/lib/config/scz180.cfg), and there is a `-subtype=cpm` subtype that implements STDIO calls using BDOS, but gives you z180 instruction optimisation where possible.

jayacotton, 2021-04-14:
@flilipu  I implemented your suggestions and after cleaning up a bunch of compile errors and warnings, I get:

>> Not to concerned about the parse warning in SPI.C but the link errors are a real problem.

$ make clean;make htlist
rm -f *.o *.err *.lis *.def *.lst *.sym *.exe *.COM  driver ifconfig dnsprnt ping dig ftp telnet wget myget get ctc ntp date https clock pingnoti telnetd mac timezone today uptime test *.map
rm -rf html irc latex email
zcc +rc2014 -subtype=cpm -Wall -pragma-include:zpragma.inc  -DFRAM -c htlist.c
zcc  +rc2014 -subtype=cpm -Wall -pragma-include:zpragma.inc  -DFRAM -c  w5500.c 
zcc  +rc2014 -subtype=cpm -Wall -pragma-include:zpragma.inc  -DFRAM   -c  dhcp.c 
zcc  +rc2014 -subtype=cpm -Wall -pragma-include:zpragma.inc   -DFRAM -c  spi.c 
spi.c:223: warning 85: in function spi_delay unreferenced function argument : 'time'
spi.c:316: info 218: z80instructionSize() failed to parse line node, assuming 999 bytes
'._M3   ld a,(_lbyte_out) ; get the output data byte'

spi.c:316: info 218: z80instructionSize() failed to parse line node, assuming 999 bytes
'._M1   ld (_spi_data_bit) , a ; write out the data bit'

spi.c:316: info 218: z80instructionSize() failed to parse line node, assuming 999 bytes
'or     80h'

spi.c:316: info 218: z80instructionSize() failed to parse line node, assuming 999 bytes
'and    4'

spi.c:316: info 218: z80instructionSize() failed to parse line node, assuming 999 bytes
'._M2   ld b, a ; build up a byte here'

spi.c:316: info 218: z80instructionSize() failed to parse line node, assuming 999 bytes
'or     b'

zcc  +rc2014 -subtype=cpm -Wall -pragma-include:zpragma.inc   -DFRAM  -c  socket.c 
zcc  +rc2014 -subtype=cpm -Wall -pragma-include:zpragma.inc   -DFRAM  -c  ethernet.c 
zcc  +rc2014 -subtype=cpm -Wall -pragma-include:zpragma.inc  -DFRAM -DHOSTS -c  dns.c 
zcc  +rc2014 -subtype=cpm -Wall -pragma-include:zpragma.inc  -DFRAM  -c  wizchip_conf.c 
zcc +rc2014 -subtype=cpm -Wall -pragma-include:zpragma.inc  -DFRAM -c libinet.c
zcc +rc2014 -subtype=cpm -Wall -pragma-include:zpragma.inc  -DFRAM -c libhost.c
zcc +rc2014 -subtype=cpm -create-app -pragma-include:zpragma.inc -DAMALLOC2  -ohtlist htlist.o w5500.o dhcp.o spi.o socket.o ethernet.o dns.o wizchip_conf.o libinet.o libhost.o
Error at file 'fcntl/z80/asm_vopen.asm' line 60: symbol 'asm_target_open_p1' not defined
Error at file 'fcntl/z80/asm_vopen.asm' line 127: symbol 'asm_target_open_p2' not defined
Errors in source file C:/cygwin64/home/lbmgm/projects/z88dk/lib/config/\..\..\\libsrc\_DEVELOPMENT\target\
rc2014\rc2014_crt.asm.m4:
Error at file 'fcntl/z80/asm_vopen.asm' line 60: symbol 'asm_target_open_p1' not defined
Error at file 'fcntl/z80/asm_vopen.asm' line 127: symbol 'asm_target_open_p2' not defined
make: *** [Makefile:101: htlist] Error 1

jayacotton, 2021-04-14:
And I see that I hit this before.  If the file i/o situation is not changed, then its back to the classic library for me.

feilipu, 2021-04-15:
Yes, it may be some time before file I/O is implemented in newlib so classic is the only solution.

The issue has been fixed, and you're aware of the floating point library snare should you encounter it. So I guess this issue is closed?

jayacotton, 2021-04-15:
The only  existing issue (that I know of now) is deploying the fix.  I cloned the repo,  and built it, what library should I be looking for ?

feilipu, 2021-04-15:
Commits flow into the master branch and are a) picked up by the nightly build (and so should be ready now), or b) if building from source then just you just git pull in your clone and rebuild as normal.

jayacotton, 2021-04-15:
In that case, unless I find a new z180 issue, I will close this one.

closed 2021-04-15 by jayacotton