z88dk/z88dk #1760: Can't run 'hello world' on Z80, works on Z180

jayacotton, 2021-04-16:
I pulled the latest nightly source and built it on my linux box. 
zcc - Frontend for the z88dk Cross-C Compiler - v18239-ee5024b80-20210415

Preliminary attempt to debug, 
ddtz test.com
g,2f9  <<<  this is main.

Never gets there. 

I wonder if one of the changes we made to stdlib translates into Z80 no go.

Here is the code:
#include <stdio.h>

void main()
{
	printf("Hello World\n");
}
build commands
jay@jay-Ultra-27:~/projects/inettools-z80$ make test
zcc +cpm --list --c-code-in-asm  -Wall -pragma-include:zpragma.inc  -DFRAM -c test.c
zcc +cpm -Cl -v -create-app -m -pragma-include:zpragma.inc -DAMALLOC2 -otest test.o
zpragma.inc
#pragma output REGISTER_SP = 0xc000
#pragma output USING_amalloc
#pragma printf "%x %s %02d %c %u"
#pragma scanf "%x %s %02d %c %u"

suborb, 2021-04-16:
@jayacotton I'm not having any problems running a hello world under z80 or z180 so I'm a bit baffled as to what's going on.

Could you attach the .map, .lis and .com files and I can have a poke around?

feilipu, 2021-04-16:
Perhaps also try with this little file, and not using your own makefile.
Just paste the command line. `zcc` will look after everything else

```C

#include <stdio.h>

#pragma printf = "%x %s %d %c %u" // enables %x %s %d %c %u only

// for cpm target using sccz80 C compiler and classic library, run hello.com
// zcc +cpm hello.c -o hello.com

// or more verbosely
// zcc +cpm -v -m --list hello.c -o hello.com

int main(void)
{
    fprintf(stdout,"Hello World R\nHello World C\nHello World 2\nHello World 0\nHello World 1\nHello World 4\n");
    return 0;
}
```
My suspicion is that perhaps the make file may be missing a define that is necessary.
There are [quite a few needed](https://github.com/z88dk/z88dk/blob/master/lib/config/cpm.cfg#L10).

jayacotton, 2021-04-16:
Found the bug.  Its in my http loader program.  I think its not loading the last record correctly.  Need to work on that.
Thanks for your quick response on this.  It does not look like there is a problem with the latest z88dk build.

closed 2021-04-16 by jayacotton