z88dk/z88dk #1761: Did something change WRT starting main ?

jayacotton, 2021-04-18:
I am finding that code that has been running for a few years it now starting to fail after rebuilding.

I am seeing an issue with argc,argv on startup in main.

C>wget http://server ping.com
argc 4
argv[0] 
argv[1] 
argv[2] HTTP://SERVER
argv[3] PING.COM

This is a example of the issue.  It looks like there is one extra slot used in the argv list.
I.e.  this command wget http://server ping.com should decode to

argc 3
argv[0] wget
argv[1]  http://server
argv[2] ping.com    (all in upper case of course)

This how my programs are coded, and what I have been expecting to see.  Lately this has not 
held true.  Wondering if I missed something.

Here is a code snip
int
main (int argc, char *argv[])
{
  struct sockaddr_in sin;
  struct hostent *h;
  uint16_t port = 80;
  char *pp;
  unsigned char *p;
  char *fp;
  int i;

  int of;
  int code;
  int len;
  uint8_t looped = 0;
  int stat;

  skip_dns = 0;
printf("argc %d\n",argc);
printf("argv[0] %s\n",argv[0]);
printf("argv[1] %s\n",argv[1]);
printf("argv[2] %s\n",argv[2]);
printf("argv[3] %s\n",argv[3]);

part of the build log
zcc +cpm -Wall -pragma-include:zpragma.inc  -DFRAM -c htget.c
htget.c:86:2: warning: Converting type: const char * p to char * s [-Wincompatib
le-pointer-types]
htget.c:294:9: warning: Variable 'pp' may be used before initialisation [-Wmaybe
-uninitialized]
htget.c:297:3: warning: Variable 'pp' may be used before initialisation [-Wmaybe
-uninitialized]
zcc +cpm -create-app -pragma-include:zpragma.inc -DAMALLOC2  -owget htget.o w5500.o dhcp.o spi.o socket.o ethernet.o dns.o wizchip_conf.o time.o -lm libinet.o libhost.o

suborb, 2021-04-18:
That is an odd one, it looks like changed back in February and accidentally took into account the length flag - hence argv[1] being gibberish.

argv[0] will always be an empty string BTW.

jayacotton, 2021-04-18:
I'll try the build tonight .. after 12pm gmt.

jayacotton, 2021-04-19:
I down loaded the nightly  z88dk-win32-latest.zip | 2021-04-19 03:13 | 55M |  and installed it.
I then rebuilt all my code.  

It seems to be working now, as described in K&R second edition, page 115
tnx

closed 2021-04-19 by jayacotton