z88dk/z88dk #1762: [c128][vic-ii] Graphics no longer works (at least not as before)

Fabrizio-Caruso, 2021-04-18:
VIC-II UDGs no longer work on the C128 target.

I am compiling with:
```
c128_z80_40col: $(ASSETS_PATH)/z88dk_xchase.asm 
	$(Z88DK_PATH)$(MYZ88DK) +c128 $(SCCZ80_OPTS)  \
	$(INCLUDE_OPTS) \
	-lndos -subtype=disk \
...
	$(FULL_FILES) \
	$(ASSETS_PATH)/z88dk_xchase.asm \
	$(CROSS_LIB_PATH)/display/init_graphics/z88dk/udg/udg_init_graphics.c \
	-create-app -oa40.bin
...
```

suborb, 2021-04-19:
I get this using my little test app:

[image]

suborb, 2021-04-19:
I think what's happening is that when you compile -clib=gencon then we do this;

        ld      a,$8c   ;Character generator to $3000
        ld      bc,$d018
        out     (c),a

But with -clib=ansi we don't

commit b0735553 (2021-04-19): Fix #1762: Set the character set address when we set a font
closed 2021-04-19 by suborb