z88dk/z88dk #1764: Support of Windows XP (now zcc.exe doesn't work under it)

Oleg-N-Cher, 2021-04-21:
Once upon a time, Alvin and I were faced with the fact that the zsdcc.exe does not work under Windows XP. At that time he corrected this point and said that it was solved by switching on an appropriate linker option. I request this feature for zcc.exe again (ideally - for all .exe). If it's not too difficult.

Thank you so much for your work. Very impressed, especially with the support for the 8080 processor.

suborb, 2021-04-21:
The Windows binaries have been 64 bit for a while: https://z88dk.org/forum/viewtopic.php?f=18&t=11381

I'm not sure the mingw toolchain the nightlies are cross-compiled with has all the right dependencies for generating 32 bit binaries.

Oleg-N-Cher, 2021-04-21:
Hmm... then I'm confused by the "Win32" in the archive name "z88dk-win32-latest.zip" at http://nightly.z88dk.org
Tell me plz, would you consider switching back to Win32 target entirely?

The fact is that I myself am a translator developer, and I came to the understanding that 64-bit code would have advantages if it was necessary to address > 4Gb of RAM, for example. What else did Christian Gisler, the author of Total Commander, write about at the time. That he was asked for the 64-bit version of Total, and he understands perfectly well that there will be no advantages from 64-bitness. And so he went to meet the wishes of users not for technical reasons.

I really ask you to download the 32-bit MinGW toolchain and build only by it. There are no advantages from 64-bitness, only loss of compatibility with 32-bits. I really don't think 32-bit windows is used that much (less and less recently). But sometimes we still need to.

And I even made Windows 95 support in my translator. And if it was necessary, then Win 3.1+Win32s would support too. This expands its applicability, especially since we are all fans of retro systems. :)

And yesterday, one user asked me for WinXP support, and my Oberon-to-C translator works on his machine, but z88dk does not.

Thank you for your understanding!

suborb, 2021-04-21:
There's a lot to unpack there!

For development machines, I think there's a natural tendency for people to keep up-to-date and 32 bit Windows is now not available for OEMs - the only place I've seen a 32 bit Windows install in that time is on vmware instance  that had to use VS2005 for compatibility reasons.

However, I've temporarily(*) adjusted the nightly build to put a set of 32 bit binaries into a bin.x86 directory so with a quick rename they are available. I'm a bit loathed to switch them over to the primary versions since they are completely untested.

(*) temporarily....

1. Ideally there would some sort of "Portable Executable" format containing multiple architectures (x86, x86-64, arm)
2. I need to something about the size of the zip files, I burning through the best part of a gigabyte a week with the nightlies.
3. Given the low usage of 32 bit windows, it might be worth just building a (fully formed) nightly once a week or so

Oleg-N-Cher, 2021-04-22:
Thank you for meeting us halfway!

I think the z88dk for Windows can always be 32-bit. Because the continuity of compatibility of 32-bit code on 64-bit Windows systems is transparently achievable, and users will not even feel much of a difference. And you don't have to have 32-bit Windows on your development machine at all, just 32-bit MinGW, and that only if 64-bit MinGW can't do the work (with the option -m32).

I myself am willing to make sacrifices, especially since the developer's host system must be flexible. So I personally have different Linuxes installed, and different versions of Windows (from XP to 10), and even macOS. But that my user very sincerely does not want to change his 32-bit Windows to another version.

By the way, I have several versions of MinGW installed simultaneously on my development machine: TDM-GCC (works under 32-bit, but can generate 64-bit code even on 32-bit machine), different versions of MinGW 32-bit and MinGW 64-bit. They do not interfere with each other, I just add the version of MinGW I need to the PATH using the building .bat scripts.

feilipu, 2021-04-22:
> I need to something about the size of the zip files, I burning through the best part of a gigabyte a week with the nightlies.

Just an unrelated thought about the nightlies, which age as well as yesterday's newspaper. Given their only "next-day" use is to bisect an issue and github enables precise granularity builds quite easily... and commits within a week are pretty easy to eyeball anyway.

Would it be sensible to just keep the nightlies for 7 days, and promote one per week to be a "weekly"?

That would cut the disk usage substantially.

suborb, 2021-05-02:
Yeah, I think that's probably the way to go, just got to figure out the correct find magic and I'll be able to prune weekly.

I think this issue is now dealt with as is, so I'm closing.

closed 2021-05-02 by suborb