z88dk/z88dk #1768: (sdcc) Codegen issues

suborb, 2021-04-29:
Reported on the forum: https://z88dk.org/forum/viewtopic.php?f=12&t=11519

```
#include <stdio.h>

typedef unsigned long   uint32;
typedef unsigned short  addr16;
typedef unsigned short  uint16;
typedef unsigned char   uint8;

uint8* stack = 0;
addr16 stackTop = 0;

static uint32 stackPeek() {
    uint8* location = stack + stackTop;
    return *((uint32*)location);
}

static uint32 stackPop() {
    uint32 value = stackPeek();
    stackTop += 4;
    return value;
}

void main() {
    stackPop();
    printf ("%x ", stackTop);
    stackPop();
    printf ("%x ", stackTop);
    stackPop();
    printf ("%x ", stackTop);
    stackPop();
    printf ("%x ", stackTop);
    stackPop();
    printf ("%x ", stackTop);
}
```

Compile with: `zcc +zx -vn -clib=sdcc_iy -startup=0 Bug1.c -o bug -create-app`

Expected to see "4 8 c 10 14" but got "404 808 c0c 1010 1414"

Note: using --max-allocs-per-node0 options fixes the problem

Second issue:

```
#include <stdio.h>

typedef unsigned long   uint32;
typedef unsigned short  addr16;
typedef unsigned short  uint16;
typedef unsigned char   uint8;

uint8* memory = 0;

void memWrite(uint32 sectorNumber){
    uint16 value = 200;
    uint32* byte51202 = (uint32*)(memory + 51202);
    *byte51202 = sectorNumber;
    memory[51210]=100;
    memory[51212]=value;
}

void main() {
    memWrite(0);
    printf ("%u ", memory[51210]);
    printf ("%u ", memory[51212]);
}
```

Compile with: `zcc +zx -vn -clib=sdcc_iy -startup=0 Bug2.c -o bug -create-app`

Expected to see "100 200" but got "0 0"

Using --max-allocs-per-node0 option did not help, output was "0 200"

suborb, 2021-04-29:
Second one looks like a --reserve-regs-iy bug:

```
dom@heliotrope:~/z88dk:250% zcc +test stk.c -compiler=sdcc --reserve-regs-iy
dom@heliotrope:~/z88dk:251% z88dk-ticks a.bin
0 0
Ticks: 28129
dom@heliotrope:~/z88dk:252% zcc +test stk.c -compiler=sdcc
dom@:~/z88dk:253% z88dk-ticks a.bin
100 200
Ticks: 55893
dom@:~/z88dk:254%
```

suborb, 2021-04-29:
Issue isn't seen in latest upstream sdcc so time for another upgrade...

I suspect it was fixed by r12245 or r12246 in sdcc

feilipu, 2021-04-29:
Ok. I'll try to get a clean PR for latest release up soonest.

feilipu, 2021-04-30:
Might need to rework this... It wasn't qute fixed in r12250.
https://sourceforge.net/p/sdcc/bugs/3217/#8cf6

feilipu, 2021-05-04:
Tested (on CP/M) that Bug1 is resolved. :+1: 
But Bug2 remains, AFAIK. :-1:

feilipu, 2021-05-04:
@spth

```
0000              _memWrite:
0000  DD E5       	push	ix
0002  21 00 00    	ld	hl,_memory
0005  7E          	ld	a, (hl)
0006  C6 02       	add	a,0x02
0008  5F          	ld	e, a
0009  23          	inc	hl
000A  7E          	ld	a, (hl)
000B  CE C8       	adc	a,0xc8
000D  57          	ld	d, a
000E  21 04 00    	ld	hl,4
0011  39          	add	hl, sp
0012  01 04 00    	ld	bc,0x0004
0015  ED B0       	ldir
0017  2B          	dec	hl                   <<<<  I a problem, where compiler has not realised that HL was changed by LDIR.
0018  7E          	ld	a, (hl)
0019  C6 0A       	add	a,0x0a
001B  4F          	ld	c, a
001C  23          	inc	hl
001D  7E          	ld	a, (hl)
001E  CE C8       	adc	a,0xc8
0020  47          	ld	b, a
0021  3E 64       	ld	a,0x64
0023  02          	ld	(bc), a
0024  2B          	dec	hl
0025  7E          	ld	a, (hl)
0026  C6 0C       	add	a,0x0c
0028  4F          	ld	c, a
0029  23          	inc	hl
002A  7E          	ld	a, (hl)
002B  CE C8       	adc	a,0xc8
002D  47          	ld	b, a
002E  3E C8       	ld	a,0xc8
0030  02          	ld	(bc), a
0031  DD E1       	pop	ix
0033  C9          	ret
```

spth, 2021-05-04:
Thanks. I can reproduce the memWrite issue on upstream sdcc -mz80 --reserve-regs-iy, and opened a ticket: https://sourceforge.net/p/sdcc/bugs/3223/

feilipu, 2021-05-04:
Philipp has a fix in r12284. TBC.

PR #1774 (2021-05-05): zsdcc - update to r12288 for #1768
Confirmed that #1768 Bug2 has been fixed.

Closes #1768.

Third time's a charm.