z88dk/z88dk #1772: (sam) (classic) Support for MODE3 'Fatpix"

dandoore, 2021-04-30:
SAM MODE 3 allows for pixels to be made double-wide via the FATPIX function, allowing horizontal and vertical pixels have the same thickness.

When FATPIX 1 is enabled the XRG (part of the graphics scaling subsystem's X range) is halved from 512 to 256, FATPIX 0 restores XRG to 512 note that the screen is not cleared or mode changed as per the attached:

[image]

Use case is the conversion of 64 column programs with graphics from the ZX platforms to use MODE3 (in this case, [AITTT](http://www.marcellozaniboni.net/zxaittt/index.html) , where 64 column native MODE3 text is used but where the horizontal resolution is at 256 for graphics

suborb, 2021-04-30:
I've added `sam_set_fatpix(int)` into `<arch/sam.h>` which should do what you want.

dandoore, 2021-04-30:
Cool! Will check it out on the nightly :)

dandoore, 2021-05-01:
Awsome, works like a charm - ignore the stuff on screen I haven't fixed yet :)

[image]

closed 2021-05-01 by dandoore