z88dk/z88dk #1775: zx spectrum dot files, wrong start address

varmfskii, 2021-05-05:
dot files produced by the current nightly (as of 2 May 2021) are linked as if the start address is $8000. For dot files, this start address should be $2000,
[showsimg.zip](https://github.com/z88dk/z88dk/files/6428065/showsimg.zip)

aralbrec, 2021-05-06:
Even after this fix the startup code is not being processed correctly and the org address is not properly read from the build type.

I believe all newlib compiles have been broken by the recent-ish change that does not copy an m4 processed file into the current working directory.   newlib uses a master crt for each target to decide which particular crt to select and with which options.  For the zxn target it is this one https://github.com/z88dk/z88dk/blob/master/libsrc/_DEVELOPMENT/target/zxn/zxn_crt.asm.m4 .  This file is then processed into an asm file which is picked up in the linking step by z80asm.  So I think the asm file is missing and a no crt file compile is performed.

I will take a look when I can.

varmfskii, 2021-05-07:
Thanks, I have managed a temporary workaround to allow the creation of working dot files. If you pass the 0x2000 start address at link time it works. It is a little ugly, but it is a temporary solution.

This e-mail message (including any attachments) is for the sole use of
the intended recipient(s) and may contain confidential and privileged
information. If the reader of this message is not the intended
recipient, you are hereby notified that any dissemination, distribution
or copying of this message (including any attachments) is strictly
prohibited.

If you have received this message in error, please contact
the sender by reply e-mail message and destroy all copies of the
original message (including attachments).

suborb, 2021-05-07:
I'm not at all convinced about what's being posited:

Latest:

```
% zcc +zxn world.c -subtype=dot -clib=new -m -v
...
m4 -I "/Users/dom/z88dk/lib/config//../..//libsrc/_DEVELOPMENT/target/zxn" -I "/tmp/tmpzccRhydSvkH" -I "/Users/dom/z88dk/lib/config//../..//src/m4"  --define=__STARTUP_OFFSET=256  < "/Users/dom/z88dk/lib/config//../..//libsrc/_DEVELOPMENT/target/zxn/zxn_crt.asm.m4" > "/tmp/tmpcfnFfN1I.asm"
cat "/tmp/tmpcfnFfN1I.asm" >> "/tmp/tmpcfnFfN1I.tmp"
cat "/tmp/tmpcfnFfN1I.tmp" >> "/tmp/tmpcfnFfN1I.asm"
WILL ACT AS CRT

...
z88dk-z80asm -mz80n -b -d  -o"a.bin" -m -L.  -L"/Users/dom/z88dk/lib/config//../..//libsrc/_DEVELOPMENT/lib/sccz80"   -D__ESXDOS_DOT_COMMAND -I"/Users/dom/z88dk/lib/config//../..//libsrc/_DEVELOPMENT/target/zxn" -I"/tmp/tmpzccRhydSvkH" -D__SCCZ80 -lzxn    "/tmp/tmpcfnFfN1I.o" "/tmp/tmpETfHaJET.o"

% grep __Start a.map
__Start                         = $2000 ; addr, public, , zxn_crt_asm_m4, CODE, /Users/dom/z88dk/lib/config//../..//libsrc/_DEVELOPMENT/target/zxn/zxn_crt.asm.m4:9389

% z88dk-dis -x a.map -o 0x2000 a_CODE.bin | head -20
__Start:
                    ld      l,c                             ;[2000] 69
                    ld      h,b                             ;[2001] 60
                    push    iy                              ;[2002] fd e5
                    exx                                     ;[2004] d9
                    push    hl                              ;[2005] e5
                    ld      (__sp),sp                       ;[2006] ed 73 ad 20
                    rst     $08                             ;[200a] cf
                    adc     b                               ;[200b] 88
                    ld      hl,$2089                        ;[200c] 21 89 20
                    jp      c,error_crt                     ;[200f] da 67 20
                    ld      hl,$207b                        ;[2012] 21 7b 20
                    ld      (_esx_errh),hl                  ;[2015] 22 55 38
                    rst     $08                             ;[2018] cf
                    sub     l                               ;[2019] 95
                    exx                                     ;[201a] d9
                    ld      a,h                             ;[201b] 7c
                    or      l                               ;[201c] b5
                    jr      nz,cmdline_present              ;[201d] 20 0d
cmdline_missing:

```

1.99:

```
% zcc +zxn world.c -subtype=dot -clib=new -m -v
..
m4 -I "/Users/dom/1.99/z88dk/lib/config//../..//libsrc/_DEVELOPMENT/target/zxn" -I "/Users/dom/1.99/z88dk/lib/config//../..//src/m4"  --define=__STARTUP_OFFSET=256  < "/Users/dom/1.99/z88dk/lib/config//../..//libsrc/_DEVELOPMENT/target/zxn/zxn_crt.asm.m4" > "/tmp/tmp0LekyssH"
cat "/tmp/tmp0LekyssH" >> "/Users/dom/1.99/z88dk/lib/config//../..//libsrc/_DEVELOPMENT/target/zxn/zxn_crt.asm"
cat "/Users/dom/1.99/z88dk/lib/config//../..//libsrc/_DEVELOPMENT/target/zxn/zxn_crt.asm" >> "/tmp/tmp0LekyssH.asm"
cat "/tmp/tmp0LekyssH.asm" >> "/tmp/tmp0LekyssH.tmp"
cat "/tmp/tmp0LekyssH.tmp" >> "/tmp/tmp0LekyssH.asm"
WILL ACT AS CRT
...

z80asm -D__ESXDOS_DOT_COMMAND -I"/Users/dom/1.99/z88dk/lib/config//../..//libsrc/_DEVELOPMENT/target/zxn" -D__SCCZ80  -s  -mz80-zxn -I"/Users/dom/1.99/z88dk/lib/config//../..//lib" -I"/Users/dom/1.99/z88dk/lib/config//../..//libsrc/_DEVELOPMENT/target/zxn"  "/tmp/tmp0LekyssH.asm"

z80asm -mz80-zxn -b -d  -o"a.bin" -m -L.  -L"/Users/dom/1.99/z88dk/lib/config//../..//libsrc/_DEVELOPMENT/lib/sccz80"   -D__ESXDOS_DOT_COMMAND -I"/Users/dom/1.99/z88dk/lib/config//../..//libsrc/_DEVELOPMENT/target/zxn" -D__SCCZ80 -izxn    "/tmp/tmp0LekyssH.o" "/tmp/tmpC2eb9K6j.o"

% !grep
grep __Start a.map
__Start                         = $2000 ; addr, public, , zxn_crt_asm, CODE, /Users/dom/1.99/z88dk/lib/config//../..//libsrc/_DEVELOPMENT/target/zxn/zxn_crt.asm:9389

% z88dk-dis -x a.map -o 0x2000 a_CODE.bin | head -20
__Start:
                    ld      l,c                             ;[2000] 69
                    ld      h,b                             ;[2001] 60
                    push    iy                              ;[2002] fd e5
                    exx                                     ;[2004] d9
                    push    hl                              ;[2005] e5
[... 586 lines cut ...]
> 176   0000              
> 177   0000              ENDIF
> 178   0000              
> 179   0000              ;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;
> 180   0000              
15982,15991c15936,15940
< 34    0059                  IF (__crt_model = 2)
< 35    0059                    EXTERN   asm_dzx7_standard
< 36    0059                    call nz, asm_dzx7_standard
< 37    0059                  ELIF (__crt_model = 3)
< 38    0059                    EXTERN   asm_dzx0_standard
< 39    0059                    call nz, asm_dzx0_standard
< 40    0059                  ENDIF
< 41    0059              
< 42    0059                 ENDIF
< 43    0059              
---
> 34    0059                    EXTERN   asm_dzx7_standard
> 35    0059                    call nz, asm_dzx7_standard
> 36    0059              
> 37    0059                 ENDIF
> 38    0059              
```

Which seem like the right changes

varmfskii, 2021-05-08:
I am compiling with -clib=sdcc_iy, rather than newlib. Looking at the raw disassembly I included, you can see code which immediately jumps outside of the permissible range for a dot file ($2000-$4000).
I compile a set of separate objects with:
zcc +zxn -clib=sdcc_iy -SO3 --max-allocs-per-node200000 --opt-code-size -c showsimg.c
zcc +zxn -clib=sdcc_iy -SO3 --max-allocs-per-node200000 --opt-code-size -c error.c
zcc +zxn -clib=sdcc_iy -SO3 --max-allocs-per-node200000 --opt-code-size -c getext.c
...
zcc +zxn -clib=sdcc_iy -SO3 --max-allocs-per-node200000 --opt-code-size -c writestate.c

Then link them all together with:
zcc +zxn -Cz="--clean" -create-app -subtype=dot -startup=30-clib=sdcc_iy -SO3 --max-allocs-per-node200000 --opt-code-size -o showsimg showsimg.o error.o getext.o getopts.o help.o noattr.o readstate.o remap.o restorestate.o savestate.o setimage.o setpalette.o setstate.o writestate.o
which will create code that looks as if it should be starting at $8000 even though it is loaded at $2000
however if I link with:
zcc +zxn -Cz="--clean --org=0x2000" -create-app -subtype=dot -startup=30-clib=sdcc_iy -SO3 --max-allocs-per-node200000 --opt-code-size -o showsimg showsimg.o error.o getext.o getopts.o help.o noattr.o readstate.o remap.o restorestate.o savestate.o setimage.o setpalette.o setstate.o writestate.o
instead I get a working dot file that is properly linked to load at $2000

Presumably, the issue is in z88dk-appmake since the workaround is to pass a parameter to appmake.

suborb, 2021-05-08:
GitLab is currently down, but I have an older checkout of your project.

I can see that the listings you provided are garbage and the first few bytes ('prologue') look completely wrong.

So, to sanity check, I've compiled with 1.99 and (separately) with the latest code that's committed (so including https://github.com/z88dk/z88dk/commit/86d25db3f0d9d2ac2ec220873eaaf88acc7b3d0c above), and the code prologue is the same with addresses in a sensible range (i.e. 0x2000 -> 0x4000)

I've also compared against the the checked in DOT commands within the z88dk repo and again the prologue is sensible and all addresses are in the correct 0x2000 range.

This prologue also matches up with a compilation performed by the classic library dot file target.

varmfskii, 2021-05-08:
It may have been resolved by something. At the time I pulled teh listings, thataccurately reflected teh output. It no longer does and what I am seeing on my end does indeed look correct.

Thank You for your efforts.

So, to sanity check, I've compiled with 1.99 and (separately) with the latest code that's committed (so including 86d25db<https://nam11.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2Fz88dk%2Fz88dk%2Fcommit%2F86d25db3f0d9d2ac2ec220873eaaf88acc7b3d0c&data=04%7C01%7Ctheodore.alexander.evans%40emory.edu%7Ccb1867be9aca4fe462e908d91204a6b2%7Ce004fb9cb0a4424fbcd0322606d5df38%7C0%7C0%7C637560633468302142%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C1000&sdata=TD3cMplTGAmB4nZ41AUJVjqcU5NkEteheM4uIyTThRM%3D&reserved=0> above), and the code prologue is the same with addresses in a sensible range (i.e. 0x2000 -> 0x4000)

suborb, 2021-05-08:
> It may have been resolved by something. At the time I pulled teh listings, thataccurately reflected teh output. It no longer does and what I am seeing on my end does indeed look correct. Thank You for your efforts.

No worries, it was the fix above that resolved it - a continuation of #1746 where I focussed on the size not the actual code being generated.

commit 86d25db3 (2021-05-05): Fix #1775: Related to #1746 - I only tested with decimal values.

But of course, hex is used somewhere...
closed 2021-05-08 by suborb