z88dk/z88dk #1778: rc2014 - SIO/2 failure

feilipu, 2021-05-09:
Documenting a "bit rot" issue. For some reason the RC2014 SIO/2 drivers are no longer working.
This has consumed two days, and I'm still no closer to identifying what is going wrong.

- hardware checks out as ACIA builds work using the base RC2014,
- and RomWBW code using the SIO/2 module work.
- compilers check out, as being the same sccz80 and sdcc produce the same faulty result.
- crt_0 files check out and the necessary configuration for the IM2 vectors seems correct.
- line by line review of the listing from a simple `hello_world.c` shows nothing obvious (for either sccz80 or zdcc)

The only thing remaining AFAIK, is that somehow the actual drivers have become broken, but they [haven't been touched for years](https://github.com/z88dk/z88dk/commits/master/libsrc/_DEVELOPMENT/target/rc2014/device/sio).

I'm at a loss, but will keep looking.

suborb, 2021-05-09:
I think this was in 1.99 - is it working there?

feilipu, 2021-05-09:
Good idea.
Next step find where `hello_world.c` last worked.

feilipu, 2021-05-14:
I suspect that it is still not quite right.

Ordered a new SIO/2 Module to rule out a weird hardware failure (that works with RomWBW but not for z88dk IM2). Possible but unlikely.

Otherwise the symptoms look like an unbalanced stack in SIO device driver, but that's hardly likely either.

feilipu, 2021-05-15:
Something is broken with the Tx Interrupt logic. But I don't believe that code can just "fail", so probably I've damaged my SIO/2 device. Pending a new device.

closed 2021-05-26 by feilipu