z88dk/z88dk #1779: Dockerfile doesn't delete the build_deps correctly.

rdebath, 2021-05-09:
The `.build_deps` are installed in one `RUN` command and deleted in the next one. This means that they are in different layers and the packages installed still take up space in the final image.

If you merge the two commands like this the total image is 283MB smaller. (About 1/3rd smaller)
```
RUN apk add --no-cache build-base libxml2 m4 \
    && apk add --no-cache -t .build_deps bison flex libxml2-dev git subversion boost-dev texinfo \
    && git clone --depth 1 --recursive https://github.com/z88dk/z88dk.git ${Z88DK_PATH} \
    && cd ${Z88DK_PATH} \
    && chmod 777 build.sh \
    && ./build.sh \
    && rm -fR ${SDCC_PATH} \
    && apk del .build_deps
```

```
IMAGE ID      SIZE    REPOSITORY:TAG                   CREATED
4f73caed5b88  592MB   testimage:latest                 About a minute ago
eac87f4de0b5  875MB   z88dk/z88dk:latest               3 months ago
```

suborb, 2021-05-10:
I think this was originally by design to allow people to override configuration.

However, I suspect that people aren't actually doing that so I'll comment out that bit out and replace with your snippet above.

Thank you!

rdebath, 2021-05-13:
If I wanted to allow for that configuration I'd probably change the call to `build.sh` so you can give it arguments from a Docker "build-arg" and then add script options to move the data from a copy command location.

suborb, 2024-04-28:
Looks like I finally did this here: https://github.com/z88dk/z88dk/blob/master/z88dk.Dockerfile

closed 2024-04-28 by suborb