z88dk/z88dk #1780: [vz200] won't compile and gives: FATAL Compiler Internal Error in file 'gen.c

Fabrizio-Caruso, 2021-05-09:
If I compile Cross Snake for the vz200 I get:
```
games/snake/main.c:1237: warning 126: unreachable code
games/snake/main.c:1237: error 9: FATAL Compiler Internal Error in file 'gen.c'
line number '7196' : code generator internal error
Contact Author with source code
make: *** [Makefile.snake:1653: vz200_24k] Error 1
```

spth, 2021-05-10:
If you can provide a small code sample to reproduce the issue, I suggest to report the bug upstream (https://sourceforge.net/p/sdcc/bugs/).

Fabrizio-Caruso, 2021-05-10:
@spth I have no idea how to reproduce this problem with a smaller example. The compilation does not fail on all my games. It was not failing on the previous version of ZSDCC. So the update broke something.
Moreover SDCC (in Z88DK) was and still is very unstable even when it can produce a binary.
I have opened issues about SDCC. I always always use SCCZ80 as it is more reliable.

Fabrizio-Caruso, 2021-05-16:
I have opened:
https://sourceforge.net/p/sdcc/bugs/3232/
No much reaction from the devs.

Fabrizio-Caruso, 2021-05-16:
The line that breaks (z)sdcc is:
`level_bonus = (uint16_t) (((uint16_t) snake_length)<<1)+(((uint16_t) energy)<<3) +(((uint16_t) coin_count)<<5) + (((uint16_t) level)<<2); `

feilipu, 2022-02-13:
Is this one fixed now?
I didn't try snake with #1503 and sdcc.

spth, 2022-02-13:
The upstream sdcc bug report is still open (since AFAIK upstream doesn't have a small sample to reproduce the issue yet).
Since SDCC is lacking in interprocedural optimizations, it should be possible to reproduce nearly any issue in sdcc when compiling a single C function.

feilipu, 2022-02-13:
Sorry @spth was asking whether the issue was still reproducible in the source.

Fabrizio-Caruso, 2022-02-13:
@spth I did find the line that the compiler cannot handle but, if I remember well, the problem does not appear if only that line is used. I have try to find some smaller code that shows the issue.

@feilipu snake is now fixed thanks to your report (dos2unix fixed it).

Fabrizio-Caruso, 2022-02-13:
@feilipu @spth I confirm the issue is still present.

It can be reproduced by:
(0) have `python` and GNU `make` installed
(1) `git clone https://github.com/Fabrizio-Caruso/CROSS-LIB`
(2) enter `src` directory
(3) run `xl snake vz200_zsdcc` or equivalently `make zsdcc -f games/snake/Makefile.snake`

The problem is somehow in 
```
level_bonus = (uint16_t) (((uint16_t) snake_length)<<1)+(((uint16_t) energy)<<3) +(((uint16_t) coin_count)<<5) + (((uint16_t) level)<<2); 
```
but I am not able to build a significantly smaller failing example because it won't fail but removing the line above fixes in my game.

feilipu, 2022-02-14:
I'm getting something, but not the compiler error.

In `sleep_macros.c` the [`_XL_SLEEP()`](https://github.com/Fabrizio-Caruso/CROSS-LIB/blob/master/src/cross_lib/sleep/sleep_macros.c#L41) function has no decoration, and in [`sleep_macros.h`](https://github.com/Fabrizio-Caruso/CROSS-LIB/blob/master/src/cross_lib/sleep/sleep_macros.h#L54) it is defined as `sleep()`.
This conflicts with the standard definition of [`sleep() __z88dk_fastcall`](https://github.com/z88dk/z88dk/blob/master/include/stdlib.h#L305) found in `stdlib.h`.

This needs to be adjusted, and then we can look further.

Fabrizio-Caruso, 2022-02-14:
I don't understand the conflict. Do I redefine sleep? My own sleep can be different from standard sleep. Is the fat call tag important? I cannot put there unless I protect it with some directives to avoid it when using other compilers.

feilipu, 2022-02-14:
> I don't understand the conflict. Do I redefine sleep? My own sleep can be different from standard sleep.

The `stdlib.h` definition of sleep can't be redefined, so best to give your own sleep function another name, then you're free to use any definition. But, if you want to use the z88dk platform sleep, then your macro definition has to match the z88dk `stdlib.h` definition.

> Is the fast call tag important? I cannot put there unless I protect it with some directives to avoid it when using other compilers.

The `__z88dk_fastcall` decoration means that the parameter is passed in registers, rather than on the stack. Yes it is important that the definitions match. If you don't want to use the decoration, then you should use another function name for your `sleep()` function, and all will be good.

Fabrizio-Caruso, 2022-02-14:
Sorry, I got it now. It is a mistake on my side. If _XL_SLEEP is defined as sleep in the header than I should not redefine it in the .c file and the definition signature has to match the one used for sleep.
This seems unrelated to the issue here, though, but it is good to remove this additional problem.

Fabrizio-Caruso, 2022-02-14:
Do you mean that you no longer get the compilation error after fixing the sleep function ?

feilipu, 2022-02-14:
> Do you mean that you no longer get the compilation error after fixing the sleep function ?

Yes, after I changed the `sleep()` function to match the definition, then no more error, and the binaries appear in the `/build` directory as they should. I don't know whether they're working or not though. Not sure how to emulate the machine.

But, anyway, I haven't been able to produce the error in `gen.c`. so IDK :man_shrugging:

suborb, 2022-02-14:
Use vzem to emulate.

feilipu, 2022-02-14:
> I don't know whether they're working or not though.

Tested with vzem and snake is working as expected. 👍

Fabrizio-Caruso, 2022-02-14:
@feilipu for some reason if I avoid the (stupid) redefinition of sleep, it compiles fine for me and after passing some more paramters to fix the iy register and the keyboard, it works fine.
I have tested it with https://nippur72.github.io/laser310-emu/ which is probably the most accurate emulator (by @nippur72).

Fabrizio-Caruso, 2022-02-14:
@feilipu  I am getting the compilation error on any environment (just tested on a Ubuntu VM and Cygwin). In order to reproduce the bug you have to leave the stupid bug of mine with sleep being redefined.
For some arcane reason, the bug is contributing to the error message that seems totally unrelated.

feilipu, 2022-02-14:
> For some arcane reason, the bug is contributing to the error message that seems totally unrelated.

Computer + Compiler = Arcane imho.

Anyway good that it works. Another one closed.

BTW, not sure whether there is a need to avoid `IY` in this machine. It depends on whether it is being used and not preserved. IDK the machines at all to comment.

Fabrizio-Caruso, 2022-02-14:
@feilipu https://nippur72.github.io/laser310-emu/ only requires a browser. No need to use wine.

Fabrizio-Caruso, 2022-02-14:
@suborb, @nippur72 if I use VZEM and I play my games, I see the keys echos on the screen and pollute the screen.
@suborb, which VZEM version are you using?
On @nippur72  emulator I do not see this bug and I see a more precise emulator (snow effect on the screen).
I tend to trust @nippur72 's emulator. Nino, have you tested my games on real hardware?

I have included the binaries of the game Cross Snake compiled with SCCZ80 and ZSDCC.

Concerning this issue, I am closing it as it is "solved" by avoiding redefining sleep (no idea why it solves the issue)
[build.zip](https://github.com/z88dk/z88dk/files/8064150/build.zip)
.

closed 2022-02-14 by Fabrizio-Caruso