z88dk/z88dk #1781: [vz200] Broken binaries (?) Problem with input and/or visible cursor on screen

Fabrizio-Caruso, 2021-05-09:
The vz200 seems to suffer of multiple problems when I compile my Cross-Lib games for it.

For example "make vz200 -f Makefile.chase" builds Cross Chase for the vz200.

I can reproduce the problem(s) with Antonino's emulator, which is, by the way, a very accurate emulator (the only one to reproduce the screen snow effect): https://nippur72.github.io/laser310-emu/
which can be used by dragging and dropping the binary.

I have attached two pre-built binaries in a zip file.
[build.zip](https://github.com/z88dk/z88dk/files/6448552/build.zip)

Fabrizio-Caruso, 2021-05-10:
How?
If I add the `--hardware-keyboard` option I get:

```
Error at module 'rocket_c': symbol 'getk' already defined in module 'vz_crt0'
Errors in source file c:/z88dk/\lib\target\vz\classic\vz_crt0.asm:
Error at module 'rocket_c': symbol 'getk' already defined in module 'vz_crt0'
make: *** [makefiles.chase/makefiles_z88dk/Makefile_z88dk_zsdcc:376: vz200_18k] Error 1
```

suborb, 2021-05-10:
I had this diff:

```
--- a/src/makefiles.chase/makefiles_z88dk/Makefile_z88dk_zsdcc
+++ b/src/makefiles.chase/makefiles_z88dk/Makefile_z88dk_zsdcc
@@ -374,13 +374,14 @@ vz200: vz200_18k

 vz200_18k:
        $(Z88DK_PATH)$(MYZ88DK) +vz $(ZSDCC_OPTS)  -vn \
-       -compiler=sdcc \
+       -compiler=sccz80 \^M
+       --hardware-keyboard \^M
        $(INCLUDE_OPTS) \
        -DSOUNDS -D__VZ__ -clib=ansi \
        -DFULL_GAME \
        -DINITIAL_SLOWDOWN=31500 \
        -DCONIO -DNO_INIT_GRAPHICS \
-       -lndos \
+       -lndos -m \^M
        -create-app -Cz--audio -o $(BUILD_PATH)/X$(GAME_NAME)_vz200_18k.vz \
        $(FULL_FILES)
        rm $(BUILD_PATH)/X$(GAME_NAME)_vz200_18k.cas
```

Though I do get your problem if I don't run with sccz80 which is interesting...

Fabrizio-Caruso, 2021-05-10:
@suborb by the way under VZEM emulator (memory set to 18k) I see the cursor:
[image]

Fabrizio-Caruso, 2021-05-10:
Holding shift and pressing keys:
[image]

suborb, 2021-05-10:
It's looks like it's a memory problem, in vzem it fails for 18 + 24k, but works for 34k.

nippur72, 2021-05-10:
I tried XChase 18K in VZEM build 20210204 (newest) and it behaves just like the in-browser emulator: the keyboard is very unresponsive with keypresses registered randomly.

Note: to avoid keys being printed on screen, I loaded the game in VZEM not from .VZ file but from .WAV (which requires a conversion to 22Khz 8 bit mono). This because when a program is loaded from VZ file it is launched without clearing the "immediate mode", so the cursor stays active (this is my I guess).

suborb, 2021-05-10:
I've changed the routine that's called for scanning the keyboard so all it does is scan the keyboard.

The previous routine that was being called was additionally setting up some variables which I think must be being interpreted by the interrupt and printing on screen, doing random things etc - the routine now pretty much matches what --hardware-keyboard was doing.

Now, Chase seems to work with in 18k mode, so presumably the other ones will as well.

Fabrizio-Caruso, 2021-05-12:
@suborb the cursor problem in VZEM is still present and it is probably not a bug as @nippur72 reports VZEM not to show the cursor when the game is loaded through a wav file (which has to be 22khz 8-bit). It has to be with the way VZEM loads vz files. It seems a bug of the emulator.

closed 2021-06-04 by suborb