z88dk/z88dk #1784: [CP/M] (classic): argc, argv[] setup exceeds the argument count

zx70, 2021-05-10:
When an argument is passed to the command line, the argument count overflows and picks a non-existing parameter.
```
#include <stdio.h>

int x;

int main(int argc, char *argv[])
{
	printf("Arg test\n");
	printf("\n");
	
		printf ("argc: %u\n",argc);
		
	for (x=0; x<argc; x++)
		printf ("argv[%u]: %s\n",x,argv[x]);
}
```

When run with no arguments it works as expected, but when if we run:
C> TEST A B C

we get:
```
Arg test

argc: 5
argv[0]:
argv[1]: 
argv[2]: A
argv[3]: B
argv[4]: C
```

zx70, 2021-05-10:
sorry, the issue was local, updating z88dk to the latest nightly build solved the problem

zx70, 2021-05-11:
the culprit seems to be the fancy way I solved a previous bug  <blush> !
After the parameter loop I check if the first argument has been forgotten. .  better not to loose it at all..

zx70, 2021-05-12:
It seems to work well, now.
@suborb,   I used LDDR.. is it ok to have it in the classic CP/M startup code or it will hit the 8080 targets?

suborb, 2021-05-12:
Lddr will be emulated on 8080 targets (I hope!)

zx70, 2021-05-12:
+new option to specify an optional program name for argv[0]
#pragma string name pgmname

suborb, 2021-06-26:
This seems to have broken non-redirected argc, argv handling.

zx70, 2021-06-27:
how frustrating  😖

suborb, 2021-06-27:
Yeah. I think I’m going to revert to the original code since redirection isn’t often used.

zx70, 2021-06-27:
mm ok, just do.   I'll have a crack on the current version and try to fix.

zx70, 2021-06-30:
oddly I couldn't reproduce the problem on my current copy.   How did you spot it ?

suborb, 2021-06-30:
Came out of this comment here: https://z88dk.org/forum/viewtopic.php?p=19715&sid=cb29a6710f1f13ba68956cdb30582c44#p19715

argv[1] ended up being garbage for CP/M. For ticks argv[1] had the first character cut off.

All a bit odd...

zx70, 2021-06-30:
thanks for explaining,  I was checking argc only

feilipu, 2026-11-08:
Fixed in 72b1c6ee91 (`(crt0) Improve command line parsing a little`).

Same change as #2063. `cpm_crt0.asm` NUL-terminates the DMA tail, and `crt_command_line.inc` drops each redirect from the count, so one redirect no longer invents an extra argument.

closed 2026-11-08 by feilipu