z88dk/z88dk #1785: (sam) Current screen mode not respected when calling code

dandoore, 2021-05-11:
When CALLing the code for a SAM compiled program the current screen mode is not respected as it defaults to MODE4.

Can this please be changed to probe the current mode and set that so that when console_ioctl(IOCTL_GENCON_SET_MODE,&mode) isn't used (as it clears the screen) it can keep the current screen contents as per ZX behaviour.

SAMC has a function getgraphmode() in misc.c (https://github.com/dandoore/samclibs/blob/main/c/misc.c) which does this:
```
//
// get video mode number 1..4
//
#ifdeclared getgraphmode
int getgraphmode(void )
{
    #asm
    in a,(&fc)
    and %01100000
    rlca
    rlca
    rlca
    inc a
    ld l,a
    ld h,0
    #endasm
}
```

Use case: With a program running under MODE3 (hi res) I need to load a SCREEN$ file before calling the code so cannot set the mode in the code as it clears it, when running it sets to MODE4 so screen and text are broken. 

[image]

suborb, 2021-05-11:
I've just added a pragma to control this behaviour, so try this from the next nightly: https://github.com/z88dk/z88dk/wiki/Platform--Sam-Coupe.md#startup-mode

dandoore, 2021-05-12:
Sorted - cheers!
[image]

closed 2021-05-12 by dandoore