z88dk/z88dk #1789: regression issue compiling for msx2 target

vipoo, 2021-05-14:
In a previous build of z88dk, the following command would compile just fine:

`zcc +msx -create-app -subtype=msxdos2 -compiler=sdcc -lmath32 -Cc-D__MATH_MATH32 -D__MATH_MATH32 -pragma-define:CLIB_32BIT_FLOAT=1 lines.c v9958.c msx.asm v9958.asm -o lines.com`

But I just updated to latest version and I get this error:

```
Error at file '/home/dean/z88dk/lib/../lib/target/msx/classic/msxdos.asm' line 64: symbol 'defltdsk' not defined
Errors in source file /home/dean/z88dk/lib/config/../..//lib/target/msx/classic/msx_crt0.asm:
Error at file '/home/dean/z88dk/lib/../lib/target/msx/classic/msxdos.asm' line 64: symbol 'defltdsk' not defined
                   ^ ----     ld      (defltdsk),a
```

Now, I dont really understand any of the code in `msxdos.asm`, but it appears to me that some restructuring was done in and around commit  f6a7d483500114d81b3bad7f4106f97ce5c31695

I found this section of code in `msxdos.asm:99`

```
IF CLIB_MSXDOS = 1
    INCLUDE "crt/classic/crt_cpm_fcntl.asm"
ENDIF
```

I was able to get things to compile for me without error, if I changed 1 to 2. 

Dean.

suborb, 2021-05-14:
Thanks, sorry about that - I had a bit of a large rearrange of the CP/M & MSXDOS1 fcntl code and it looks like msxdos2 was a victim.

It should be a pretty simple fix (that variable is never actually used) so I'll push a change this evening. 

Ironically I'm currently working on an automatically building configurations to ensure features link correctly, and MSXDOS1/2 was next on the list to do!

suborb, 2021-05-14:
This one is now sorted and permanently regression tested so shouldn't recur.

closed 2021-05-14 by suborb