z88dk/z88dk #1790: [samcoupe][color] cyan is displayed as purple/violet + red and green inverted

Fabrizio-Caruso, 2021-05-15:
Some colors are totally wrong.
In particular:

    red and green are inverted and
    cyan is shown as purple/violet.

The screenshot below shows the problem:
[image]

suborb, 2021-05-15:
See: https://github.com/z88dk/z88dk/wiki/Platform--Sam-Coupe.md#paletteclut-mapping (it used to say 2,3,4 weren't mapped, but it turns out mode 2 is)

Given your game doesn't has pixel perfect colour I don't think mode 4 is needed.

Fabrizio-Caruso, 2021-05-15:
@suborb So do I have to map them? I will but the default ones are totally wrong.
Shouldn't they be mapped by default to at least the closest match or at least something vaguely similar?
Red and green as different as yellow and blue. It is not a matter of color perfection.
By default the samcoupe target is inverting two totally different colors.

Cyan has a less serious problem. It is mapped to purple. It could be matched to blue if no cyan or light blue or teal is available.

suborb, 2021-05-15:
No, you don't need to map them, just switch to mode 2 rather than mode 4! I don't think there's a compelling reason to use mode 4 for your game.

I can explicitly map them, however for mode4 I took the decision not to since the palette could be redefined to better match the named colours. If you can come up with a palette that works then I'm quite happy to add it!

Fabrizio-Caruso, 2021-05-15:
@suborb Sorry! I thought I was using mode 2. Stupid me. Thanks!
It works! I see correct games when I get to start the games (because of issue #1791 on kbhit/getch that randomly locks the computer)

closed 2021-05-15 by Fabrizio-Caruso