z88dk/z88dk #1791: [samcoupe][bug][getch] kbhit()/getch() freezes the computer

Fabrizio-Caruso, 2021-05-15:
I am using `getch()` and `kbhit()` to wait for a key.
The code below works on all targets across different dev-kits but freezes the samcoupe ramdonly (maybe when a key is pressed to long or too many times too rapidly):

```
#include<conio.h>
#define cgetc() getch()

void wait_for_key(void) {
    while(kbhit())
        (void) cgetc();
    while(!kbhit()) { }; 
    (void) cgetc();
}

int main(void)
{
    clrscr();
    while(1) {
        printf("press a key\n");
        wait_for_key();
    }
    return 0;
}
```

suborb, 2021-05-15:
This is an annoying side effect of #1756 - undoing the change there makes this snippet work correctly.

suborb, 2021-05-16:
Ignore the red X above: the azure pipeline is playing up.

Anyway, I've fixed #1756 a different way and it keeps the behaviour of the fix for that issue + sorts this particular issue out - let me know how it goes for you.

Fabrizio-Caruso, 2021-05-19:
It is fixed for me. Thanks!

closed 2021-05-19 by Fabrizio-Caruso