z88dk/z88dk #1792: (sdcc) Miscompilation from forum

suborb, 2021-05-18:
Nice succinct example from https://z88dk.org/forum/viewtopic.php?f=12&t=11537

```
#include <stdio.h>

typedef unsigned short uint16;

uint16 x, y;

static uint16 printHello() {
    printf("\nHello");
    return 1;
}

static uint16 hello(uint16 a, uint16 b) {
    uint16 result = 0;
    if(a + b > x - y){
        result = printHello();
    }
    return result;
}

void main() {
    x = 2000;
    y = 1000;
    hello(600,600);
}
```

Compilation yields (snipped to the problem bit):

```
        ld      hl,_y
        ld      a, (_x+0)
        sub     a, (hl)
        ld      h, a              <<<<<<< 
        ld      a, (_x+1)
        inc     hl
        sbc     a, (hl)
        ld      l, a
        ld      a, h
        sub     a, c
        ld      a, l
        sbc     a, b
        jr      NC,l_hello_00102

```

Compiling with --max-allocs-per-node0 works around this.

suborb, 2021-05-18:
Also present in latest vanilla SVN so raised as: https://sourceforge.net/p/sdcc/bugs/3242/

feilipu, 2021-05-18:
Related, there are a lot of other z80 affecting bugs recently (but after r12288 #1773) fixed.
But also lots of new code features. So it is hard to work pick a good new stable date/release.

Might be sensible to automate those builds... :smile: 

2021-05-17 Philipp Klaus Krause <pkk AT spth.de>
	  Tail call optimization for functions with no stack parameters.

2021-05-17 Philipp Klaus Krause <pkk AT spth.de>
	  Fix bugs #3235, 3239.

2021-05-17 Philipp Klaus Krause <pkk AT spth.de>
	  Fix a code generation bug in hl handling.

2021-05-17 Philipp Klaus Krause <pkk AT spth.de>
	  Tail call optimization fixes for register parameters.

2021-05-14 Philipp Klaus Krause <pkk AT spth.de>
	  Fix code generation bugs triggered by register arguments.
	  Fix cost function for byte and bit access.

2021-05-14 Philipp Klaus Krause <pkk AT spth.de>
	  Improve tail call optimization.

2021-05-14 Philipp Klaus Krause <pkk AT spth.de>
	  Fix a bug in handling of register parameters to function calls via pointer.

2021-05-13 Philipp Klaus Krause <pkk AT spth.de>
	  Fix handling of multiple register parameters.

2021-05-13 Philipp Klaus Krause <pkk AT spth.de>
	  Increase some regalloc_dry_run_cost penalties to avoid invalid allocation even when all alternatives result in expensive spilling.

2021-05-12 Philipp Klaus Krause <pkk AT spth.de>
	  Fixes for register parameter in a, l or hl.

2021-05-12 Philipp Klaus Krause <pkk AT spth.de>
	  Fix bug #3234.

2021-05-12 Philipp Klaus Krause <pkk AT spth.de>
	  Infrastructure for non-__z88dk_fastcall register arguments.

2021-05-09 Philipp Klaus Krause <pkk AT spth.de>
	  Improve stack adjustment.

2021-05-09 Philipp Klaus Krause <pkk AT spth.de>
	  Fix bug #3231.

2021-05-08 Philipp Klaus Krause <pkk AT spth.de>
	  Fix bug #3230.

2021-05-07 Philipp Klaus Krause <pkk AT spth.de>
	  Fix bug #3229.

2021-05-07 Philipp Klaus Krause <pkk AT spth.de>
	  Fix some bugs triggered by return values in other than the conventional registers.

2021-05-07 Philipp Klaus Krause <pkk AT spth.de>
	  Deduplicate return value handling in peephole optimizer vs code generation.

suborb, 2021-05-18:
> So it is hard to pick a good new stable date/release.

This seems to have been a problem over the past few years, it does feel like the last stable build we had was 9958. I really want to do a 2.1.1 z88dk release since there are some clangers in 2.1, but sdcc issues keep popping up and pushing it back.

feilipu, 2021-05-18:
It would be good to be able to help @spth to check and squash some known bugs and get to somewhere stable, but imho the sdcc code base is so intricate that touching anything z80 related could likely break something for another platform (for example).

Perhaps at least helping triage known issues is enough support?

spth, 2021-05-18:
> This seems to have been a problem over the past few years, it does feel like the last stable build we had was 9958. I really want to do a 2.1.1 z88dk release since there are some clangers in 2.1, [...]

The current (and for at least the past 15 years) SDCC development cycle is a stable x.y.0 release once per year.
Those releases tend to be quite reliable. There is a substantial code freeze period before, with extra testing by developers and users, and bug fixes. During that frozen trunk, new features are developed in branches, and merged after the release (once can sometimes even see this in benchmarks - scores tend to not change much or go down  a bit due to bugfixes disabling optimization before a release, then jump after the release).
In between releases there are still the regression tests to catch bugs (see http://sdcc.sourceforge.net/snap.php - snapshots with a green dot at the right pass regression tests). But bugs are definitely more common then than in releases.
Regression testing covers all ports, but only default options are tested for z80 related ports (except the z80 port itself, which is also tested with --reserve-regs-iy).

feilipu, 2021-05-18:
I read...
> snapshots with a green dot at the right pass regression tests.

so for z88dk v2.1.1 we should start with a sdcc snapshot that at least passes regression testing (green dot).
> only default options are tested for z80 related ports.

Would it make sense to extend these tests to cover the z88dk normal targets with z88dk specific options, and check for further "green dots"? Or would that not really achieve anything useful?

Is scanning the bug list and trying to help triage the issue going to be useful?
Or, don't bother?

spth, 2021-05-18:
> I read...
> 
> > snapshots with a green dot at the right pass regression tests.
> 
> so for z88dk v2.1.1 we should start with a sdcc snapshot that at least passes regression testing (green dot).

Yes, though there is one additional complication: To reduce load on the compile farm some targets are only tested every other day. When you click on the green dot, you can see which targets were tested, and which tests failed for which targets.
So to be on the safer side, you might want to take a snapshot where the tests also pass for the one before and the one after.

> > only default options are tested for z80 related ports.
> 
> Would it make sense to extend these tests' scope to cover the z88dk normal targets with z88dk specific options, and check for further "green dots"? Or would that not really achieve anything useful?

Which targets and options would that be?

> Is scanning the bug list and trying to help triage the issue going to be useful?
> Or, don't bother?

Usually, z80-specific SDCC bugs get fixed quickly, once reproduced. For reproducing having a small, compileable code sample is very helpful. It shouldn't depend on anything outside of SDCC (such as z88dk headers). The lack of interprocedural optimizations in SDCC is actually helpful here: Usually, one can easily isolate a single function to reproduce the bug.
Of course, patches are also helpful. Or checking if some old bug can still be reproduced in current SDCC.

I don't know if there is much triaging to be done. Which bug gets fixed first usually depends on how easy it is to reproduce and fix, not so much on how serious the impact is.

feilipu, 2021-05-19:
Original issue resolved in r12390. But Philipp is doing some additional cleanup of older issues now too.

feilipu, 2021-05-19:
>> Would it make sense to extend these tests' scope to cover the z88dk normal targets with z88dk specific options, and check for further "green dots"? Or would that not really achieve anything useful?

>Which targets and options would that be?

I was thinking of extending testing to z80n and z180 as they have different code generation paths.

Together with the iy register reservation specifically.

spth, 2021-05-19:
The z80n is not that much different from the z80. The developers of the z80n failed to use the opportunity to make real improvements useful to compilers¹, and focused on special-case instructions useful only for very specific tasks when programming in assembler. Also I don't know if there really are a lot of z80n users out there.

The differences probably don't make it worth including a separate ucz80n-resiy test target in addition to the existing ucz80n test target. But I'll look into adding a ucz180-resiy target.

¹ push nn is nice when there is no free register pair. And mlt de is better than having no multiplication at all. And sometimes even swap a, tst a, n, add rr, nn and add rr, a instructions can be used. Still, in the end they all only get used rather rarely. Also the cycle times for Z80 instruction are all the same as on the Z80 itself.

spth, 2021-05-20:
The z180 with --reserve-regs-iy has been added to regression testing on the compile farm. For z80 / z180 we now:

test "-mz80" and "-mz180 --reserve-regs-iy" on even days
test "-mz80 --reserve-regs-iy" and "-mz80" on odd days.

feilipu, 2021-05-21:
So it sounds like for the z88dk 2.1.1, two consecutive days of green dots for z80-resiy and z180-resiy would be the target zsdcc.

spth, 2021-05-21:
Most test targets are executed either on even or odd days. Currently the only exception is ds390, which is tested every day.

PR #1799 (2021-06-12): zsdcc - r12419 - bugfix
Updating sdcc addressing bugs.

Closes #1792
Closes #1794

Partial changelog from r12288 (previous zsdcc version).
```
2021-06-03 Philipp Klaus Krause <pkk AT spth.de>
	* src/SDCCast.c,
	  src/SDCCerr.c,
	  src/SDCCerr.h,
	  support/valdiag/tests/bug-2798.c:
	  Fix bug #2798.

2021-05-27 Philipp Klaus Krause <pkk AT spth.de>
	* src/z80/gen.c,
	  support/regression/tests/bug-3244.c:
	  Fix bug #3244.

2021-05-20 Philipp Klaus Krause <pkk AT spth.de>
	* support/regression/Makefile.in,
	  support/regression/ports/ucz180-resiy:
	  Test z180 port with --reserve-regs-iy.
	* s [... cut]
closed 2021-06-12 by suborb