z88dk/z88dk #1794: ZCC compilation issue/error

vipoo, 2021-05-23:
Hi,

I think i may have identified a ZCC compiler error.  I recently attempted to recompile a project that i had not worked on for a few months - and noticed some tests were now failing.

I managed to track it down to an issue with CARRY FLAG/SBC handling.

## How to reproduce:

Compile the following `zcc +cpm -clib=sdcc_iy -S  -create-app test.c`

```
#include <stdlib.h>
#include <stdio.h>
#include <stdlib.h>
#include <stdbool.h>

uint16_t p = 0x202;

void main() {
  if (p < 0x0200)
    printf("1. First time it does not evaluate to TRUE\r\n");
  else
    printf("2. First time it does evaluate to FALSE\r\n");

  if (p < 0x0200)
    printf("3. But now it incorrectly evaluates to TRUE\r\n");
  else
    printf("4. It should have been FALSE\r\n");
}
```

When you run the code, the first condition correctly evaluates to TRUE, but the same condition tested a second time, evaluates to FALSE

I get this output:

```
cpm --exec A

2. First time it does evaluate to FALSE
3. But now it incorrectly evaluates to TRUE
```

The assembly it produces for the condition is:
```
	ld	a,(_p + 1)
	sbc	a,0x02
	jr	NC,l_main_00102
```

Its using SBC, but it never clears the CARRY FLAG before the SBC, so its behaviour is a little unpredictable.

I reproduced this with commit #4f92db5666c994274057b33b0d2fee47c7e29354

I also tried this with different levels of optimisation.

If this is an issue, I not sure when it got introduced as I had not touch my project that is now erroring, for about 6 months.

Hope this makes sense and is helpful.

Dean

vipoo, 2021-05-23:
I see there is some discussion re SDCC releases and identify the a good stable version - perhaps this issue will be solved as part of that....

feilipu, 2021-05-23:
Hi Dean, I'll have a look tonight. But, I'll just end up cross posting on the SDCC tracker for Philipp to fix. Which will then flow in here.

Cheers

vipoo, 2021-05-23:
Thanks mate - appreciate yours and others efforts...

feilipu, 2021-05-23:
> Its using SBC, but it never clears the CARRY FLAG before the SBC, so its behaviour is a little unpredictable.

Yes. I see the second test doesn't reset the CARRY before the test. I'll raise a case on SDCC.
Thanks Dean.

feilipu, 2021-05-27:
Reported fixed in sdcc r12404.

PR #1799 (2021-06-12): zsdcc - r12419 - bugfix
Updating sdcc addressing bugs.

Closes #1792
Closes #1794

Partial changelog from r12288 (previous zsdcc version).
```
2021-06-03 Philipp Klaus Krause <pkk AT spth.de>
	* src/SDCCast.c,
	  src/SDCCerr.c,
	  src/SDCCerr.h,
	  support/valdiag/tests/bug-2798.c:
	  Fix bug #2798.

2021-05-27 Philipp Klaus Krause <pkk AT spth.de>
	* src/z80/gen.c,
	  support/regression/tests/bug-3244.c:
	  Fix bug #3244.

2021-05-20 Philipp Klaus Krause <pkk AT spth.de>
	* support/regression/Makefile.in,
	  support/regression/ports/ucz180-resiy:
	  Test z180 port with --reserve-regs-iy.
	* s [... cut]
closed 2021-05-29 by feilipu