z88dk/z88dk #1797: snap installation on Ubuntu does not place asmpp.pl and asmstyle.pl in /snap/bin

iratahack, 2021-05-26:
When installing z88dk on Ubuntu with the command below the two perl scripts, _asmpp.pl_ and _asmstyle.pl_ are not included in the _/snap/bin_ directory.

`sudo snap install --edge z88dk`

I assume the file below creates the snap image but I'm not familiar with the format.

https://github.com/iratahack/z88dk/blob/master/snap/snapcraft.yaml

suborb, 2021-06-04:
Yes, that is what is needed, though I feel it would be politer to expose them as z88dk-asmpp.pl and z88dk-asmstyle.pl

iratahack, 2021-06-04:
@suborb , I agree. Which brings up another point. When installing the executables after building from source all binaries except zcc, sccz80, z80asm, and the asmpp.pl and asmstyle.pl perl scripts have the z88dk prefix.  Should these 5 files also have a z88dk prefix? The name changes would then transfer through to snap.
What are your thoughts?

suborb, 2021-06-04:
I went through a thought process a few months ago when creating the snap which is how some things got renamed:

* zcc: This ought to stay as it is - there's far too many tutorials and guides to rewrite
* z80asm: This is double exported I think - internally we use z88dk-z80asm and z80asm is left for compatibility purposes
* sccz80: This could be renamed since it's usually just invoked internally, we should really do the same to zdcc as well then
* Perl extra files: Easy choice - these should definitely be renamed!

iratahack, 2021-06-04:
I'll take a look and open a pull request if that's OK with you?

closed 2021-06-12 by suborb