z88dk/z88dk #1804: #asm/#endasm cannot be used in a macro

Oleg-N-Cher, 2021-06-06:
Dear z88dk developers,
When working on zcc, I came across its difference from SDCC. In particular, zcc has constructs #asm/#endasm, whereas in SDCC they are designated __asm/__endasm

There is nothing wrong with this difference, but let's say I need to develop a macro that will implement a single machine command (for example, HALT). In SDCC I can wrote:

`#define WAITSYNC __asm halt __endasm`

And it will work.

Whereas, a similar macro containing a service symbol # will not work in zcc:

`#define WAITSYNC #asm halt #endasm`

Maybe I certainly do not understand something or have not studied the documentation well. In that case, correct me. However, if I am right, I would like to request such possibility. The simplest thing is to make the __asm/__endasm directives available in zcc as synonyms of #asm/#endasm.

Thank you.

zx70, 2021-06-06:
I think this hint will answer you needs:

https://github.com/z88dk/z88dk/blob/ae3a72750e9101ce5c326ede44b90f3aa82cb887/include/stdlib.h#L270

suborb, 2021-06-06:
sdcc uses a bit of nasty "hack" to get __asm/__endasm to work in macros: https://github.com/z88dk/z88dk/issues/1268 

Thankfully as Stefano pointed out there's an alternative which is using the `asm("")` function style invocation. Although the tree is using `asm()`, `__asm__()` works with both compilers. There's a multi-line example of the usage here: https://github.com/z88dk/z88dk/blob/master/include/_DEVELOPMENT/proto/im2.h#L23

feilipu, 2021-06-06:
There are some further examples of the ways to do assembly macros in [im2.h](include/_DEVELOPMENT/proto/im2.h).

Oleg-N-Cher, 2021-06-07:
Thank you, guys. Yes, it will really help me.

Why is the \t symbol so widely used inside? Can there be a space? (this will be more visual).

Oleg-N-Cher, 2021-06-07:
P.S. zcc doesn't compile this:

`         do{ __asm defb 12h  __endasm; }while(0) ;`

> Bolder16K.c:75:20: fatal error: Unknown symbol: __asm
> Compilation aborted

**suborb**, you wrote here (https://github.com/z88dk/z88dk/issues/1268#issuecomment-525888948) that it must works  for -compiler=sccz80

Where am I wrong? Or do I just need to update the z88dk?

Oleg-N-Cher, 2021-06-07:
Please, answer: can I declare an aligned array in zcc?
SDCC allows such construction. but zcc looks like not.

`unsigned char __at(0xFF00) Rsrc_field [256];`

suborb, 2021-06-07:
> Please, answer: can I declare an aligned array in zcc?
> SDCC allows such construction. but zcc looks like not.
> 
> `unsigned char __at(0xFF00) Rsrc_field [256];`

sccz80 is a bit picky about the order it seems, try moving the `__at`:

```
extern __at(0x4000) char x[200];
```

Oleg-N-Cher, 2021-06-09:
Thank you very much for the useable hint, **suborb**.
I think I can close the issue.

closed 2021-06-09 by Oleg-N-Cher