z88dk/z88dk #1806: [bug?][sound][SN76489] big bang on sms/gamegear targets and other targets with PSG SN76489

Fabrizio-Caruso, 2021-06-06:
I am not able to get bit bang sounds on the sms and gamegear targets.

I am using `examples/sound/fx.c` 
I can get sounds and see the printed text if I compile it for the zx target but I hear and see nothing on the sms (tested on Kega Fusion and Emulicious).

I compilethe `fx.c` example with:
```
zx_bit_bang: 
	zcc.exe +zx -clib=default -lndos $(SOURCE_PATH)/../../test/fx.c -create-app

sms_bit_bang: 
	zcc.exe +sms -clib=default -lndos $(SOURCE_PATH)/../../test/fx.c -create-app
```

suborb, 2021-06-07:
[sms_fxdemo.zip](https://github.com/z88dk/z88dk/files/6607034/sms_fxdemo.zip)

The attached zip file contains an SMS rom that makes sound for me using emulicious with the latest WhatsNew.txt containing:

```
==========
2021-05-31
==========
- Improved Game Boy Color color correction and made it configurable (thanks to Eievui)
```

Fabrizio-Caruso, 2021-06-07:
Indeed it was a problem with the emulator(s). Sorry for reporting a false issue.

closed 2021-06-07 by Fabrizio-Caruso

suborb, 2021-06-07:
@Fabrizio-Caruso you're not having a lot of luck with emulators lately!