z88dk/z88dk #1807: sccz80 - sizeof compilation

feilipu, 2021-06-07:
I'm having trouble with a `sizeof()` call with sccz80.

```c
#include <stdint.h>

uint8_t *pucData;

size_t test_function( void * Buffer,
                        size_t xDataLengthBytes,
                        uint16_t xTicksToWait );

void main()
{
    size_t xReturned;
    xReturned = test_function( xMessageBuffer, sizeof( pucData[ 0 ] ), 0 );
}

// zcc -mz80 -compiler=sccz80 -c test1807.c
```
[test1807.c.txt](https://github.com/z88dk/z88dk/files/6608342/test1807.c.txt)

```sh
$ zcc +test -mz80 -compiler=sccz80 -c test1807.c
test1807.c:12:65: error: Missing token, expecting ) got 0
test1807.c:12:65: fatal error: Too few arguments to call to function 'test_function'
```

suborb, 2021-06-07:
See #704 - I think sizeof (*pucData) should work?

feilipu, 2021-06-07:
```
 xReturned = test_function( xMessageBuffer, sizeof( * pucData ), 0 );
```
Doesn't throw an error. Thanks.
But I've 99 other problems so can't tell whether it is actually working. ;-)

closed 2021-06-07 by feilipu