z88dk/z88dk #1813: z88dk - check z88dk-bugfix peepholes

feilipu, 2021-06-13:
In #1811 it was noted that the sdcc bugfix peepholes refer to r9570.
We are now at r12419 and possibly (probably) some of these will be obsolete.

This issue to track progress.

My plan is to compile all the available example code, and find if any of the `peephole z88dk-bugfix-xx` appear in any of the compiled versions. I expect to do this with `--list --fverbose-asm` compile options.

feilipu, 2021-06-13:
Below substantial C libraries and programs have been compiled. No mention of any of the `z88dk-bugfix-  ` rules being triggered.

 - [ChaN FatFS](https://github.com/feilipu/z88dk-libraries/tree/master/ff) - nothing.
 - [Duane Rice time](https://github.com/feilipu/z88dk-libraries/tree/master/time) - nothing.
 - [FreeRTOS](https://github.com/feilipu/z88dk-libraries/tree/master/freertos) - nothing.

 - [yabios](https://github.com/feilipu/yaz180/tree/master/yabios) - nothing.

 - [Whetstone](https://github.com/z88dk/z88dk/tree/master/libsrc/_DEVELOPMENT/EXAMPLES/benchmarks/whetstone/z88dk-new) - nothing.
 - [Dhrystone](https://github.com/z88dk/z88dk/tree/master/libsrc/_DEVELOPMENT/EXAMPLES/benchmarks/dhrystone21/z88dk-new) - nothing.

 - [Startrek](https://github.com/z88dk/z88dk/blob/master/libsrc/_DEVELOPMENT/EXAMPLES/startrek.c) - nothing.

Not exhaustive, but certainly indicative that they're not useful.

feilipu, 2021-06-15:
@Fabrizio-Caruso could you please test whether Cross Chase or Cross Lib use any of the `z88dk-bugfix-` peephole rules?

You just need to add `--list --fverbose-asm` to your compile instructions, and when you have the `~.lis` files ready, just `grep` for any appearance of the rules. If there's none it would be good to know. If there's any reference and they're being used, even better to know.

Would be good to do it using a current nightly with r12419 zsdcc.

Thanks. :pray:

feilipu, 2021-06-22:
@suborb looking at some of the peepholes in the aggressive set, they are just shuffling registers and operand order with no performance or code size outcome.

I thought they were trying to achieve some kind of second pass reordering outcome, but as they all include an inline assembly language comment further optimisation by a restart of the peephole optimisation process seems to be precluded.

The one that struck my mind recently is [621](https://github.com/z88dk/z88dk/blob/master/libsrc/_DEVELOPMENT/sdcc_peeph.3#L27540) (as an example). It simply reorders code which makes the resultant assembly from my C code more difficult to read.

There are a few other peepholes in this same basket.

 - [610](https://github.com/z88dk/z88dk/blob/master/libsrc/_DEVELOPMENT/sdcc_peeph.3#L27178)
 - [618](https://github.com/z88dk/z88dk/blob/master/libsrc/_DEVELOPMENT/sdcc_peeph.3#L27453)
 - [619](https://github.com/z88dk/z88dk/blob/master/libsrc/_DEVELOPMENT/sdcc_peeph.3#L27489)

Is there a benefit to reordering code that I'm not aware of?
Was there a plan to optimise around assembly comments that would make this code reordering more useful?

feilipu, 2022-02-08:
No benefit from removing junk here, and it might actually break things.

closed 2022-02-08 by feilipu

aralbrec, 2022-02-08:
The rules that shuffle instruction order are needed.  The peephole rules are pattern matchers and the order of instructions determine whether rules can be applied or not.  There is a strategy in the peephole set to try move `pop` and some other instructions upward so that important instructions like `ld hl,xxx` are pushed downward to perhaps match another rule set.  These rules are coming from observation of the code sdcc generates where it will sometimes insert pop instructions in the middle of another code generation idiom that prevents rules that should be applied from being applied.

In the past I saw this change as well:

```
replace restart
{
	ld	hl,#%1
	ld	a,(hl)
	ld	(hl),a
} by {
	ld	a,(#%1)
---	ld	(hl),a
	ld	hl,#%1
	; peephole z88dk-bugfix-12a
}
```
where the load to hl in the replacement has been removed.  This is incorrect as well.

The bugfix rules are repairing nonsensical code produced by sdcc and the replacement code is restoring the intention of the compiler rather than reproducing equivalent behaviour for the nonsensical stuff it is replacing.

The code being replaced:

```
	ld	hl,#%1
	ld	a,(hl)
	ld	(hl),a
```

is code that would normally crash because sdcc is generating improper code in these instances.  The intention is the replacement:

```
	ld	a,(#%1)
	ld	(hl),a
	ld	hl,#%1
```

This corrects the output of sdcc so that the program will run fine.  What sdcc wanted to do as read a value from an address, store it in the current address pointed by hl and then load hl with the source address.

I would not remove any of the bugfix peephole rules.  It's possible they are no longer needed but they were rare occurrences when they were created and they were only spotted after investigating buggy compiles.  A few were reported bugs in sdcc's bug tracker.

feilipu, 2022-02-08:
> There is a strategy in the peephole set to try move `pop xx` and some other instructions upward so that important instructions like `ld hl,xxx` are pushed downward to perhaps match another rule set.

I thought this was the intention, but it seems unworkable as the trailing comment breaks the pattern matching in every case. How would it work?

aralbrec, 2022-02-08:
Which trailing comments?  The `; peephole` comments are ignored by the peepholer.

Ignoring instructions after assembly were likely patched by myself but I don't see that instance here.  I don't know what the state of the sdcc instruction matching is at the moment; in the past our patch was much more robust.  I'd have to look at it again.

feilipu, 2022-02-08:
I spent some time looking at this, and the trailing `; peephole` comments were breaking further optimisation in what I saw.

Perhaps I was doing it wrong? But if time is available it would be worth checking that second and later passes overlook code comments.

aralbrec, 2022-02-09:
Without `--fverbose-asm` the standalone comments will not be emitted but with it, they will be emitted but they should be ignored by the peepholer.  That's definitely a bug if that has changed.  `--fverbose-asm` is all about documenting what rules were applied in order to track down errors in them.

It is quite common you will see a bunch of `; peephole z88dk-n` comments in a row in the output and if that's the case then the rules have been applied one after the other ignoring the comment.

feilipu, 2022-02-09:
Looking back at this for reference, it was the #1809 issue that started off the thought process.
SDCC changed its code generation process, which triggered the [bugfix-12a/b](https://github.com/z88dk/z88dk/pull/1811/files) and they ended up doing the wrong thing.
They changed the code flow, setting the `(hl)` to `a`, without first setting `hl`.