z88dk/z88dk #1814: perl scripts and snap

iratahack, 2021-06-13:
@suborb , looks like it isn't going to be so simple to add the perl scripts (asmpp and asmstyle) to snap. I get the following errors when trying to run them. Unfortunately, I'm running Ubuntu under WSL and I can't install snapcraft because multipass requires KVM which apparently is not supported by my environment.

Both scripts give the following errors. It seems the perl modules need to be part of the snap image build.

```
Can't locate Capture/Tiny.pm in @INC (you may need to install the Capture::Tiny module) (@INC contains: /home/devel/perl5/lib/perl5 /etc/perl /usr/local/lib/x86_64-linux-gnu/perl/5.26.1 /usr/local/share/perl/5.26.1 /usr/lib/x86_64-linux-gnu/perl5/5.26 /usr/share/perl5 /usr/lib/x86_64-linux-gnu/perl/5.26 /usr/share/perl/5.26 /usr/local/lib/site_perl /usr/lib/x86_64-linux-gnu/perl-base) at /snap/z88dk/2468/bin/z88dk-asmpp line 28.
BEGIN failed--compilation aborted at /snap/z88dk/2468/bin/z88dk-asmpp line 28.
```

suborb, 2021-06-13:
The search path there is your system - so I think you need to install the perl modules in your system?

iratahack, 2021-06-13:
Yeah. I have them installed. I can run the scripts fine directly from the git repo build but not from the snap install. This also fails in the GitHut action build the same way.

I've tried the command below with and without sudo but no luck.

cpan install App::Prove Modern::Perl Capture::Tiny Capture::Tiny::Extended Path::Tiny File::Path Template Template::Plugin::YAML Test::Differences CPU::Z80::Assembler Test::HexDifferences Data::HexDump Object::Tiny::RW Regexp::Common List::Uniq Text::Table

iratahack, 2021-06-13:
@suborb , I have things building locally now. Changing "confinement:" to "classic" resolves the issue but also gives the warning below when installing.  I'm not sure if this is the correct fix or if there is a way to tell snapcraft in include the required perl modules then confinement could remain as strict, which sounds like a better solution.

error: This revision of snap "./z88dk_18432-f24dd45b9-20210613_amd64.snap" was published using
       classic confinement and thus may perform arbitrary system changes outside of the security
       sandbox that snaps are usually confined to, which may put your system at risk.

       If you understand and want to proceed repeat the command including --classic.

closed 2021-06-17 by suborb