z88dk/z88dk #1817: [C128] border and background colors no longer work. It was ok on 2021/06/04

Fabrizio-Caruso, 2021-06-23:
The following code shows the problem:
```
#include<stdio.h>
bordercolor(BLACK);
textbackground(BLACK);
int main() {
    printf("Hello World!\n");
    while(1){};
    return 0;
}
```

This works on previous snapshots, e.g., 2021/06/04 is OK.

The current snapshot shows grey border and background.

suborb, 2021-06-25:
@z70 made this change: https://github.com/z88dk/z88dk/commit/dfb8f2e2b227a058af3e4201320255e55c342d00 a couple of weeks ago which is probably the cause of it.

Fabrizio-Caruso, 2021-06-25:
@Z70 could you please revert or fix your change?

zx70, 2021-06-25:
The behavior changed but it lets you a better control. 
please try adding a clear screen code (12).
The difference hassling you is thd DEFAULT background color (now set to light gray at startup).   Originally you simply couldn't change it at all (the textbackground function had no effect).  Now the whole background colour is updated at the next cls event, which is closer to the way the other targets work.

Fabrizio-Caruso, 2021-06-25:
@zx70 I get the same grey color in my games that do clear the screen several times.

zx70, 2021-06-25:
Are you setting textbackground(0) and textcolor() before clearing the screen ?

Fabrizio-Caruso, 2021-06-25:
@zx70 I am but I need to check. I am away from the pc now. The problem I am having now is very recent and it affects all my games that all use the same code across all targets.

zx70, 2021-06-25:
well, reverting to the default colour settings is obviously possible, the new cls behaviour can still stay.   But you're right,  you should have had the same problem on other targets (e.g. the ZX Spectrum, which is white by default) .

Fabrizio-Caruso, 2021-06-25:
@zx70 
Clearing the screen does not fix the recent issue.

You can try:
```
#include<stdio.h>
bordercolor(BLACK);
textbackground(BLACK);
int main() {
    printf("%c",12);
    printf("Hello World!\n");
    while(1){};
    return 0;
}
```

zx70, 2021-06-28:
Background is now black again, please check after the tonight's build.

Fabrizio-Caruso, 2021-06-29:
The bug is fixed for me.

closed 2021-06-29 by Fabrizio-Caruso