z88dk/z88dk #1818: Extrange behaviour with extern structs

mojontwins, 2021-06-28:
I'm in the process of recompiling code I wrote for z88dk 1.10 with the latest nightly build and I've found something odd - or maybe I'm getting this completely wrong.

It has to do with extern structs, i.e.

```
typedef struct {
	unsigned char a;
	unsigned char b;
} MY_STRUCT;

extern struct MY_STRUCT *foo;

#asm
	._foo
		defb 0, 0
#endasm
```

I used to make extern structs in my code that way when I used 1.10, and either foo->a and foo.a compiled and worked fine. Now, in the latest nightly build, 

```
void main (void) {
	foo.a = 7;
	foo.b = 8;
}
```

Produces this error when compiling:

```
$ zcc +zx -vn minimal.c -O3 -crt0=crt.asm -o minimal.bin
minimal.c:16:8: fatal error: Member reference to 'a' via 'struct MY_STRUCT * foo' is a pointer; did you mean to use '->'?
Compilation aborted
```

which, to my knowledge, is right - as "foo" is a pointer. But if I change the code to:

```
void main (void) {
	hola->a = 7;
	hola->b = 8;
}
```

Then I get:

```
$ zcc +zx -vn minimal.c -O3 -crt0=crt.asm -o minimal.bin
minimal.c:16:11: fatal error: Unknown member: 'a' of struct 'MY_STRUCT'
Compilation aborted
```

Is this a bug or am I doing this in the wrong way?

Thanks.

suborb, 2021-06-28:
The declparser was rewritten so things that worked through luck now report errors. For your example this matches up with what I think you want to do:

```
typedef struct {
        unsigned char a;
        unsigned char b;
} MY_STRUCT;

extern MY_STRUCT foo;

#asm
._foo
        defb    0,0
#endasm

int main() {
        foo.a = 7;
        foo.b = 8;
}
```

This is the main change: `extern MY_STRUCT foo;` - you've allocated space in the asm for foo, so it's not a pointer. Secondly `struct MY_STRUCT foo` sets up a weak definition of a structure with a tag of MY_STRUCT: hence the diagnostics you were getting. So you can just use the typedef directly.

mojontwins, 2021-06-28:
Thanks for the comments.

You are doing an excellent job!

closed 2021-07-12 by suborb