z88dk/z88dk #1819: math48 - trigonometric function accuracy?

feilipu, 2021-07-04:
Recently I've been building a very [rough planetary motion calculator](https://github.com/feilipu/planets) to test the RC2014 APU Module and benchmark the Am9511A against soft floating point. Whilst the calculator is very inaccurate, because it does no perturbations of any type, it does serve to test timing and exercise functions, and the results can be compared with an [accurate planetary motion calculator](https://cosinekitty.com/solar_system.html) for general correctness.

So I've just [tested with the math48](https://github.com/feilipu/planets/blob/main/planet48.txt) library, and I'm finding that some of the results are way off the correct [Cosine Kitty outcome](https://cosinekitty.com/solar_system.html). And, most interestingly as the algorithms and code are identical, they don't match either [math32](https://github.com/feilipu/planets/blob/main/planet32.txt) or the [Am9511A APU](https://github.com/feilipu/planets/blob/main/planetapu.txt) results.

I was of the belief that math48 was pretty accurate, so this strikes me as odd.

Am I doing something wrong to get these very odd numbers?

suborb, 2021-07-04:
My immediate hunch was in the conversion that takes place between the IEEE format for sdcc and the math48.

I do get different results when I use +test/math48/sccz80. The first row of results is more in line with the other libraries. It fails after that first line but I think that's due to the rev() implementation  making some assumptions.

Maybe a thread to pull?

feilipu, 2021-07-05:
I've [tweaked the build](https://github.com/feilipu/planets/commit/20dbfa10b72d2812b92980a6d49cc989593afac9) to enable sccz80 / newlib. The `rev()` function looks very efficient in sccz80, so hardly worth converting to assembly.

Not sure how to get printing working with +test z88dk-ticks though.

And, oddly enough I'm getting yet another [set of results](https://github.com/feilipu/planets/blob/main/planetnew.txt). Most interestingly, these results are the most correct (according to Cosine Kitty). The large errors I am getting for Jupiter & Neptune that I'd put down to perturbations are not present in the sccz80/newlib results.

So that means to me that all the sdcc/newlib math library combinations are broken. :disappointed:

feilipu, 2021-07-05:
I'm suspicious of the `atan2()` interface implementation in math48. Adding `rev()` to the sun calculation slightly changes the output for math32 and am9511, but completely changes output for math48.

And then, if math48 can be repaired, I'm suspicious of the overall calculation accuracy with IEEE single precision because the math32 and am9511 APU results are similar, but in some cases quite wrong (where sccz80/newlib/math48 is correct).

suborb, 2021-07-05:
> Not sure how to get printing working with +test z88dk-ticks though.

The printf pragma needs to include %.2f instead of %f - printf flags are a flag in classic.

feilipu, 2021-07-05:
I've added [results](https://github.com/feilipu/planets/blob/main/planetticks.txt) from a `+ticks` build, with results quite close to `scz80/newlib/math48`. Something works. :smile: 

But this just shows the `sdcc/newlib/math48` [results](https://github.com/feilipu/planets/blob/main/planet48.txt) are completely crazy.

And the `sdcc/newlib/math32` and most interestingly the `sdcc/newlib/am9511` [results](https://github.com/feilipu/planets/blob/main/planetapu.txt) are mostly right, but still often wrong.
Math isn't supposed to a raffle, so this isn't good enough either. :disappointed:

Checked math48 `atan2()` and it looks correct. So, don't know?
Could be a long thread to pull.

suborb, 2021-07-05:
So if I follow correctly:

Working:

- sccz80/genmath/classic
- sccz80/math48/newlib

Semi working:

- sdcc/math32/newlib
- sdcc/am9511/newlib

Wrong:

- sdcc/math48/newlib

Unknown:

- sccz80/math32/classic
- sccz80/am9511/classic (I don't think this one can be built?)

So far, this suggests to me that the core math48 library is working and it's just the sdcc bindings that have a mistake?

feilipu, 2021-07-05:
> So far, this suggests to me that the core math48 library is working and it's just the sdcc bindings that have a mistake?

Yes. I agree. But it is a surprise that sdcc/math48 isn't working for this code. Hence my thought I'm doing something odd that breaks it.

Seeing sccz80/math48 works gives some comfort that thh hee code ir right and should work. If there was no bug.

suborb, 2021-07-05:
Have you flipped between sdcc_ix and sdcc_iy? The bindings are different implementations.

feilipu, 2021-07-05:
> Have you flipped between sdcc_ix and sdcc_iy?

No. I'll try that.

And I think a trip into the sdcc way back machine is necessary too.

feilipu, 2021-07-06:
I'll edit this comment again when I'm sure, but it seems the main issue (apart from the IEEE single precision limitations), is the various ways that atan2(y,x) can be defined. Whether horizontal-anticlockwise `atan2(y,x)` or north-clockwise `atan2(x,y)` has an impact on the outcome, and causes gross location errors if it is used incorrectly.

I think this is my issue, and when I'm sure I'll have learned something.

feilipu, 2021-07-06:
Ok. I've [adjusted](https://github.com/z88dk/z88dk/commit/6cf640ebacc399f3614776c39289627a73e69d8e) math48 to get correct results.

That provides a consistent set of results across `z88dk-ticks`, `sccz80`, and `sdcc` across all the standard floating point libraries. Whether this consistent set of results is correct, I'll leave open to discussion. :smile:

Case closed.

closed 2021-07-06 by feilipu