z88dk/z88dk #1821: Wrong warning using ZSDCC and GameBoy target

Oleg-N-Cher, 2021-07-21:
zcc +gb -create-app -compiler=sdcc -o gbdemo.gb gbdemo.c

> Warning at file 'gbdemo.c' line 578: integer '65344' out of range
> Warning at file 'gbdemo.c' line 580: integer '65344' out of range
> Program requires cartridge with 2 ROM banks
> Notice: Available RAM space is 7928 bytes ignoring the stack
> Adding main banks 0x00,0x01 (25688 bytes free)

After compiling [gbdemo.zip](https://github.com/z88dk/z88dk/files/6852379/gbdemo.zip)

Yesterday nightly build of z88dk for Windows was used.

Wrong lines look as:

[image]

suborb, 2021-07-21:
Looking at the z80asm test files, it looks like z80asm expects the numeric parameter to be in the range 0-255 rather than actually pointing to the $ff00 page.

For sdcc this can probably be fixed up in the translation rules, but I do suspect that a similar problem may exist with sccz80 when it uses expressions rather than constants.

@pauloscustodio - could you take a look at fixing up z80asm?

pauloscustodio, 2021-07-21:
According to http://www.devrs.com/gb/files/opcodes.html, the warning is correct, as the instructions are described as:
`LDH (n),A | ($FF00+n) <-- A`
`LDH A,(n) | A <-- ($FF00+n)`

Is this source correct? If yes, the source code should be changed to
`LDH A, (0xFF40 & 0xFF)`

To fix z80asm I need to create a new record type in the object file and update the linker and z80nm to handle these high-page addresses.

pauloscustodio, 2021-07-22:
Fixed. Now z80asm accepts either a value in the range 0x00..0xff or 0xff00,,0xffff

commit 69f7a4d0 (2021-07-22): Merge pull request #1822 from z88dk/issue1821

Fix #1821 - create a new record type in the object file to handle gbz…
commit 8da6b402 (2021-07-22): Fix #1821 - create a new record type in the object file to handle gbz80 offsets to 0xff00
closed 2021-07-22 by pauloscustodio