z88dk/z88dk #1823: z80asm - 8085  B issue opcodes

feilipu, 2021-07-23:
I'm building a 8085 based CPU board for RC2014, and I'm planning to use the CA80C85B device at the same clock as standard in the RC2014. This is the device that formally adds the [undocumented opcodes](https://en.wikipedia.org/wiki/Intel_8085#Undocumented_instructions) to the 8085 instruction set.

[CA80C85B](https://github.com/z88dk/z88dk/files/6866198/CA80C85B-TUNDRA.pdf)

I don't think z80asm assembles these instructions when using the `-m8085` option.

Is that correct?
Would it be possible to add them to z80asm, please, if they're not already there?

suborb, 2021-07-23:
It does understand the Z80 style equivalent opcodes and I’m pretty certain that intel is supported too.

feilipu, 2021-07-23:
That's good. I would prefer Zilog style.

Is there any source or doc description of what they are in Zilog codes?

pauloscustodio, 2021-07-23:
Please look up the z80asm test file, https://github.com/z88dk/z88dk/blob/69f7a4d0185827c016ad5a18c2ea0b920af35f06/src/z80asm/dev/cpu/cpu_test_8085_ok.asm

feilipu, 2021-07-26:
Looking a bit further at this, there seems to be some opportunities in the 8085 support libraries to use the undocumented instructions more extensively.

I've attached some reference documents, an original [Dehnhardt/Sorensen](https://github.com/z88dk/z88dk/files/6875306/UnDoc85.pdf), and an excerpt from [the 8085B datasheet](https://github.com/z88dk/z88dk/files/6875310/8085_Extended_Opcodes.pdf).
[Ken Shriff's blog](https://www.righto.com/2013/02/looking-at-silicon-to-understanding.html) also adds to the story.

Some things to think about (and then do).
- [ ] use the Underflow Indicator (K) flag with 16 bit decrement and `JUI` `JNUI` instructions to manage loops, like `LDIR` emulation, more cleanly. 16 bit decrement flag is set on overflow to -1, not to 0, note to self.
- [ ] use the `LD DE,SP+n` instruction with `LD HL,(DE)` to grab from and `LD (DE),HL` store parameters on the stack. Can use this with a math library to make it reentrant, for example, and also relieves pressure on the small number of registers.

suborb, 2021-07-26:
There's not too many 8085 library optimisations in place, so plenty of scope for improvement. 

I've been somewhat lazy when adding the 808x support. For small tweaks I've just IF/ENDIF up the z80/*.asm files, for more substantial rewrites there's a separate 8080 directory.

feilipu, 2021-07-27:
>  for more substantial rewrites there's a separate 8080 directory.

Would it be appropriate to start a separate 8085 directory, considering that these optimisations will run in neither 8080 nor on z80?

suborb, 2021-07-27:
> Would it be appropriate to start a separate 8085 directory, considering that these optimisations will run in neither 8080 nor on z80?

Just take a call on the divergence and decide then.

> Is there a complete reference to the `l_xxx` function list that sccz80 calls?
> I assume it is [here](https://github.com/z88dk/z88dk/blob/master/lib/z80_crt0.hdr) in `z80_crt0.hdr`, right?
> And for 8080 / 8085 we are just replicating these functions?

Yeah that hdr file should have all of them (and probably some unused ones). The l_ files are still a bit of a mess (better than they were, not as good as they should be) mainly due to to avoiding duplication.

feilipu, 2021-08-02:
@pauloscustodio

> Please look up the z80asm test file.

Would it be possible to add jp k,n (FF nn) and jp nk,n (FD nn) syntax, please? I don’t think these are available.

pauloscustodio, 2021-08-03:
> Would it be possible to add jp k,n (FD nn nn) and jp nk,n (DD nn nn) syntax, please? I don’t think these are available.

@felipu, done.

feilipu, 2021-08-04:
@pauloscustodio

I think that the LD DE,HL+* and LD DE,SP+* instructions should have 2s complement tests.

Currently they're tested to unsigned values.

I'm assuming these instructions are made in same vein as the other offset instructions though. Do you know for sure?

feilipu, 2021-08-04:
@pauloscustodio

Just noting, it would be better to assemble LD DE,HL as LD D,H and LD E,L.

Using the LD DE,HL+0 (28 00) is 10 ticks vs. 8 ticks.

pauloscustodio, 2021-08-05:
> I think that the LD DE,HL+* and LD DE,SP+* instructions should have 2s complement tests.

According to https://electronicerror.blogspot.com/2007/08/undocumented-flags-and-instructions.html these opcodes add 00bb, i.e. an unsigned value to HL and SP.

Do you have different information?

z80asm interprets the argument as an unsigned byte, but if a negative value is used, it silently accepts the 2s complement, i.e. ld de,hl-1 results in 28 FF.

feilipu, 2021-08-05:
> These opcodes add 00bb, i.e. an unsigned value to HL and SP.

> Do you have different information?

No. Just a feeling based on Zilog relative instructions. Which was quite wrong.

Also had your reference confirmed as correct. And also the idea that decrement in high byte is cheap, so to create eg. -1 it is simply decrement (-100h), then add +FFh. 4 cycles more, but ok it's still cheaper than 8080.

pauloscustodio, 2021-08-05:
> Just noting, it would be better to assemble LD DE,HL as LD D,H and LD E,L.

@feilipu , Done.

feilipu, 2021-08-06:
Thoughts on creating a stack frame for reentrant functions.

```asm
                               ; LHS,RET and dehl registers
ld bc,de                       ; 8  ; save de (only for long or float)
ld de,sp+($100-stack_needed)   ; 10 ; stack_needed is the number of working variable bytes
dec d                          ; 4  ; add $100 bytes
ex de,hl                       ; 4  ; function parameter now in bcde
ld sp,hl                       ; 6  ; now we have stack_needed bytes to work with

...                            ; function parameter in bcde
...                            ; function return in bcde

ex de,hl                       ; 4
ld de,sp+($100-stack_needed)   ; 10 ; remove offset
ex de,hl                       ; 4
ld sp,hl                       ; 6  ; now we have stack offset removed
ld hl,bc                       ; 8
ex de,hl                       ; 4  ; return in dehl
ret
```

feilipu, 2021-08-15:
@pauloscustodio

I've been flipping backward and forward about whether to make this request, but on balance I think it is worth doing.

Would it be possible to implement a `LD HL,SP+*` 4 byte pseudo instruction for 8085, please?

`EX DE,HL`, `LD DE,SP+*`, `EX DE,HL`.

My rationale is that it can be implemented without any side effects, and takes just 18 cycles to complete (which is no more than a `CALL **` instruction).

And the justification is that it would be a useful tool to abstract moving "long" register contents in `BCDE` to and from the stack, without going via the `A` register.

Using the `A` register and `LD (DE),A` would still be faster for "int" though.

What do you think?

pauloscustodio, 2021-08-15:
@feilipu 
Done. Please let me know of any issues.

feilipu, 2021-08-19:
@zx70 
I'm playing with the 8085 `lpush2` function, and I've a couple of questions. Perhaps you can help?

First, I don't think the function is referenced anywhere, so perhaps it hasn't been exercised. But, is there a bug there?
```asm
.lpush2 
    ld      (__retloc),hl
    pop     hl      ;save next item on stack (BUG? should be pop bc).
    push    de      ;dump our long - MSW
    ex      de,hl   ;de is now return address
    ld      hl,(__retloc)
    push    hl
    ex      de,hl
    push    bc      ;store back "next item on stack"
    push    de
    ld      hl,(__retloc)
    ret
```
Second, a more general question, does the `lpush2` function have to preserve `dehl` as well as pushing `dehl` under the second return? Or can it destroy the registers?
I can't find any use of the function, so it is difficult to guess what is needed.

suborb, 2021-08-19:
Lpush2 is meant to be used when constructing a function pointer call and a long parameter needs to be passed. 

That area has had a few changes over the years so it may be redundant now. 

I can’t dig further at the moment since I’ve temporarily left the country.

suborb, 2021-09-20:
I've removed lpush2 since I can't see it being used anywhere.

feilipu, 2021-09-21:
> I've removed lpush2 since I can't see it being used anywhere.
Thanks. 

After [starting down the wrong track](https://github.com/z88dk/z88dk/pull/1845) with 8085, and getting distracted into redoing the hardware to support wait states and PAUSE, not much actual progress to report. Back onto it soon.

feilipu, 2021-09-27:
@suborb is it true to say that the files in the [libsrc/z80_crt0s/sccz80](https://github.com/z88dk/z88dk/tree/master/libsrc/z80_crt0s/sccz80) directory have been wholly replaced by those in the [libsrc/_DEVELOPMENT/l/sccz80](https://github.com/z88dk/z88dk/tree/master/libsrc/_DEVELOPMENT/l/sccz80) directory?

Is there a specification of which registers need to be preserved by these functions when called by sccz80? Even a piece of sccz80 code would be good to have as reference.

Specifically most of the `l_` functions don't use `de`, but using `ld de,sp+n` for stack access would be useful for 8085, particularly if `de` didn't need to be preserved.

suborb, 2021-09-27:
I don't think everything has been moved across, I've got a memory of running into trouble when I tried to switch over completely.

The knowledge of what registers used isn't documented explicitly anywhere. However l_gint* needs to preserve de which is the obvious optimisation point that can't be made.

suborb, 2021-09-27:
I think I've spent the evening going round in circles again with these routines - thanks for your efforts at starting to sort them out.

The main differences for crt0_z80 are:

- The shift routines are smaller in classic (newlib `jp`s to other routines)
- There's some ez80 tweaks about the place (no great loss)
- Some routines are inlined - eg l_cm_de inlines l_neg_de to avoid needing l/z80 on non-z80 machines.

I've not checked crt0_long_z80 but I think they are broadly the same.

For a while I've not been happy with how they've been organised in classic. The problem is I'm still not sure how else to arrange them. I was trying to avoid listing 100 routines in a .lst file, but I suspect that might end up being the best thing to do here. 

Thoughts/ramblings...

l/sccz80:

At the moment everything is arranged in terms of z80 as a base. However, I think our true base is 8080, bar the gbz80 (missing `ld hl,()/ld (),hl`), 8080 code can be compiled to any supported cpu.

So, for example for z180, we'd include `z180.lst` and `z80.lst` (which includes `8080.lst` + overrides) making use of the "first symbol in file that matches" linking rule that z80asm has so that "unoptimised" routines are ignored.

Conditional assembly makes life hard and for these routines it's probably best to try to avoid it, apart from to distinguish gbz80/8080 to avoid assembly errors (I'm thinking here of the long arithmetic routines which use the missing opcodes on gbz80).

l/z80:

We seem to have a similar issue, though it's mostly z80 code with very little 8080 base. I'm tempted to leave this one for a while.

feilipu, 2021-09-28:
OK. I'll take a rest until you've had a chance to think this through a bit, and recover from laptop rebuilding.
Then, if you let me know a plan of action I can help to implement.

I'll just tweak about on the rc2014-8085 target specifically for the time being.

suborb, 2021-09-30:
I'm nicely bogged down in this one, I think I can see the end though.

feilipu, 2021-09-30:
Thinking about using `inc (hl)` / `dec (hl)` for 8080/8085 atomic mutex, and how the K flag signed comparison can improve the `cmp` and `ucmp` intrinsic functions.

feilipu, 2021-10-02:
Have finished the two [rc2014-8085 subtypes](https://github.com/z88dk/z88dk/blob/master/lib/config/rc2014.cfg#L15) now. `basic85` and `acia85` are now both working using their stdio interfaces.

There are still a lot of library related errors when trying to build the example programs, due to missing library functions, but hopefully this can be washed out following the `l_` function clean up.

suborb, 2021-10-12:
I’ll look this evening but it might be the handle flags bit that’s not there? The width and +,* etc are handled by that.

feilipu, 2021-10-12:
Just leave this here...
[image]

feilipu, 2021-10-13:
> I'm debating just adding that file into the repo. Especially if we end up with z80 and 8085 enhancements.

Probably no reason not to upload it. It has copyright info attached, and [MS open sourced GW Basic](https://github.com/microsoft/GW-BASIC) already.

BTW, I tried to get MS CP/M Basic85 5.29 too open sourced too, but no luck.

suborb, 2021-10-13:
> > I'm debating just adding that file into the repo.
> 
> Probably no reason not to upload it. It has copyright info attached, and [MS open sourced GW Basic](https://github.com/microsoft/GW-BASIC) already.

Consider it done: https://github.com/z88dk/z88dk/commit/31454a8fafe160932d5f07d72bc015dc545b1eb6

I'd forgotten that I got it working on the gbz80 as well.

feilipu, 2021-10-13:
> Since you're using mbf32, you could provide a list of entry points into BASIC to use the ROM maths library.

Just thinking this can be a useful shortcut for Am9511 support too, as I've now got a 8085+Am9511 hardware with MS Basic working.

I am planning on a native 8085 am9511 library at some point, but hadn't thought of using MS Basic as the intermediary.

suborb, 2021-10-13:
> I'm planning on "improving" it slightly while learning 8085, so they might be a bit unstable for a while, but when I lose interest I'll do that! 

We need 16 entry points + 4 RAM variable addresses - any chance you could sneak a jp table in? It's only 48 bytes.

feilipu, 2021-10-13:
> I'm not sure I would have thought about that route either - daimath32 would be my first thought.

Given MS Basic with 8085/Am9511 is the basic85 ROM host, it will be quite cheap to use ROM math routines.

feilipu, 2021-10-19:
@suborb If you're sure that the test suite is going to exercise `l_long_cmp` and `l_long_ucmp`, and  the `l_long_bools`, I'd be keen to do some work with the 8080 versions too.

There is very little in the [8085 versions I've just done](https://github.com/z88dk/z88dk/commits/master) that couldn't be replicated into 8080 too. But I don't have a way to test it on hardware.

feilipu, 2021-10-19:
> added some, interesting results.

Odd. Should be ~40 cycles per comparison cheaper for 8085 currently. Sleuthing time!

suborb, 2021-10-19:
| Instruction | 8085 | 8080 |
|-|-|-|
| push | 12 | 11 |
| jp cc | 7/10 | 10/10 |
| call | 18 | 17 |
| call cc | 9/18 | 11/17 |
| ret cc | 6/18 | 5/16 |
|  jp (hl) | 6 | 5 |

I'd be surprised if we had 5000 pushes though.

feilipu, 2021-10-20:
8080 i32 cmp/ucmp code now - common code removced - 8085 timing
total: 146 cycles
```asm
    ex      (sp),hl         ; 18
    ld      (__retloc),hl   ; 16
    pop     bc              ; 10
    pop     hl              ; 10
    ld      (__retloc2),hl  ; 16

    ld      hl,sp           ; 20

;---snip---

.retloc
    ; We need to preserve flags in af
    ld      hl,(__retloc2) ; 16
    push    hl             ; 12
    ld      hl,(__retloc)  ; 16
    push    hl             ; 12
```

8085 i32 cmp/ucmp code now - common code removed - 8085 timing
total: 82 cycles - reentrant
```asm
    ld      bc,de       ;  8
    ld      de,sp+4     ; 10

;---snip---

; We need to preserve flags in af
.l_long_cmp2
    pop     bc          ; 10
    pop     de          ; 10
    pop     hl          ; 10
    pop     hl          ; 10
    push    de          ; 12
    push    bc          ; 12
```
I'll convert the 8080 code to use `ld hl,sp+4` 20 cycles. It should still be 54 cycles better than currently, and be reentrant.

feilipu, 2021-10-21:
Reference point current ACK 8080 multiply.
```sh
$ z88dk-ticks -m8085 test_8080_mult.bin 
Starting suite Multiplication Tests (2 tests)
2 run, 2 passed, 0 failed

Ticks: 255405
```

feilipu, 2021-10-30:
Extended the `l_div` [tests](https://github.com/z88dk/z88dk/commit/fd75b18f71635d3db884807fbbbb9d858504977e) to include negative divisor tests, to capture the correct remainder sign handling.

I'm not 100% sure that the remainder sign handling method is actually correct, but now at least classic/8080 and classic/8085 have been modified produce results consistent with newlib/rc2014 and classic/gbz80 results. :shrug:

zx70, 2021-10-30:
Felipu, I used that BASIC in the past to ..emulate the Nascom o  the Spectrum  :)
It was also the starting point for tuning basck, later integrated with the results of my MSX BASIC disassembly. 
BASCK, in turn, now permits me to use skoolkit and reverse-engineer most of the basic80 implementations quite quickly.
I'm currently trying to use the Alphatronic PC cp/m version (5.22) to get a recent source for the cp/m version to play with (e.g. to add graphics, etc).
It is halfway between the MSX and the Nascom, but a good compromise.

feilipu, 2021-11-05:
> I'm currently trying to use the Alphatronic PC cp/m version (5.22) to get a recent source for the cp/m version to play with (e.g. to add graphics, etc). It is halfway between the MSX and the Nascom, but a good compromise.

@zx70

I found this [Basic 5.2 Source](https://winworldpc.com/download/c2bfc380-c2b2-c2b4-7d00-11c3a7c29d25) recently. Perhaps it is useful for you? Don't know if it is completely real, or a disassembly, though it looks real enough. 🤷‍♂️

zx70, 2021-11-19:
I'll look at it.  There's also something interesting in the Sharp MZ world to look at.   Usually the MZ systems did not use the MS BASIC but few versions existed, and sources are available.  The CP/M versions are useful because they're naked, just the file support and a special line editor usable via terminals.

zx70, 2021-11-19:
This is the current status of my work ..  https://github.com/z88dk/techdocs/blob/master/targets/cpm/mbasic.asm

zx70, 2021-11-21:
The fp code could help in supporting the 8080 targets, but it uses SMC tricks in the division related code.
https://github.com/z88dk/techdocs/blob/869b8e967f61827f41c80c1438c0771ddc787beb/targets/cpm/mbasic.asm#L5986

zx70, 2021-12-02:
The MBASIC source is now clean enough to support a bit of editing.
As a proof of concept I merged in a bit of code coming from the MSX rom, dealing with the VDP.
It does very little, just the 'color' command in text mode shows some effect.. but the syntax command and the current video mode status are checked correctly.
It should be fairly easy to add custom commands in a similar way.

Note: this implementation of FP Division code was an impressive mess, it is full of SMC tricks !

suborb, 2021-12-02:
> Note: this implementation of FP Division code was an impressive mess, it is full of SMC tricks !

Yes, when mbf32 was imported I just treated as a blob: https://github.com/z88dk/z88dk/blob/master/libsrc/math/mbf32/z80/divsup.asm 

Don't suppose you know of an MSX disassembly? Playing around with the BCD maths there might be fun.

zx70, 2021-12-03:
Sure I do, look in our z88dk documentation git area, targets/msx.
By the way, It's almost ready:
[image]
..

zx70, 2021-12-13:
@suborb, please wait a bit more before looking at the MSX disasm.   I just discovered that the recently released GWBASIC sources include comments which can be retrofitted to BASIC-80.

feilipu, 2021-12-13:
> @suborb, please wait a bit more before looking at the MSX disasm. I just discovered that the recently released GWBASIC sources include comments which can be retrofitted to BASIC-80.

The [CP/M MS Basic 5.2 Source](https://winworldpc.com/download/c2bfc380-c2b2-c2b4-7d00-11c3a7c29d25) is also available. It has extensive comments etc on all the code sections. For example from the floating point file in the package `F4.MAC`.

```
;
;	EXTERNAL LOCATIONS USED BY THE MATH-PACKAGE
;	;THE FLOATING ACCUMULATOR
;IFN	LENGTH&2,<
;	BLOCK	1		;[TEMPORARY LEAST SIGNIFICANT BYTE]
;DFACLO:	BLOCK	4>		;[FOUR LOWEST ORDERS FOR DOUBLE PRECISION]
;FACLO:	BLOCK	3		;[LOW ORDER OF MANTISSA (LO)]
;				;[MIDDLE ORDER OF MANTISSA (MO)]
;				;[HIGH ORDER OF MANTISSA (HO)]
;FAC:	BLOCK	2		;[EXPONENT]
;				;[TEMPORARY COMPLEMENT OF SIGN IN MSB]
;IFN	LENGTH&2,<
;	BLOCK	1		;[TEMPORARY LEAST SIGNIFICANT BYTE]
;ARGLO:	BLOCK	7		;[LOCATION OF SECOND ARGUMENT FOR DOUBLE
;ARG:	BLOCK	1>		; PRECISION]
;FBUFFR:	BLOCK	^D13		;BUFFER FOR FOUT
;IFN	LENGTH&2,<BLOCK	^D<35-13>>
;
;
;THE FLOATING POINT FORMAT IS AS FOLLOWS:
;
;THE SIGN IS THE FIRST BIT OF THE MANTISSA
;THE MANTISSA IS 24 BITS LONG
;THE BINARY POINT IS TO THE LEFT OF THE MSB
;NUMBER = MANTISSA * 2 ^ EXPONENT
;THE MANTISSA IS POSITIVE, WITH A ONE ASSUMED TO BE WHERE THE SIGN BIT IS
;THE SIGN OF THE EXPONENT IS THE FIRST BIT OF THE EXPONENT
;THE EXPONENT IS STORED IN EXCESS 200 I.E. WITH A BIAS OF 200
;SO, THE EXPONENT IS A SIGNED 8-BIT NUMBER WITH 200 ADDED TO IT
;AN EXPONENT OF ZERO MEANS THE NUMBER IS ZERO, THE OTHER BYTES ARE IGNORED
;TO KEEP THE SAME NUMBER IN THE FAC WHILE SHIFTING:
;	TO SHIFT RIGHT,	EXP:=EXP+1
;	TO SHIFT LEFT,	EXP:=EXP-1
;
;SO, IN MEMORY THE NUMBER LOOKS LIKE THIS:
;	[BITS 17-24 OF THE MANTISSA]
;	[BITS 9-16 OF THE MANTISSA]
;	[THE SIGN IN BIT 7, BITS 2-8 OF THE MANTISSA ARE IN BITS 6-0]
;	[THE EXPONENT AS A SIGNED NUMBER + 200]
;(REMEMBER THAT BIT 1 OF THE MANTISSA IS ALWAYS A ONE)
;
;ARITHMETIC ROUTINE CALLING CONVENTIONS:
;
;FOR ONE ARGUMENT FUNCTIONS:
;	THE ARGUMENT IS IN THE FAC, THE RESULT IS LEFT IN THE FAC
;FOR TWO ARGUMENT OPERATIONS:
;	THE FIRST ARGUMENT IS IN B,C,D,E I.E. THE "REGISTERS"
;	THE SECOND ARGUMENT IS IN THE FAC
;	THE RESULT IS LEFT IN THE FAC
;
;THE "S" ENTRY POINTS TO THE TWO ARGUMENT OPERATIONS HAVE (HL) POINTING TO
;THE FIRST ARGUMENT INSTEAD OF THE FIRST ARGUMENT BEING IN THE REGISTERS.
;MOVRM IS CALLED TO GET THE ARGUMENT IN THE REGISTERS.
;THE "T" ENTRY POINTS ASSUME THE FIRST ARGUMENT IS ON THE STACK.
;POPR IS USED TO GET THE ARGUMENT IN THE REGISTERS.
;NOTE: THE "T" ENTRY POINTS SHOULD ALWAYS BE JUMPED TO AND NEVER CALLED
;BECAUSE THE RETURN ADDRESS ON THE STACK WILL BE CONFUSED WITH THE NUMBER.
;
;ON THE STACK, THE TWO LO'S ARE PUSHED ON FIRST AND THEN THE HO AND SIGN.
;THIS IS DONE SO IF A NUMBER IS STORED IN MEMORY, IT CAN BE PUSHED ON THE
;STACK WITH TWO PUSHM'S.  THE LOWER BYTE OF EACH PART IS IN THE LOWER
;MEMORY ADDRESS SO WHEN THE NUMBER IS POPPED INTO THE REGISTERS, THE HIGHER
;ORDER BYTE WILL BE IN THE HIGHER ORDER REGISTER OF THE REGISTER PAIR, I.E.
;THE HIGHER ORDER BYTE WILL BE POPPED INTO B, D OR H.
;%
```

zx70, 2021-12-15:
Felipu, that source comments are a goldmine.
the version I'm hacking is slightly more modern, but they're almost the same. 
The MSC BASIC has several similarities to that version too but I don't think it has the BCD FP variant.
That could possibly be understood looking at the GWBASIC sources.

.. a bit more of graphics teaser in attachment  :D
[image]

feilipu, 2021-12-29:
Well, following a circuitous route through the `l_` functions and the string library, I've finally arrived at the desired outcome.

[__CPM-IDE for the 8085__](https://github.com/RC2014Z80/RC2014/tree/master/ROMs/CPM-IDE) built with z88dk.

I think that's pretty well all of the major things done here that warrant holding this issue open.
AFAIK, 8085 support for the RC2014 is now done.

Thanks again for all the support along the way. :+1: :100:

closed 2021-12-29 by feilipu