z88dk/z88dk #1824: -list generating fake instructions

headkaze, 2021-07-24:
Z80 asm code generated by SCCZ80 using the -list option will create some fake instructions that when compiled using SjASM will split into two instructions.

Here's an example:
```
sub	a, (hl)
```
Will become:
```
sub a
sub (hl)
```
Here's some other examples and what they should be:
```
sub a, e => sub e
sub a, c => sub c
sub a,0x08 => sub 0x08
xor a,a => xor a
```
You can see the pattern here.

suborb, 2021-07-24:
Although not part of the Zilog notes I’d never heard of an assembler that didn’t treat the a, part of the 8 bit alu opcodes as optional. 

Not reporting a violation of the spec (if not actually a violation of common usage) as a warning and actually generating two instructions seems to me to be a bug in sjasm. 

List files aren’t meant to be assembled - they are intended for use either for eyeballing or parsing for debugging.

closed 2021-07-24 by suborb

headkaze, 2021-07-24:
> List files aren’t meant to be assembled - they are intended for use either for eyeballing or parsing for debugging.
It's useful for our team to be able to assemble them. so much so that I have written a tool called Lis2Asm. SjASM does have an option called  --nofakes (deprecated as --syntax=F) but it does not detect these issues. Can we perhaps add an option such as --list-strict to convert these cases to SjASM compatible source files?

ped7g, 2021-07-24:
@headkaze ah, you mean zoom's sjasmplus (sjasm has no --syntax options and has overall a bit different syntax and features).
You can use `--syntax=a` to change multi-arg delimiter from single comma to double-comma, then non-Zilog syntax like `sub a,e` will produce `sub e` instruction. (but `push de,hl,bc` then requires double-commas like `push de,,hl,,bc` .... I always recommend for new projects to use `--syntax=abf`).

Also it's not fake instructions, those are something different, like `ld de,hl` and similar, which are split into Z80 instructions by sjasmplus, while what you have hit is "multi-arg" feature, which I'm not huge fan of ... something like `ld h,d,l,e,a,43,b,22` does look to me as mess, but some users use it, so I didn't dare to remove it, just added that "a" option to hide it behind other delimiter.

And C compilers usually can produce asm file as output (like `cc -S`), although z80asm and sjasmplus syntax is different enough to make this pain one way or another, but processing list file sounds a bit suspicious.